RPi-Distro/repo #276: Hard Drive Eject

g1gsw, 2021-11-23:
I have 2 hard drives and when I plug them in to a USB port they will fail to eject after use. Clicking the eject button or doing it from the command line give the same error of "Failed to eject XXXX" Before installing Bullseye both ejected fine and I did do a clean install not an upgrade.

Both drives are WD Scorpio Blue WD2500BEVT  I am usng the adapters to connect them from Te PiHut
https://thepihut.com/collections/raspberry-pi-sd-cards-and-adapters/products/ssd-to-usb-3-0-cable-for-raspberry-pi

g1gsw, 2022-01-10:
This now appears to be fixed.

XECDesign, 2022-01-11:
Hmm, not sure if anything has changed on our end that would fix this. Glad it's working though.