RPi-Distro/repo #278: Direwolf package not compiled for ARM 6L on Bullseye - gives "Illegal instruction" when trying to run Direwolf 1.6 on Raspberry Pi 1

dranch, 2021-11-30:
Hello Rpi packagers,

It seems the Direwolf package in the Bullseye repo is being compiled for the ARM 7 instruction set and not the ARM6L for compatibility.  This is impacting users using the Rpi v1, 1+, Zero, and ZeroW SBCs.  Can a new package be published which is compatible for your legacy SBCs?  Maybe in the future, there can be different repos published for different generations of SBCs to maximize performance (ARM6L, ARM7, ARM8, etc).  Anyway.. thank you for all that you all do!

References:
   https://groups.io/g/direwolf/topic/87165711#5832
   https://groups.io/g/direwolf/topic/87406630#5855

--David

XECDesign, 2022-01-13:
Hello,

Since this is a Raspbian package and not an archive.raspberrypi.org one, we don't build or maintain it.

Raspbian issues should be reported here: https://bugs.launchpad.net/raspbian

Cheers

closed 2022-01-13 by XECDesign

dranch, 2022-07-08:
This issue has been reported at https://bugs.launchpad.net/raspbian/+bug/1980899?comments=all by another user.  Adding this here to tie the two tickets together.