RPi-Distro/repo #279: libdeflate0:armhf & libgcrypt20:armhf are compiled targeting armv8 architecture on bullseye

wagnerch, 2021-12-05:
This doesn't affect me since I have Pi4's, but thought I would pass it on if you guys want to keep supporting the older devices.  This is likely the cause of the Chromium issues on Pi0 & Pi1.

libgcrypt20 is a dependency for Chromium.

```
$ readelf -A /lib/arm-linux-gnueabihf/libdeflate.so.0
Attribute Section: aeabi
File Attributes
  Tag_CPU_name: "8-A"
  Tag_CPU_arch: v8
  Tag_CPU_arch_profile: Application
  Tag_ARM_ISA_use: Yes
  Tag_THUMB_ISA_use: Thumb-2
  Tag_FP_arch: VFPv2
  Tag_ABI_PCS_wchar_t: 4
  Tag_ABI_FP_denormal: Needed
  Tag_ABI_FP_exceptions: Needed
  Tag_ABI_FP_number_model: IEEE 754
  Tag_ABI_align_needed: 8-byte
  Tag_ABI_align_preserved: 8-byte, except leaf SP
  Tag_ABI_enum_size: int
  Tag_ABI_VFP_args: VFP registers
  Tag_CPU_unaligned_access: v6
  Tag_MPextension_use: Allowed
  Tag_Virtualization_use: TrustZone and Virtualization Extensions
```

```
$ readelf -A /lib/arm-linux-gnueabihf/libgcrypt.so.20
Attribute Section: aeabi
File Attributes
  Tag_CPU_name: "8-A"
  Tag_CPU_arch: v8
  Tag_CPU_arch_profile: Application
  Tag_ARM_ISA_use: Yes
  Tag_THUMB_ISA_use: Thumb-2
  Tag_FP_arch: FP for ARMv8
  Tag_Advanced_SIMD_arch: NEON for ARMv8
  Tag_ABI_PCS_wchar_t: 4
  Tag_ABI_FP_denormal: Needed
  Tag_ABI_FP_exceptions: Needed
  Tag_ABI_FP_number_model: IEEE 754
  Tag_ABI_align_needed: 8-byte
  Tag_ABI_enum_size: int
  Tag_ABI_VFP_args: VFP registers
  Tag_CPU_unaligned_access: v6
  Tag_MPextension_use: Allowed
  Tag_Virtualization_use: TrustZone and Virtualization Extensions
```

XECDesign, 2021-12-06:
As far as I can tell, there's a run-time test for whether to use a NEON implementation of a function or not, so readelf will be a bit misleading in such cases. Running the binaries from libdeflate-tools seems to work on a pi zero.

Chromium is a separate issue, it's build specifically for armv7. There are some patches needed to armv6 support and I can take a look at bringing those back when there's more time. Given how poorly it runs with limited RAM, it might not be worth the effort.

XECDesign, 2021-12-06:
libgcrypt seems to be used by a lot of programs. If it didn't work on armv6, I think there is no way it wouldn't have been spotted earlier, so I'm certain it's also using run-time cpu detection.

Raspbian packages go through an armv7 contamination check and AIUI packages need to be checked and whitelisted before they can end up in their repo.

wagnerch, 2021-12-06:
@XECDesign Yeah, your right I went through the implementation in both libraries and they do appear to do this at runtime.  libdeflate is using __attributes__ to call out armv8+crc versions of crc32, but it compiles 2 functions and selects them at runtime.  So I think both are false flags, apologies for the distraction.

closed 2021-12-06 by wagnerch

XECDesign, 2021-12-06:
No worries, it doesn't hurt to be cautious about these things.