RPi-Distro/repo #280: dhcpcd.service restarts ntp.service after a router advertisement

Dschubba, 2021-12-16:
Hi,
I've got a DCF77 receiver set up on my Raspberry Pi, which get the (somewhat) accurate time from the sender in Mainflingen. 

The ntp needs around 4 Minutes to receive and validate the signal and use it as a primary timesource aka. sys.peer.

My router (a fritz.box) sends advertisements every 5 Minutes or so. This means the ntp server is restarted from dhcpcd after it has set the correct time after 4 Minutes. 

This behavior is new in Bullseye and did not happen in the Distribution release with Buster.

Here is a log snippet from the actual behavior:

[NtpLog.txt](https://github.com/RPi-Distro/repo/files/7727029/NtpLog.txt)

My current workaround is to disable the dhcpcd - this leaves the ntp server/daemon alone and running.

I would be happy to know if this is a bug or a feature ;-)

XECDesign, 2022-01-06:
As far as I can tell, it's a feature.

Our bullseye dhcpcd5 package is closer to Debian's version and contains their hooks, which are meant to integrate better with different ntp clients. I believe this might be the line responsible for the restart:

https://salsa.debian.org/smlx-guest/dhcpcd5/-/blob/debian/7.1.0-2/debian/hooks/66-ntp.conf#L29

Dschubba, 2022-01-06:
Hi - thanks for the info. Commenting out this line restores the old behavior.