RPi-Distro/repo #282: Bullseye background tear around windows.

audetto, 2022-01-04:
Hi,

I've just upgraded to Bullseye and noticed tear when moving windows around.
I have a Pi400 fully upgraded, connected to a HDMI screen.
Same hardware never showed anything like this before with Buster.

Look carefully just above the mouse pointer and you will see an area with a lighter shade to the left.
And a darker line running horizontally to the left just below the task bar.

The dark bits are good, the light bits are bad. So the window causes a 1cm lighter area on the desktop which should not be there.
When the window is moved, this area flickers and eventually settles.

Happens as well when the terminal undergoes a lot of text scrolling.

I see the window decoration to have a gradient, of an intensity comparable to the tear. Could it be some window effect that overflows on the area outside the window?

[image]

XECDesign, 2022-01-04:
@spl237

spl237, 2022-01-04:
I can see it on a system here. My assumption is that it is some weird interaction between how mutter draws window shadows and how pcmanfm draws the desktop. Given it is basically a transient cosmetic effect, and it would probably be days of work to find out why it happens, I'm inclined to mark this as won't fix; it may well just go away as a result of future changes to X / window manager anyway.

closed 2022-01-04 by spl237

audetto, 2022-01-04:
Can we at least switch off any non necessary cosmetic effects?
If is a bit of a nonsense to enable shadows if the only thing they achieve is a tear.

spl237, 2022-01-04:
This effect isn't even noticeable on many systems - it is only visible with a light contrasting background; I've been running mutter for nearly a year, and hadn't seen it until you pointed it out! The window shadows make a big difference to the look and feel of the desktop. So no, we won't be turning off the window shadows; they're an inherent feature of mutter anyway, and it may not even be possible to disable them.

audetto, 2022-01-04:
I disagree. The effect is very visible when a window is moved over another window. Not just when the background image is light.
Open a File Manager and move any other window over it, towards the top of the desktop.

The pattern is very strange, only happens at the top of the screen, but once it happens, it continues even when the window is moved down.

If this is the price to pay for shadows, it is way too high.

spl237, 2022-01-04:
Window shadows are a hard-coded feature of mutter - as far as I am aware, it is not possible to disable them without changing and rebuilding the code. 

This is a minor graphical glitch which is barely noticeable in most circumstances. It may well be fixed as we further tweak mutter and the underlying graphic drivers, but it isn't a priority.

If you really cannot put up with it, you will need to either find how to turn shadows off in mutter and rebuild it yourself, or switch back to openbox.