RPi-Distro/repo #285: [Bullseye]: regression in OpenGL ES 2 performance, down more than 30%

audetto, 2022-01-15:
I have a Pi400 and recently updated to Bullseye, from Buster.

I noticed that the emulator I develop had a big drop in performance.
It uses SDL2 + OpenGL ES 2

When running fullscreen inside X, it could reach a fantastic speed of 140 FPS (with vsync disabled).
Now it only gets to 95 FPS when running with openbox, and only to 70 FPS with the default mutter.

A non fullscreen window always run slower and the speed dropped from 70 FPS to 60 or 45.

This brings the speed very close to 60 FPS when it starts to droop frames.

In all cases I use the full KMS driver: `dtoverlay=vc4-kms-v3d` (which, at least in Buster) made a massive improvement to OpenGL ES 2 performance

It seems that the 2 distros use the same kernel 5.10.63.
Has anything changed between the 2 scenarios?

popcornmix, 2022-01-18:
Lots has no doubt changed, but identifying it may be tricky.
My vague understanding is 3d rendering is done by a mixture of mesa and kernel.
glamor provides gl acceleration for X (and is required for using gl under X).
There may be a combination of fbturbo, xcompmgr, mutter, openbox used for composition and window manager.

You've identified that mutter costs some performance (but not all).
What's the version of mesa on each? It should be possible to build your own mesa to ensure they are the same.
Checking which of the other bits are running on each system (and versions) may indicate a difference.

Have you tried rendering without X, e.g. using GBM (see [kmscube](https://gitlab.freedesktop.org/mesa/kmscube)).
I suspect that may have better performance and rules out some of the X libs.

audetto, 2022-01-30:
It is curious that on Buster I get

mesa 21.2.2
sdl 2.0.14

while on bullseye I get

mesa 20.3.5
sdl 2.0.14

I double checked and they are both updated.

Moreover, running without X is not something I know how to do, nor I would like to.
Using X has so many advantages, and in buster, the speed is more than enough.

There must be some reference GLES app used for regression testing when creating a new Pi Distro or Kernel? Do you know what is used?

audetto, 2022-02-09:
You are right. I must have got the versions while I was ssh into Ubuntu 21.10.

So, the versions on Buster are

Mesa 19.3.2, SDL 2.0.9

audetto, 2022-09-26:
Something has changed.
Now it is back at the original good speed.

Nothing has changed on my code, so must have been a fix on the OS side.

XECDesign, 2022-09-27:
Hmm, I don't think anything has changed on our end either.

If the issue comes back, could you please send the exact list of packages that was updated to re-introduce the problem? You should be able to find that in /var/log/apt/history.log

closed 2022-09-27 by XECDesign