RPi-Distro/repo #286: dhcpcd spams logs with "is reachable again" message [patch]

normanr, 2022-01-16:
Logs get `dhcpcd: eth0: fe80::aabb:ccff:fedd:eeff is reachable again` every time there's a router announcement. This was fixed upstream in https://github.com/NetworkConfiguration/dhcpcd/commit/0228659a8a539852e41781155d229cc976474ba5 (backported to be included in the dhcpcd-8.1.3 release as https://github.com/NetworkConfiguration/dhcpcd/commit/aee911790a9303a74b5cad78226744170a7f3c9b). Please could you apply that to the raspberrypi bullseye package. (Note that dhcpcd-8.1.9 is available, so perhaps worth upgrading to that too).

XECDesign, 2022-01-17:
Always great when a bug report contains the fix. Many thanks!

The updated package should be live now.

I didn't opt to upgrade dhcpcd to a newer version because it has never gone smoothly in the past.

normanr, 2022-01-17:
Yea, at least upgrading to a newer minor version (8.x) should be less pain than a major version (9.x).

closed 2022-02-01 by no-response[bot]