RPi-Distro/repo #287: Bullseye 64 Bit - xrdp presents a black screen

MrYsLab, 2022-02-03:
This problem does not present itself in buster. When I try to connect using xrdp, I do not see the desktop, but a black screen. Below is the xrdp log:
```
[20220202-16:07:01] [INFO ] address [0.0.0.0] port [3389] mode 1
[20220202-16:07:01] [INFO ] listening to port 3389 on 0.0.0.0
[20220202-16:07:01] [INFO ] xrdp_listen_pp done
[20220202-16:07:03] [INFO ] starting xrdp with pid 1339
[20220202-16:07:03] [INFO ] address [0.0.0.0] port [3389] mode 1
[20220202-16:07:03] [INFO ] listening to port 3389 on 0.0.0.0
[20220202-16:07:03] [INFO ] xrdp_listen_pp done
[20220202-16:07:15] [INFO ] Socket 12: AF_INET6 connection received from ::ffff:192.168.2.104 port 53632
[20220202-16:07:16] [INFO ] Using default X.509 certificate: /etc/xrdp/cert.pem
[20220202-16:07:16] [INFO ] Using default X.509 key file: /etc/xrdp/key.pem
[20220202-16:07:16] [ERROR] Cannot read private key file /etc/xrdp/key.pem: Permission denied
[20220202-16:07:16] [INFO ] Connected client computer name: theLab
[20220202-16:07:16] [WARN ] Received [MS-RDPBCGR] TS_UD_HEADER type 0xc006 is unknown (ignored)
[20220202-16:07:16] [WARN ] Received [MS-RDPBCGR] TS_UD_HEADER type 0xc00a is unknown (ignored)
[20220202-16:07:16] [INFO ] xrdp_load_keyboard_layout: Keyboard information sent by the RDP client, keyboard_type:[0x04], keyboard_subtype:[0x00], keylayout:[0x00000409]
[20220202-16:07:17] [INFO ] xrdp_load_keyboard_layout: model [] variant [] layout [us] options []
[20220202-16:07:17] [INFO ] Non-TLS connection established from ::ffff:192.168.2.104 port 53632: encrypted with standard RDP security
[20220202-16:07:17] [INFO ] xrdp_caps_process_pointer: client supports new(color) cursor
[20220202-16:07:17] [INFO ] xrdp_caps_process_codecs: RemoteFX, codec id 3, properties len 49
[20220202-16:07:17] [INFO ] Loading keymap file /etc/xrdp/km-00000409.ini
[20220202-16:07:17] [WARN ] local keymap file for 0x00000409 found and doesn't match built in keymap, using local keymap file
[20220202-16:07:17] [INFO ] connecting to sesman ip 127.0.0.1 port 3350
[20220202-16:07:17] [INFO ] xrdp_wm_log_msg: sesman connect ok
[20220202-16:07:17] [INFO ] sesman connect ok
[20220202-16:07:17] [INFO ] sending login info to session manager, please wait...
[20220202-16:07:17] [INFO ] xrdp_wm_log_msg: login successful for display 10
[20220202-16:07:17] [INFO ] login successful for display 10
[20220202-16:07:17] [INFO ] loaded module 'libxup.so' ok, interface size 10296, version 4
[20220202-16:07:17] [INFO ] started connecting
[20220202-16:07:17] [INFO ] lib_mod_connect: connecting via UNIX socket
[20220202-16:10:47] [INFO ] connection problem, giving up
[20220202-16:10:47] [INFO ] some problem
[20220202-16:10:47] [ERROR] xrdp_iso_send: trans_write_copy_s failed
[20220202-16:10:47] [ERROR] Sending [ITU T.125] DisconnectProviderUltimatum failed

```

XECDesign, 2022-02-16:
@spl237 Another xrdp issue.

spl237, 2022-02-16:
xrdp is not something we support, I'm afraid - I've got no idea why this won't work with mutter! Suggest you open a discussion on the forums; I know there have been some discussions about xrdp problems and workarounds on there.

closed 2022-02-16 by XECDesign

spl237, 2022-02-16:
FWIW, https://github.com/neutrinolabs/xrdp/discussions/2050 is a discussion about the details of xrdp running on bullseye, so people have got it working; it may be worth raising an issue with Neutrino Labs?

MrYsLab, 2022-02-16:
@spl237  Thanks. I will take a look.