RPi-Distro/repo #288: Add support for PCM 5.1 channel output to HDMI (patch)

horazont, 2022-02-05:
So I hope this is the right place. I have a proposal to modify the `/usr/share/pulseaudio/alsa-mixer/profile-sets/rpi-hdmi.conf` shipped by the `pulseaudio` package on Raspberry Pi OS. This file is not provided by stock Debian, so I assume that it is an add-on provided by Raspberry Pi OS, and hence this may be the right place to report it.

On bullseye, I see the following contents:

```ini
[General]
auto-profiles = yes

[Mapping digital-stereo]
description = Digital Stereo
device-strings = hw:%f
fallback = yes
priority = 1
direction = output
channel-map = left,right
paths-output = hdmi-output-0
```

I created an alternative version:

```ini
[General]
auto-profiles = yes

[Mapping digital-stereo]
description = Digital Stereo
device-strings = hw:%f
priority = 1
direction = output
channel-map = left,right
paths-output = hdmi-output-0

[Mapping hdmi-surround]
description = Digital Surround 5.1
device-strings = hw:%f
paths-output = hdmi-output-0
channel-map = front-left,front-right,lfe,center,rear-left,rear-right
priority = 3
direction = output
```

Changes:

- Added Mapping hdmi-surround:

  - The channel map was discovered by trial and error with a connected Denon AVR-1910.
  - The template was taken from some other profile with hdmi-surround

- Removed `fallback = yes` from `digital-stereo`: Otherwise, `digital-stereo` is not offered when `hdmi-surround` is detected correctly, as `fallback = yes` options are only shown if no non-fallback options are available.

In `pavucontrol`, this then offers `Digital Stereo` and `Digital Surround 5.1` profiles for the HDMI sink, which allows online switching between the profiles and to output 5.1 PCM sound via the HDMI output via pulseaudio, without having to resort to any `pacmd load-module module-alsa-sink […]` hacks.

It would be really nice if this change could be shipped with the OS so that users don't have to tinker as much with the options to get it to work.

XECDesign, 2022-02-05:
I don't know enough about alsa to know whether there is a reason the file is as it is. @popcornmix, should we pick up these changes?

horazont, 2022-02-05:
FWIW, I tested (and am using) this profile set on a Raspberry Pi 2B.

popcornmix, 2022-02-05:
It looks plausible. I'll test it Monday with AVR. Might be worth just checking it doesn't break anything with normal stereo use.

XECDesign, 2022-02-17:
@popcornmix how did it go?