RPi-Distro/repo #290: When changing volume settings from the terminal with `pactl` the volume panel applet is not updated

lucatrv, 2022-02-19:
This issue can be reproduced by issuing the following command from the terminal:
```
pactl set-sink-volume @DEFAULT_SINK@ 50%
```

The volume panel applet is not updated and still shows the previous volume setting. You need to right click on it to update.

When trying the same under GNOME, its volume panel applet is updated automatically as expected.

XECDesign, 2022-02-21:
@spl237

spl237, 2022-02-21:
wontfix - the plugin would need to constantly poll the volume to see if it has changed, which is inefficient. The volume should only be changed using the plugin, which is what it is there for!

closed 2022-02-21 by spl237

lucatrv, 2022-03-30:
Apparently this issue has been fixed anyway, the volume panel applet is now automatically updated when changing volume settings from the terminal with `pactl`.