RPi-Distro/repo #294: No mirrors in sync

isaacngym, 2022-03-24:
Hi,

I just created a fresh install of raspberry pi os lite (32-bit) an a pi zero w. Trying to run `sudo apt update` with a variety of mirrors in my `/etc/apt/sources.list` file has resulted in the following error message: 

```
E: Failed to fetch http://103.1.138.206/archive.raspberrypi.org/debian/dists/bullseye/main/binary-armhf/Packages.gz  File has unexpected size (264264 != 264427). Mirror sync in progress? [IP: 103.1.138.206 80]
   Hashes of expected file:
    - Filesize:264427 [weak]
    - SHA256:65aff89f7ec7f8c049c6ff1891d0544bc92bf14bac06c4f7b76e6b635d48260c
    - SHA1:d49148ce5f3f98cc83eda81c34e9865c325b9fd1 [weak]
    - MD5Sum:2ed396d193ecfce6bc9beb6f3514cb2d [weak]
   Release file created at: Tue, 22 Mar 2022 14:53:57 +0000
E: Some index files failed to download. They have been ignored, or old ones used instead.
```

I've seen that there was someone facing a similar issue in the past which was caused by the mirrors to fail to sync. Was wondering if that's what's going on here or what I could do with it.

isaacngym, 2022-03-25:
running `dig archive.raspberrypi.org +short` gives me:
```
pi@raspberrypi:~ $ dig archive.raspberrypi.org +short
lb.raspberrypi.org.
lb.raspberrypi.com.
93.93.135.117
176.126.240.84
176.126.240.167
46.235.231.151
93.93.135.118
46.235.231.111
46.235.230.122
46.235.227.39
93.93.135.141
176.126.240.86
46.235.231.145
93.93.130.212
```

XECDesign, 2022-03-25:
All I can say is that everything looks okay on this end. It may be better to take general network issues to the forum: https://forums.raspberrypi.com/

I'd start with trying a different nameserver. It looks like your ISP is trying to cache things, but failing to do so properly.

closed 2022-03-25 by XECDesign