RPi-Distro/repo #295: Bullseye does not wake from the screen saver on input activity

mphare, 2022-03-28:
I am using a Pi 3B+, a fresh install of 32-bit Bullseye, no other apps loaded. I have a Logitech K120 keyboard, a Logitech B100 mouse and an Acer 27" HDMI FHD monitor. 

Mouse or keyboard activity does not wake the Pi up after the screen saver kicks in. The only recourse is to power down and then back up.

I reverted back to 32-bit Buster with the same hardware set and everything works as expected.

Garylhawkins2, 2022-04-21:
Just to add on , i have 3 x 3B  one with a sense hat installed and all 3 with wired KB and mouse or the Logitech k400r wireless kb/mouse does not wake up from the screen blanking . This only seems to have started after a apt-get update and a fresh image of the OS. 
I will be trying a fresh install of an older build to test and will update my comment .

Garylhawkins2, 2022-04-22:
> 

OK, so as a follow up . I did some tests last night with base installs from the last 3 builds of 32 and 64 bit versions and both the 4-4-22 and 1-28-22 builds exhibit the issue where the screen will blank after idle time and will not "wake" back up. I also noticed that it acted like it had "locked" up as i was not able to light up the num lock or caps lock keys .  After disconnecting and reconnecting them the KB seemed to work again for a short time but still did not wake the video back up .  
The 10-28-21 build did not seem to show either of these behaviors.
3 Pi's tested , all 3B models , 2 with wired KB and mouse (various brands of what ever was laying around) and 3rd had Logitech k400r with unifying receiver . All functioned normally at first boot . 

I did test one and it was available via SSH connection so the Pi is still active and i do see activity lights on all of them.

I hope this helps whoever is monitoring this  

If i can provide more info let me know ,

Garylhawkins2, 2022-04-22:
Ok, so this may help, rebooted one of the nodes and started a Tail on the log files and saw this happen when it went into "sleep" mode. See the attached text file 

[New Text Document.txt](https://github.com/RPi-Distro/repo/files/8543007/New.Text.Document.txt)

MarkusPiotrowski, 2022-07-07:
There are several people reporting the same problem: https://forums.raspberrypi.com/viewtopic.php?t=328904&sid=554ae59c2ca60a5be97a9a8d06b7645d&start=25

I also experience this problem on a Raspberry Pi 3B (+?) and the recent version of Pi OS (32 bit).

When using VNC I can see that the Raspberry is still alive and active (mouse and keyboard working) but it seems it is not sending a "wake up" signal to the monitor.

popcornmix, 2022-07-08:
From log of @MarkusPiotrowski it looks like a duplicate of https://github.com/raspberrypi/linux/issues/4962

closed 2022-07-08 by XECDesign