RPi-Distro/repo #296: isc-dhcp-server.service fails upon installation on Bullseye

mattdibi, 2022-03-30:
As per title isc-dhcp-server.service fails upon installation on Raspberry Pi OS Bullseye.

The issue stems from the fact that:
- The server is started as soon as it's installed
- The configuration generated at installation time doesn't specify any interface on which to run the server.

**To Reproduce**
Steps to reproduce the behavior:
1. On a fresh Raspberry Pi OS Bullseye install `isc-dhcp-server` with `sudo apt install isc-dhcp-server`
2. See error in the installation log

```
pi@raspberrypi:~ $ sudo apt install isc-dhcp-server
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
...
Setting up isc-dhcp-server (4.4.1-2.3) ...
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_ALL to default locale: No such file or directory
Generating /etc/default/isc-dhcp-server...
Job for isc-dhcp-server.service failed because the control process exited with error code.
See "systemctl status isc-dhcp-server.service" and "journalctl -xe" for details.
invoke-rc.d: initscript isc-dhcp-server, action "start" failed.
● isc-dhcp-server.service - LSB: DHCP server
     Loaded: loaded (/etc/init.d/isc-dhcp-server; generated)
     Active: failed (Result: exit-code) since Tue 2022-03-29 08:32:17 BST; 49ms ago
       Docs: man:systemd-sysv-generator(8)
    Process: 4048 ExecStart=/etc/init.d/isc-dhcp-server start (code=exited, status=1/FAILURE)
        CPU: 80ms

Mar 29 08:32:15 raspberrypi dhcpd[4063]: before submitting a bug.  These pages explain the proper
Mar 29 08:32:15 raspberrypi dhcpd[4063]: process and the information we find helpful for debugging.
Mar 29 08:32:15 raspberrypi dhcpd[4063]:
Mar 29 08:32:15 raspberrypi dhcpd[4063]: exiting.
Mar 29 08:32:17 raspberrypi isc-dhcp-server[4048]: Starting ISC DHCPv4 server: dhcpdcheck syslog for diagnostics. ...
Mar 29 08:32:17 raspberrypi isc-dhcp-server[4068]:  failed!
```

**Additional Information**

Device: Raspberry Pi 4

```
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 5.10.103-v7l+ #1530 SMP Tue Mar 8 13:05:01 GMT 2022 armv7l GNU/Linux
```

```
pi@raspberrypi:~ $ lsb_release -a
No LSB modules are available.
Distributor ID:	Raspbian
Description:	Raspbian GNU/Linux 11 (bullseye)
Release:	11
Codename:	bullseye
```

Complete installation log:[dhcp-server-install-log.txt](/uploads/17b0d8f581db9f7810805e72c0fb30d5/dhcp-server-install-log.txt)

Complete error log:

```
-- Journal begins at Fri 2022-01-28 01:22:52 GMT, ends at Wed 2022-03-23 10:22:30 GMT. --
Mar 23 10:21:32 raspberrypi systemd[1]: Starting LSB: DHCP server...
Mar 23 10:21:32 raspberrypi isc-dhcp-server[2226]: Launching both IPv4 and IPv6 servers (please configure INTERFACES in /etc/default/isc-dhcp-server if you o>
Mar 23 10:21:32 raspberrypi dhcpd[2241]: Wrote 0 leases to leases file.
Mar 23 10:21:32 raspberrypi dhcpd[2241]:
Mar 23 10:21:32 raspberrypi dhcpd[2241]: No subnet declaration for eth0 (192.168.1.63).
Mar 23 10:21:32 raspberrypi dhcpd[2241]: ** Ignoring requests on eth0.  If this is not what
Mar 23 10:21:32 raspberrypi dhcpd[2241]:    you want, please write a subnet declaration
Mar 23 10:21:32 raspberrypi dhcpd[2241]:    in your dhcpd.conf file for the network segment
Mar 23 10:21:32 raspberrypi dhcpd[2241]:    to which interface eth0 is attached. **
Mar 23 10:21:32 raspberrypi dhcpd[2241]:
Mar 23 10:21:32 raspberrypi dhcpd[2241]:
Mar 23 10:21:32 raspberrypi dhcpd[2241]: Not configured to listen on any interfaces!
Mar 23 10:21:32 raspberrypi dhcpd[2241]:
Mar 23 10:21:32 raspberrypi dhcpd[2241]: If you think you have received this message due to a bug rather
Mar 23 10:21:32 raspberrypi dhcpd[2241]: than a configuration issue please read the section on submitting
Mar 23 10:21:32 raspberrypi dhcpd[2241]: bugs on either our web page at www.isc.org or in the README file
Mar 23 10:21:32 raspberrypi dhcpd[2241]: before submitting a bug.  These pages explain the proper
Mar 23 10:21:32 raspberrypi dhcpd[2241]: process and the information we find helpful for debugging.
Mar 23 10:21:32 raspberrypi dhcpd[2241]:
Mar 23 10:21:32 raspberrypi dhcpd[2241]: exiting.
Mar 23 10:21:34 raspberrypi isc-dhcp-server[2226]: Starting ISC DHCPv4 server: dhcpdcheck syslog for diagnostics. ...
Mar 23 10:21:34 raspberrypi isc-dhcp-server[2246]:  failed!
Mar 23 10:21:34 raspberrypi isc-dhcp-server[2247]:  failed!
Mar 23 10:21:34 raspberrypi systemd[1]: isc-dhcp-server.service: Control process exited, code=exited, status=1/FAILURE
Mar 23 10:21:34 raspberrypi systemd[1]: isc-dhcp-server.service: Failed with result 'exit-code'.
Mar 23 10:21:34 raspberrypi systemd[1]: Failed to start LSB: DHCP server.
```

Resulting configuration:

```
pi@raspberrypi:~ $ cat /etc/default/isc-dhcp-server
# Defaults for isc-dhcp-server (sourced by /etc/init.d/isc-dhcp-server)

# Path to dhcpd's config file (default: /etc/dhcp/dhcpd.conf).
#DHCPDv4_CONF=/etc/dhcp/dhcpd.conf
#DHCPDv6_CONF=/etc/dhcp/dhcpd6.conf

# Path to dhcpd's PID file (default: /var/run/dhcpd.pid).
#DHCPDv4_PID=/var/run/dhcpd.pid
#DHCPDv6_PID=/var/run/dhcpd6.pid

# Additional options to start dhcpd with.
#	Don't use options -cf or -pf here; use DHCPD_CONF/ DHCPD_PID instead
#OPTIONS=""

# On what interfaces should the DHCP server (dhcpd) serve DHCP requests?
#	Separate multiple interfaces with spaces, e.g. "eth0 eth1".
INTERFACESv4=""
INTERFACESv6=""
```

XECDesign, 2022-03-30:
```
Mar 23 10:21:32 raspberrypi dhcpd[2241]: No subnet declaration for eth0 (192.168.1.63).
Mar 23 10:21:32 raspberrypi dhcpd[2241]: ** Ignoring requests on eth0.  If this is not what
Mar 23 10:21:32 raspberrypi dhcpd[2241]:    you want, please write a subnet declaration
Mar 23 10:21:32 raspberrypi dhcpd[2241]:    in your dhcpd.conf file for the network segment
Mar 23 10:21:32 raspberrypi dhcpd[2241]:    to which interface eth0 is attached. **
Mar 23 10:21:32 raspberrypi dhcpd[2241]:
Mar 23 10:21:32 raspberrypi dhcpd[2241]:
Mar 23 10:21:32 raspberrypi dhcpd[2241]: Not configured to listen on any interfaces!
```

Looks like you just haven't configured it, so there's nothing for it to do.

mattdibi, 2022-03-30:
@XECDesign Hi, thank you for the quick feedback.

> Looks like you just haven't configured it, so there's nothing for it to do.

Yes, I know. The service fails **at installation time**, because it's immediately launched without having a configuration (i.e. it uses the one generated at installation time).

I expected it either:
- After installation to have a valid configuration auto-generated
- The service should not be launched immediately after installation.

XECDesign, 2022-03-30:
Oh right. Yeah, but that seems to be the intended behaviour of the package, even if it's a bit unusual.

The package is not from archive.raspberrypi.org (what this issue tracker is for), so it's probably best discussed with its maintainers over at bugs.debian.org.

mattdibi, 2022-03-30:
@XECDesign Ok thanks. I'll ask them about it.

closed 2022-06-29 by XECDesign