RPi-Distro/repo #297: Bullseye: udev rule gets truncated for long device paths

zackyancey, 2022-04-06:
Copy-paste of a [post](https://forums.raspberrypi.com/viewtopic.php?p=1990446#p1990446) in the Raspberry Pi OS forum, I was pointed here as a potentially better place to report the bug.

There's an issue with a udev rule messing up some device permissions: when I plug a device in through a usb hub, the entire PCI bus gets its permissions changed and can't be read anymore.

The problem is in `/usr/lib/udev/rules.d/60-rpi.gpio-common.rules`, line 3:

```
SUBSYSTEM=="bcm2835-gpiomem", KERNEL=="gpiomem", GROUP="dialout", MODE="0660"
SUBSYSTEM=="gpio", KERNEL=="gpiochip*", ACTION=="add", PROGRAM="/bin/sh -c 'chown root:dialout /sys/class/gpio/export /sys/class/gpio/unexport ; chmod 220 /sys/class/gpio/export /sys/class/gpio/unexport'"
SUBSYSTEM=="gpio", KERNEL=="gpio*", ACTION=="add", PROGRAM="/bin/sh -c 'chown root:dialout /sys%p/active_low /sys%p/direction /sys%p/edge /sys%p/value ; chmod 660 /sys%p/active_low /sys%p/direction /sys%p/edge /sys%p/value'"
```
enabling debug logging for udev, I see this when the device gets plugged in:

```
Mar 31 13:52:16 raspberrypi systemd-udevd[1553]: gpiochip2: /usr/lib/udev/rules.d/60-rpi.gpio-common.rules:3 Running PROGRAM '/bin/sh -c 'chown root:dialout /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4/1-1.2.4.4.1/1-1.2.4.4.1:1.0/gpiochip2/active_low /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4/1-1.2.4.4.1/1-1.2.4.4.1:1.0/gpiochip2/direction /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4/1-1.2.4.4.1/1-1.2.4.4.1:1.0/gpiochip2/edge /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4/1-1.2.4.4.1/1-1.2.4.4.1:1.0/gpiochip2/value ; chmod 660 /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4/1-1.2.4.4.1/1-1.2.4.4.1:1.0/gpiochip2/active_low /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4/1-1.2.4.4.1/1-1.2.4.4.1:1.0/gpiochip2/direction /sys/devices/platform/scb/fd500000.pcie'
Mar 31 13:52:16 raspberrypi systemd-udevd[1553]: gpiochip2: Starting '/bin/sh -c 'chown root:dialout /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4/1-1.2.4.4.1/1-1.2.4.4.1:1.0/gpiochip2/active_low /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4/1-1.2.4.4.1/1-1.2.4.4.1:1.0/gpiochip2/direction /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4/1-1.2.4.4.1/1-1.2.4.4.1:1.0/gpiochip2/edge /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4/1-1.2.4.4.1/1-1.2.4.4.1:1.0/gpiochip2/value ; chmod 660 /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4/1-1.2.4.4.1/1-1.2.4.4.1:1.0/gpiochip2/active_low /sys/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.2/1-1.2.4/1-1.2.4.4/1-1.2.4.4.1/1-1.2.4.4.1:1.0/gpiochip2/direction /sys/devices/platform/scb/fd500000.pcie'
```

The command that's executed doesn't match the command that's in the rules file--it gets truncated at 1023 characters. In this case, that happens to be right on a directory name, so the `chmod 660` gets applied to `/sys/devices/platform/scb/fd500000.pcie` instead of a specific gpio file.

I was able to work around the bug by splitting up that last line:

```
SUBSYSTEM=="bcm2835-gpiomem", KERNEL=="gpiomem", GROUP="dialout", MODE="0660"
SUBSYSTEM=="gpio", KERNEL=="gpiochip*", ACTION=="add", PROGRAM="/bin/sh -c 'chown root:dialout /sys/class/gpio/export /sys/class/gpio/unexport ; chmod 220 /sys/class/gpio/export /sys/class/gpio/unexport'"
SUBSYSTEM=="gpio", KERNEL=="gpio*", ACTION=="add", PROGRAM="/bin/sh -c 'chown root:dialout /sys%p/active_low'"
SUBSYSTEM=="gpio", KERNEL=="gpio*", ACTION=="add", PROGRAM="/bin/sh -c 'chown root:dialout /sys%p/direction'"
SUBSYSTEM=="gpio", KERNEL=="gpio*", ACTION=="add", PROGRAM="/bin/sh -c 'chown root:dialout /sys%p/edge'"
SUBSYSTEM=="gpio", KERNEL=="gpio*", ACTION=="add", PROGRAM="/bin/sh -c 'chown root:dialout /sys%p/value'"
SUBSYSTEM=="gpio", KERNEL=="gpio*", ACTION=="add", PROGRAM="/bin/sh -c 'chmod 660 /sys%p/active_low'"
SUBSYSTEM=="gpio", KERNEL=="gpio*", ACTION=="add", PROGRAM="/bin/sh -c 'chmod 660 /sys%p/direction'"
SUBSYSTEM=="gpio", KERNEL=="gpio*", ACTION=="add", PROGRAM="/bin/sh -c 'chmod 660 /sys%p/edge'"
SUBSYSTEM=="gpio", KERNEL=="gpio*", ACTION=="add", PROGRAM="/bin/sh -c 'chmod 660 /sys%p/value'"
```

That lets `%p` get a lot larger before we start running into the apparent 1023-character limit for `PROGRAM=` values.

XECDesign, 2022-04-13:
The rpi.gpio package seems to come from Debian now, so it's not in our repo.

The issue should be reported over at https://www.debian.org/Bugs/Reporting

If it gets fixed there, but doesn't make it to Bullseye, I could pull the updated package into our repo.

closed 2022-04-13 by XECDesign

zackyancey, 2022-04-13:
Thanks for pointing me in the right direction. Debian bug [1009654](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1009654) for anyone interested.