RPi-Distro/repo #298: Is there a virtual file to detect total RAM size?

Botspot, 2022-04-18:
I'm [working](https://github.com/hakandundar34coding/system-monitoring-center/issues/40#issuecomment-1101554347) with some other developers on a task manager. Many data values are empty because the RPi's kernel is not generating virtual files where a normal X86 machine would.

For example, the Pi does not seem to have an easy way to detect the entire RAM's capacity. We don't want just the amount available to the CPU, we want it all.
On an X86 machine this information would be found at `/sys/devices/system/memory/block_size_bytes`, but on the Pi there is no such file. Is there a Pi-specific way to do this? Maybe with `vcgencmd`?

XECDesign, 2022-06-29:
@popcornmix, @pelwell any idea about this one? I was hoping there would be something like `vcgencmd get_mem total`, but there doesn't seem to be.

pelwell, 2022-06-29:
> We don't want just the amount available to the CPU, we want it all.

What does this mean? There are many small memories scattered through the SoC - are you saying you want those included in the total? Why would inaccessible memory be of any interest?

Botspot, 2022-07-01:
> I was hoping there would be something like `vcgencmd get_mem total`, but there doesn't seem to be.

That's actually what we [ended up doing.](https://github.com/hakandundar34coding/system-monitoring-center/issues/78) `vcgencmd get_config total_mem`
This command seems to return the entire RAM chip's capacity, including inaccessible portions, which in this case is desirable. @pelwell.

At this point I would suggest that general system statistics should be reflected in standardized locations. It should not be necessary for a cross-platform task manager to use a proprietary, RPi-specific program to acquire data that is easily accessible on nearly all other Linux kernels.

XECDesign, 2022-07-01:
I'm not sure that `block_size_bytes` that qualified as a standardised way of doing this. It's a part of the memory hotplug driver, which can only be enabled on 64bit platforms. And it doesn't make much sense to have on the pi at all. I don't know what "the" cross-platform way of doing this is.

pelwell, 2022-07-01:
I'm not sure there needs to be a cross-platform way to describe how much RAM is fitted, only how much Linux is able to use.

Memory size is also encoded in the board revision number - `/proc/device-tree/system/linux,revision`. a big-endian uint32_t. Bits 20-22 encode the memory size, as explained in https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#new-style-revision-codes.

XECDesign, 2022-07-01:
Going to close this since it's more of a forum discussion than an issue with a package in the repo.

closed 2022-07-01 by XECDesign