RPi-Distro/repo #299: rpi-imager still at 1.7.1 for Buster - end of support?

beta-tester, 2022-05-02:
i have installed rpi-imager on my old RPi3 running RasPi OS Buster.
but i only get version 1.7.1 as newest versioninstalled.
and every time i start rpi-imager i get the annoying message, that there is a newer version availanle i have to install.
[image]

i thought with apt-get i always get the newest version installed and get updates as soon there is one available.
it buster already end of support?

XECDesign, 2022-05-03:
> i thought with apt-get i always get the newest version installed and get updates as soon there is one available.
> it buster already end of support?

Buster is now a ['legacy' release](https://www.raspberrypi.com/news/new-old-functionality-with-raspberry-pi-os-legacy/) and will only be getting security updates.

@ghollingworth What do you want to happen here?

I think we should patch out the update check in RPi OS, since they the updates should be handled entirely by apt. Then we can leave the Buster version behind without users getting the nag screen.

beta-tester, 2022-05-10:
> @ghollingworth What do you want to happen here?
> 
> I think we should patch out the update check in RPi OS, since they the updates should be handled entirely by apt. Then we can leave the Buster version behind without users getting the nag screen.

or change the "_action_" from visiting website to do an `apt update`/`upgrade` for rpi-imager in case rpi-imager is for a system that gets its rpi-imager via `apt`.

XECDesign, 2022-05-10:
Updated buster package to 1.7.2 and removed the version check. I'll remove the version check for the bullseye version when there's a new release.