RPi-Distro/repo #300: dhcpcd5 logging issue when inform option used

seamusdemora, 2022-05-13:
I'm a bit confused here, so if I'm in the wrong place, please advise. The README says `Issue tracking for the archive.raspberrypi.org repo`; that's where the `dhcpcd5` sources reside, so that's why I'm bringing this up here. 

This issue is re. `dhcpcd5`; it was brought up on April 4 in [another area](https://github.com/raspberrypi/linux/issues/4360#issuecomment-1087715222). That was clearly the wrong place to bring it up, but I only learned this repo existed yesterday. As I haven't seen any feedback, and just discovered this repo I thought I'd bring the issue here - which is _hopefully_ where it belongs?

As I mentioned, I submitted a PR against the upstream dhcpcd repo on Mar 31. Since then, I've patched my sources & tested on `bullseye` and `buster` hosts. I took this as an opportunity to learn a bit about the Debian packaging system, and I've created a patch file & updated the changelog. I'd like to share this with you (& anything else you need) to get this patch incorporated and released. 

If you're agreeable with that, please let me know how you'd like me to transfer the files to you, and if you need anything beyond the .patch file & the changelog.  There's a tab for pull requests here, but I'm under the impression that the `dhcpcd5` sources aren't in git?? AFAIK, all of the packages in the `archive.raspberrypi.org repo` are tarballs & .dsc files. Shall I attach the patch file & changelog here?

seamusdemora, 2022-05-13:
> I've mentioned in the other issue that there's a test build in the 'untested' component waiting for confirmation that it works as you expect. 

That isn't how I understood it... the Q&A went like this:

>>And finally, any idea as to when this will make it into the 'main' repo for update by apt?

>When I have confidence and independant verification that it does what's intended and doesn't cause other issues. 

Are you saying now that it's down to me? Have I been promoted to `independent verifier`?  :)

seamusdemora, 2022-09-11:
Out of curiosity, I noticed earlier today that a patch to `dhcpcd5` has finally been released. It contains the patch I submitted back on Mar 31, and the `changelog` indicates another patch also incorporated - described as follows:

>Add drop-in service to prevent dhcpcd from conflicting with NetworkManager 

Is this the "double logging" issue [reported here](https://github.com/RPi-Distro/raspi-config/issues/173)? 

Thnx - just trying to keep up.

XECDesign, 2022-09-14:
Yup, that patch has been hanging around the testing and internal repos for a while. Although it doesn't seem like it was an issue that anybody cared about enough to do any testing, it didn't seem to have much potential to cause issues.

The drop-in service just does what it says there. If NetworkManager is enabled, it will now prevent dhcpcd from starting. Nothing to do with the double logging issue, which was fixed in `1:8.1.2-1+rpt4`.

closed 2022-11-11 by XECDesign