RPi-Distro/repo #302: GPSD 3.22 does not function correctly with some Ublox devices

jeffreyscurtis, 2022-07-02:
Hello,

Running Raspberry PI Raspbian bullseye and Ublox compatible USB GPS receiver.

Install gpsd via apt-get which installs version gpsd 3.22 tune of system services for GPSD

disable gpsd system startup
 
Starting gpsd with the following command on:

gpsd -G -n -F /var/run/gpsd.sock /dev/serial0 /dev/pps0 

Run cgps

cgps will show 3D Fix for status but latitude, longitude, and altitude are all zero assuming the GPS can get lock.  If run after GPS already has lock then the lat/long/alt wont be zero but will be the value at startup and will not change

Run xgps

xgps shows 3d fix and updating correct ecef values, but latitude, longitude and altitude are all zeros. If run after GPS already has lock then the lat/long/alt wont be zero but will be the value at startup and will not change

Capture raw log with command gpspipe -R -x 20 > log.raw

Decode with gpsdecode < 'log.raw'

Latitude, Longitude, and Altitude are present and updating

telnet to port 2947 and issue command ?WATCH={"enable":true,"json":true}; 

Latitude, Longitude, and Altitude values are all zeros.

If you instead add the -p switch on starting gpsd all function normally, but the GPS receiver will not be initialized as a Ublox device and will run in NEMA mode only. -p is for broken device mode

Building gpsd 3.24 from source and installing corrects the issue.  prior version on Buster worked correctly as well

Can the distro package for gpsd be updated?

XECDesign, 2022-07-06:
This is a bit tricky since there are ABI changes in the new version. So any software built against the current version of libgps may not work with the new version. There are quite a few packages in the repo which this would apply to. Luckily, with the way the packaging has been done, it looks like this might not be a major issue in this case.

I've uploaded a test build to the ['untested' component](http://archive.raspberrypi.org/debian/pool/untested/g/gpsd/).

XECDesign, 2022-08-31:
No feedback, removing.

closed 2022-08-31 by XECDesign

TheSkorm, 2025-04-05:
Just for future reference this appears to be an issue with gpsd 3.22. Running gpsd in docker in bookworm (3.22) doesn't work, while gpsd in docker in trixie (3.25) works fine. It's discussed briefly here : https://marc.info/?t=165658652400002&r=1&w=2