RPi-Distro/repo #306: Raspberry Pi Screen Won't Wake Up

tonymartino, 2022-08-22:
Please see this thread:
https://forums.raspberrypi.com/viewtopic.php?t=328904
People have been experiencing this issue since the bulls4eye release. I just installed, waited for the screen to turn off, and could not turn it on by using mouse or keyboard. After rebooting, I updated the OS, rebooted, and again waited for the screen to turn off. It still could not be turned on by using the keyboard or mouse. This is a Raspberry Pi 3 Model B Plus Rev 1.3 but the problem has been reported on other models.

XECDesign, 2022-08-23:
Are you able to ssh in and see if dmesg says anything interesting?

ghollingworth, 2022-08-23:
@spl237 @popcornmix @6by9 @pelwell 

Sounds like it only affects Pi 3B+ and before, likely to be something FKMS related...

XECDesign, 2022-08-23:
Aren't we using KMS on all models now?

Also might be relevant:
https://github.com/RPi-Distro/repo/issues/295
https://github.com/raspberrypi/linux/issues/4962

I recall seeing something about switching to openbox rather than mutter potentially fixing it, but I'm not sure which one Pi 3B+ used by default.

popcornmix, 2022-08-23:
> Are you able to ssh in and see if dmesg says anything interesting?

do this and post the output (to a paste type site).
If it contains "flip_done timed out" then it's probably a variation of https://github.com/raspberrypi/linux/issues/4962 and is a kernel issue.

I'd also be interested if you can provoke the same hang by manually switching display to standby and back, which I'd expect to be the case if it is the linked issue.

tonymartino, 2022-08-23:
> > Are you able to ssh in and see if dmesg says anything interesting?
> 
> do this and post the output (to a paste type site). If it contains "flip_done timed out" then it's probably a variation of [raspberrypi/linux#4962](https://github.com/raspberrypi/linux/issues/4962) and is a kernel issue.
> 
> I'd also be interested if you can provoke the same hang by manually switching display to standby and back, which I'd expect to be the case if it is the linked issue.

I did see that yesterday. I'll grab the output shortly...

tonymartino, 2022-08-23:
dmesg output is [here](https://pastebin.com/xUjCEhUf). I tried to wake it up at 990.307740 and you can see the flip_done messages after that.

popcornmix, 2022-08-23:
Can you test if adding `video=HDMI-A-1:D` to end of `/boot/cmdline.txt` (on same line) works around the issue?

tonymartino, 2022-08-23:
Tried that. Same results:
`[ 1015.671000] logitech-hidpp-device 0003:046D:4024.0004: HID++ 2.0 device connected.
[ 1025.768487] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:89:crtc-3] flip_done timed out
[ 1036.008611] [drm:drm_crtc_commit_wait [drm]] *ERROR* flip_done timed out
[ 1036.009074] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CRTC:89:crtc-3] commit wait timed out
[ 1046.248672] [drm:drm_crtc_commit_wait [drm]] *ERROR* flip_done timed out
[ 1046.249131] [drm:drm_atomic_helper_wait_for_dependencies [drm_kms_helper]] *ERROR* [CONNECTOR:32:HDMI-A-1] commit wait timed out
[ 1056.488756] [drm:drm_crtc_commit_wait [drm]] *ERROR* flip_done timed out
[ 1056.489019] vc4-drm soc:gpu: [drm] *ERROR* Timed out waiting for commit
[ 1066.728864] [drm:drm_atomic_helper_wait_for_flip_done [drm_kms_helper]] *ERROR* [CRTC:89:crtc-3] flip_done timed out
`

tonymartino, 2022-08-29:
Fixed! Thank you.

closed 2022-08-31 by XECDesign