RPi-Distro/repo #307: Problem with installing using Orca as a blind user

krperry, 2022-08-29:
Currently when installing as a blind user we must first create a wpa_supplicant.conf  to be able to install when using Orca.  If the RPI is not connected to WI-FI the CTRL-ALT-Space prompt is not spoken. and there is no way to turn Orca on. 

I assume this is do to the installer using atp to install Orca once a person presses ctrl-alt-space.  

This makes installing Raspbion more difficult for a blind user than for a sighted user.  A sighted user does not have to ccreate a wpa_supplicant.conf to install for the first time.  They can but they do not have to. 

Orca should be already installed by default in the two desktop images and all CTRL-alt-space should do is turn it on or off for any user.

The expected behavior whether or not the RPI is connected to the internet is when installing it should ask to run Orca.

XECDesign, 2022-08-30:
@spl237 Any way around this? Is the orca install particularly large?

We could have it available in the apt package cache, so when the software tries to install it, it doesn't need to be downloaded, but also doesn't eat up as much space as if it was fully pre-installed.

XECDesign, 2022-08-30:
Yes, those options have been discussed internally. We'll look for a suitable solution.

krperry, 2022-10-11:
Thing.  I just ran apt show on orca on the rpi.  It said 1800k  for the install package.  I have no idea what that means over all for size.  I am betting it is not even close to 5 MB.

XECDesign, 2022-10-12:
The cost of pre-installing orca and its dependencies on the standard armhf image is 60.5 MB. I've also been told that enabling/disabling or uninstalling and re-installing orca doesn't reliably get you back to where you were. That makes pre-installing it a decision that's not quite as simple as it might seem.

When we had a quick look at pre-caching the necessary packages we found that it doesn't work with the way we're currently using packagekit.

I still think we can make it work, but we'll need to set some time aside to take a proper look.

krperry, 2022-10-12:
If not on the smaller images to start with it at least should be installed  by default on the Full desktop image.    65 Mb wouldn’t even dent that beast.  Then there would be no need for WI-FI for that image.    I also wonder why it can not be made as a config option in the config.txt.    Which could fix a few problems.  Part of the problem with the PI install and blind users is trying to get audio set so that it comes out the 3.5MM jack rather than HDMI because most of us don’t even hook it up to a monitor.   

 

As I said before it might be easier to make a version of the regular images that have Orca installed by default and that just come up working.  It would look good on the PI folks to have accessible  images.  In fact there could be a couple one for Blind (Orca) and one setup in Low contrast mode).  Would that really be so bad?  I mean we already have images for playing music etc.   I can tell you schools are being pushed to have stuff  accessible more and more  and  this is something that is used in schools.

spl237, 2022-10-12:
> If not on the smaller images to start with it at least should be installed by default on the Full desktop image.

As above - it is not as simple as merely being an install size issue. Orca is not particularly well-behaved in terms of resource usage, and there is a question as to whether it can cause performance degradation if installed but not enabled. For this reason, I am not happy about preinstalling it.

>It would look good on the PI folks to have accessible images. In fact there could be a couple one for Blind (Orca) and one setup in Low contrast mode). Would that really be so bad? I mean we already have images for playing music etc. 

Every extra image we have to produce has to be tested on every release. While creating multiple images is relatively trivial, testing them is time-consuming, and the effect of adding a new configuration is to multiply the number of images, not just to add to them - if we added an Orca image and a low-contrast image, we would need to create them as both 32-bit and 64-bit versions; we would need to produce a screen-reader compatible lite image, and so on. 

Given the number of people who actually use accessibility under Linux - which is really not huge, due to it not actually working all that well; accessibility is far better developed on Windows and Mac - this is more overhead than we are going to be able to handle, I'm afraid. 

We make it as easy as possible to install Orca - a single keystroke combination - and if there is anything we can do to make this even easier, we will investigate that, but we are not going to produce dedicated accessible images; the demand simply does not justify the effort required.

krperry, 2022-10-12:
It is true Orca can be a problem if you run it and just silence it but if you do not run it It affects your OS no more than not using LS.   

 

Second saying that not many Blind people use Linux is both right and wrong.   Few use it in the USA as a desktop but many around the world use it because it is low cost.  Further more Schools use it for courses and  those need  to be accessible.  I have been using Linux since a year after I went blind in 1991. Started with Slackware 0.9.  I work with blind developers around the world that use  it.  We are working on creating electronics classes and programming classes to run on the RPI  and right now we are creating our own Images.  If you would like I could have a survey out just to show you many people really are using these devices that are low vision and blind or going blind.  I think your discounting something that you do not have the data for.    The company I work for is one of the largest companies that create services for the blind and low vision in the world not just the USA and while there may not be as many Blind users of Linux as there are sighted the percentage that are Blind computer users and that use Linux are a lot better than I think you know and are growing as the prices of things like tactile displays come down and need a good OS to support them. The Canute 9 line braille display has at its heart a RPI.  The more blind developers we can get on these devices the more devices can be created to support blind users.  WE have a if you build it they will come problem right now because for a sighted users it takes 10 minutes to get up and running.  For a blind low vision user it takes much more knowledge and much more time.