RPi-Distro/repo #308: libraspberrypi-bin package is not included in the Raspbian image for x*

crokett, 2022-08-30:


crokett, 2022-08-30:
This is the listing of the packages for Raspbian on AMD64. http://[archive.raspberrypi.org/debian/dists/buster/main/binary-amd64/Packages](http://archive.raspberrypi.org/debian/dists/buster/main/binary-amd64/Packages)

It has one package that refers to libraspberrypi-bin, but doesn't have the actual libraspberrypi-bin.    I am looking for that package built for AMD64 or x86_64

XECDesign, 2022-08-30:
Those are generally only useful when running on an actual pi. Is there a specific tool you're after?

crokett, 2022-08-30:
> Those are generally only useful when running on an actual pi. Is there a specific tool you're after?

Yes.  I was looking for vcgencmd, but I figured out that it won't run on other boards. I will close this issue.

closed 2022-08-30 by crokett