RPi-Distro/repo #309: Recent RaspianOS image does not boot on CM4 from NVMe

framps, 2022-09-15:
I just upgraded my RaspberyOS Bulsseye on my CM4 and finally wasn't able to boot the system any more :-(. 

Then I dd'd the image on eMMc of the Cm4 averything works fine. When I dd'd the image on NMVe the system doesn't boot.

The image I used is 2022-09-06-raspios-bullseye-armhf.img`.

When I restore a backup I have created some time ago the system boots from NVMe.

XECDesign, 2022-09-16:
@timg236 is this something you might know about?

timg236, 2022-09-16:
Not really, but flashing latest RPi OS 32-bit on a CM4 NVMe works for me

framps, 2022-09-16:
> works for me

Thank you very much for your reply. Looks like there is something wrong on my CM4. How can I debug the issue?

peterharperuk, 2022-09-16:
2022-09-06-raspios-bullseye-armhf-lite.img works for me, as does upgrading from 2022-04-04-raspios-bullseye-armhf-lite.img to 2022-09-06-raspios-bullseye-armhf-lite.img.

Can you give more details of your setup? Are you using a Raspberry PI compute module? What's the make and model of your NVMe drive? Do you see anything on the screen when it fails?

You could edit /boot/cmdline.txt to remove the words "quiet splash". This will show the kernel booting on the screen if it's getting that far. If you have a UART cable I can tell you how to get logs from the bootloader and firmware.

For the avoidance of doubt you can run the following commands to give us some debug info...

rpi-eeprom-config
cat /etc/rpi-issue
vcgencmd bootloader_versionv
vcgencmd version
uname -a

sudo apt-get install nvme-cli
sudo nvme list
sudo nvme id-ctrl -H /dev/nvme0
sudo nvme list-ns /dev/nvme0
sudo nvme id-ns -H /dev/nvme0 --namespace-id=1

timg236, 2022-09-16:
Adding enable_uart=1  and uart_2ndstage=1 to config.txt will also provide more information
https://www.raspberrypi.com/documentation/computers/config_txt.html#uart_2ndstage
https://www.raspberrypi.com/documentation/computers/config_txt.html#enable_uart

framps, 2022-09-16:
> Can you give more details of your setup?

It's a CM4 with 1GB RAM, 32 GB eMMC and 128 GB NVMe assembled on a Waveshare CM4-IO-BASE-A (See [here](https://linux-tips-and-tricks.de/images/stories/CM4Parts.jpg) for a picture of all components). 

> What's the make and model of your NVMe drive?

It's a hynix. See the linked picture and the `nvme` command results below.

> Do you see anything on the screen when it fails?

Nothing. Green LEDs M.2 and ACT flash but the red boot LED is dark. Looks like it's just not booting.

So that's what I did: I copied the 09-06 image on eMMC and started the CM4 that way in order to be able to execute the following commands:

```
pi@cm4-bullseye:~ $ rpi-eeprom-config 
[all]
BOOT_UART=0
WAKE_ON_GPIO=1
POWER_OFF_ON_HALT=0

# Boot Order Codes, from https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#BOOT_ORDER
# Try SD first (1), followed by, USB PCIe, NVMe PCIe, USB SoC XHCI then network
BOOT_ORDER=0xf25641
# Try NVMe first (6), followed by, USB PCIe, SD , USB SoC XHCI then network
#BOOT_ORDER=0xf25146

# Set to 0 to prevent bootloader updates from USB/Network boot
# For remote units EEPROM hardware write protection should be used.
ENABLE_SELF_UPDATE=1
```
Note: I use the second BOOT_ORDER when I want to boot from NVMe.
```
i@cm4-bullseye:~ $ cat /etc/rpi-issue 
Raspberry Pi reference 2022-09-06
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 66255495f29be5d09b765d081aff6fc0f11e59b4, stage2
```
```
i@cm4-bullseye:~ $ vcgencmd bootloader_version
2022/08/02 16:55:05
version 91b6280c53c0b6dff9e3d58810f439e577408845 (release)
timestamp 1659455705
update-time 1663329705
capabilities 0x0000007f
```
```
pi@cm4-bullseye:~ $ vcgencmd version
Aug 26 2022 14:03:33 
Copyright (c) 2012 Broadcom
version 102f1e848393c2112206fadffaaf86db04e98326 (clean) (release) (start_cd)
```
```
pi@cm4-bullseye:~ $ uname -a
Linux cm4-bullseye 5.15.61-v7l+ #1579 SMP Fri Aug 26 11:13:03 BST 2022 armv7l GNU/Linux
```
```
pi@cm4-bullseye:~ $ sudo nvme list
Node             SN                   Model                                    Namespace Usage                      Format           FW Rev  
---------------- -------------------- ---------------------------------------- --------- -------------------------- ---------------- --------
/dev/nvme0n1        FNABN43081010B54Z BC711 NVMe SK hynix 128GB                1         128.04  GB / 128.04  GB    512   B +  0 B   41002131
pi@cm4-bullseye:~ $ sudo nvme id-ctrl -H /dev/nvme0
NVME Identify Controller:
vid       : 0x1c5c
ssvid     : 0x1c5c
sn        :    FNABN43081010B54Z
mn        : BC711 NVMe SK hynix 128GB               
fr        : 41002131
rab       : 3
ieee      : ace42e
cmic      : 0
[... 229 lines cut ...]
dlfeat  : 0
  [4:4] : 0	Guard Field of Deallocated Logical Blocks is set to 0xFFFF
  [3:3] : 0	Deallocate Bit in the Write Zeroes Command is Not Supported
  [2:0] : 0	Bytes Read From a Deallocated Logical Block and its Metadata are Not Reported

nawun   : 0
nawupf  : 0
nacwu   : 0
nabsn   : 0
nabo    : 0
nabspf  : 0
noiob   : 0
nvmcap  : 0
nsattr	: 0
nvmsetid: 0
anagrpid: 0
endgid  : 1
nguid   : 00000000000000000000000000000000
eui64   : ffffffffffffffff
LBA Format  0 : Metadata Size: 0   bytes - Data Size: 512 bytes - Relative Performance: 0 Best (in use)
LBA Format  1 : Metadata Size: 0   bytes - Data Size: 4096 bytes - Relative Performance: 0 Best 
```
> Adding enable_uart=1 and uart_2ndstage=1 to config.txt will also provide more information

I never used the serial to see messages from the CM4. Let me know if this information is important for you and I will connect my USB2TTL adapter to the CM4.

peterharperuk, 2022-09-16:
It sounds like it's failing in the bootloader or firmware, in which case the uart output would be helpful. If you add enable_uart=1 and uart_2ndstage=1 to config.txt it'll tell us if it's getting as far as firmware.

If you can edit your eeprom config then setting BOOT_UART=1 will tell us if it fails in the bootloader.

peterharperuk, 2022-09-16:
Ok. I've ordered a drive like yours (seemed suspiciously cheap) and will let you know if it works for me.

timg236, 2022-09-16:
If you add uart_2ndstage=1 then you should get some more output from start4.elf

framps, 2022-09-16:
It's already in there:

```
cat /media/framp/boot1/cmdline.txt
console=serial0,115200 console=tty1 root=PARTUUID=be6c5e24-02 rootfstype=ext4 fsck.repair=yes rootwait enable_uart=1 uart_2ndstage=1
```

framps, 2022-09-16:
That's in config.txt:

```
cat /media/framp/boot1/config.txt 
# For more options and information see
# http://rpf.io/configtxt
# Some settings may impact device functionality. See link above for details

# uncomment if you get no picture on HDMI for a default "safe" mode
#hdmi_safe=1

# uncomment the following to adjust overscan. Use positive numbers if console
# goes off screen, and negative if there is too much border
#overscan_left=16
#overscan_right=16
#overscan_top=16
#overscan_bottom=16

# uncomment to force a console size. By default it will be display's size minus
# overscan.
#framebuffer_width=1280
#framebuffer_height=720

# uncomment if hdmi display is not detected and composite is being output
#hdmi_force_hotplug=1

# uncomment to force a specific HDMI mode (this will force VGA)
#hdmi_group=1
#hdmi_mode=1

# uncomment to force a HDMI mode rather than DVI. This can make audio work in
# DMT (computer monitor) modes
#hdmi_drive=2

# uncomment to increase signal to HDMI, if you have interference, blanking, or
# no display
#config_hdmi_boost=4

# uncomment for composite PAL
#sdtv_mode=2

#uncomment to overclock the arm. 700 MHz is the default.
#arm_freq=800

# Uncomment some or all of these to enable the optional hardware interfaces
#dtparam=i2c_arm=on
#dtparam=i2s=on
#dtparam=spi=on

# Uncomment this to enable infrared communication.
#dtoverlay=gpio-ir,gpio_pin=17
#dtoverlay=gpio-ir-tx,gpio_pin=18

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Automatically load overlays for detected cameras
camera_auto_detect=1

# Automatically load overlays for detected DSI displays
display_auto_detect=1

# Enable DRM VC4 V3D driver
dtoverlay=vc4-kms-v3d
max_framebuffers=2

# Disable compensation for displays with overscan
disable_overscan=1

[cm4]
# Enable host mode on the 2711 built-in XHCI USB controller.
# This line should be removed if the legacy DWC2 controller is required
# (e.g. for USB device mode) or if USB support is not required.
otg_mode=1

[all]

[pi4]
# Run as fast as firmware / board allows
arm_boost=1

[all]
gpu_mem=16
enable_uart=1
```

peterharperuk, 2022-09-16:
Sorry - uart_2ndstage=1 has to go in config.txt not cmdline.txt

framps, 2022-09-16:
Oops - sorry.

But I still don't get any other debug lines. 

cmdline.txt
```
console=serial0,115200 console=tty1 root=PARTUUID=7a1b13df-02 rootfstype=ext4 fsck.repair=yes rootwait
```

At the end of config.txt I have
```
[all]
gpu_mem=16
enable_uart=1
uart_2ndstage=1
```

framps, 2022-09-16:
I tried the 64 bit lite image and didn't terminate the dangling boot. All of the sudden I got some additional output (see the attached file).

[nvme_boot.txt](https://github.com/RPi-Distro/repo/files/9586711/nvme_boot.txt)

The last line is very interesting:

`[    3.072075] Waiting for root device PARTUUID=be6c5e24-02...`

```
sudo blkid 
...
/dev/sde1: LABEL_FATBOOT="boot" LABEL="boot" UUID="A07B-BB23" TYPE="vfat" PARTUUID="be6c5e24-01"
/dev/sde2: LABEL="rootfs" UUID="93282bcd-0b56-4477-aed5-dfb0038f9ca8" TYPE="ext4" PARTUUID="be6c5e24-02"
/dev/sdf1: LABEL_FATBOOT="boot" LABEL="boot" UUID="A07B-BB23" TYPE="vfat" PARTUUID="ac1488a6-01"
/dev/sdf2: LABEL="rootfs" UUID="93282bcd-0b56-4477-aed5-dfb0038f9ca8" TYPE="ext4" PARTUUID="ac1488a6-02"
```
```
sudo fdisk -l /dev/sde
Disk /dev/sde: 29.12 GiB, 31268536320 bytes, 61071360 sectors
Disk model:                 
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xbe6c5e24

Device     Boot  Start      End  Sectors  Size Id Type
/dev/sde1         8192   532479   524288  256M  c W95 FAT32 (LBA)
/dev/sde2       532480 61071359 60538880 28.9G 83 Linux
```
```
sudo fdisk -l /dev/sdf
Disk /dev/sdf: 119.25 GiB, 128035676160 bytes, 250069680 sectors
Disk model:                 
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: dos
Disk identifier: 0xac1488a6

Device     Boot  Start     End Sectors  Size Id Type
/dev/sdf1         8192  532479  524288  256M  c W95 FAT32 (LBA)
/dev/sdf2       532480 3661823 3129344  1.5G 83 Linux
```

My eMMC has 32GB and the NVMe has 128GB. This means 
1) Boot takes a long time now (approx 30 seconds). Usually it boots after some seconds.
2) At the end of the boot process the eMMC root partition will be looked up as rootfs even the NVMe image boots.

I checked /boot/cmdline,txt and /etc/fstab on the NVMe image and both have the correct UUID of the NVMe partition (ac1488a6-02).

Not sure whether this helps to locate the root cause for the issue. It's also strange I don't get any additional output even I have 

```
gpu_mem=16
enable_uart=1
uart_2ndstage=1
```
at the end of config.txt.

framps, 2022-09-17:
Just to proof my steps I execute in order to get an uart output with 2nd stage output are correct I executed following steps:

1) Restored the 04-04 lite image on eMMC and added the two uart lines in /boot/config.txt. I got 2ndstage output.
2) Restored the 09-06 lite image on eMMC and added the two uart lines in /boot/config.txt. I got again 2ndstage output. 
3) Restored the 04-04 lite image on NVMe and added the two uart lines in /boot/config.txt. Next I changed the BOOT_ODER to boot from NVMe. I got 2ndstage output. 
4) Restored the 09-06 lite image on NVMe and added the two uart lines in /boot/config.txt. BOOT_ORDER change was not required. I don't get any 2ndstage output :-( . The system just hangs. For me it sounds there is some issue in the startup phase of 2ndstage before any debug statements are executed or any files required for 2nd stage are not found and boot hangs somewhere. 
I also executed the same steps (4) and updated the eeprom with the boot order again: no 2ndstage output. 

I also noticed the system doesn't look for the eMMC rootfs any more after some time what I described in my previous comment. 

Please let me know what I should do next in order to help you to locate the root cause for the issue.

framps, 2022-09-17:
Maybe the boot messages from the 04-04 image which include the stage2 messages help?

[nvme_boot.txt](https://github.com/RPi-Distro/repo/files/9592513/nvme_boot.txt)

peterharperuk, 2022-09-26:
I ordered this drive and it works fine. It arrived with a slightly different model number BC501A despite me ordering BC711. 

Your logs indicate that the bootloader and firmware worked okay. The kernel seems to be stuck waiting for the drive to appear. It's possible the bootloader or firmware are causing issues for the kernel.

Things you could try if you're up for it.
1) A clean install - you will lose any data on the drive however.
2) Boot using the SD (BOOTORDER 1) but set the root parameter in cmdline.txt to point to the NVMe drive (root=PARTUUID=be6c5e24-02). If it still doesn't work then it implies a kernel issue with that drive?

framps, 2022-09-26:
> I ordered this drive and it works fine. It arrived with a slightly different model number BC501A despite me ordering BC711.

Thank you very much you tried to recreate my issue.

> The kernel seems to be stuck waiting for the drive to appear.

It's not a busy loop. The heat sink stays warm as usual. Looks like the kernel waits for an interrupt.

> 2\. set the root parameter in cmdline.txt to point to the NVMe drive

Clean install will be done on NVMe. Should I then change cmdline.txt on the NVMe boot partition or eMMC boot partition? Just guessing eMMC boot partition may be used with `BOOTORDER 1`and thus cmdline.txt of eMMC boot partition will be used.

I'm still busy right now but I will follow your instructions mid next week.

framps, 2022-09-29:
I'm back home now.

That's what I did:

1) Prime the emmc storage with the 09 lite image and set BOOTORDER to 1 to boot from emmc
2) Booted the CM4. Got messages a ssh key is generated. then the system rebooted and got a second time a message a ssh key is generated. Next I was prompted for a user and password. Then I was prompted for the keyboard I changed to a German keyboard. 
3) Next I got a login prompt
4) I logged in an verified the system booted from emmc

5) Primed the nvme with the 09 lite image and kept the boot order.
6) Tried to boot - failure
7) Modified cmdline.txt on the emmc storage to use the nvme partuuid
8) Tried to reboot - success

So that's what I actually asked for in my previous comment: When I change the BOOTORDER to 1 (SD) the emmc storage will be used and thus the cmdline.txt on this storage has to use the nvme root filesystem.

The interesting part is that the system now uses /boot from the nvme storage

```
pi@raspberrypi:~ $ mount | grep nvme
/dev/nvme0n1p2 on / type ext4 (rw,noatime)
/dev/nvme0n1p1 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
```

But if I change the BOOTORDER back to 6 now system doesn't boot any more.

I have no clue what's going on there but it seems to me the discovery of the boot partition on my nvme storage fails. If I use the boot partition of the emmc storage which can be discovered and use the nvme root partition it works.

It's also interesting the ssh key is generated twice. First on the emmc root partition and second on the nvme root partition? 

> 2\. If it still doesn't work then it implies a kernel issue with that drive?

It works - but with a mixture of emmc and nvme partitions. That's the way it's possible to use a USB disk with an older Raspberry which is not USB boot capable: Just insert a SD card with /boot and modify cmdline.txt to use the USB rootpartition. 

But that's not the way it should work with nvme. Did you execute your test with a CM4 which has emmc and nvme or with a CM4 which has no emmc but nvme?

peterharperuk, 2022-09-30:
I probably used a cm4 lite. If I send you a test bootloader/firmware, are you able test something for me? I don't really know why switching release would make a difference, but we have recently spotted a timing issue ultimately caused by us not disabling nvme after we've finished with it.

framps, 2022-09-30:
> I probably used a cm4 lite.

Ok. So my issue may not be an issue with my hynix nvme card but the fact I have emmc and nvme storage. Maybe somebody can test this configuration to verify it's not an issue of a cm4 equipped with eMMC and NVMe?

> If I send you a test bootloader/firmware, are you able test something for me?

Sure. Just give me a download URL and instructions how to flash this code on my cm4. Please also let me know how I can revert and go back to the current bootloader/firmware - just in case ... Should I save the current code first? If yes, I need instructions how to do this.

peterharperuk, 2022-09-30:
[pieeprom.original.tipwithfix.zip](https://github.com/RPi-Distro/repo/files/9683888/pieeprom.original.tipwithfix.zip)

The attached has a bootloader and firmware. Copy start4.elf and fixup4.dat to the /boot partition on the nvme drive. Backup the old files. Update the eeprom with the pieeeprom* image.

programming the bootloader...

To program the bootloader you need to use usbboot git@github.com:raspberrypi/usbboot.git. Fetch this e.g. on a rpi4 and build the code by running make. Copy the new bootloader into the usbboot/recovery folder. You have to replace the file pieeprom.original.bin. Sign the bootloader and run rpiboot, like this...

cd usbboot/recovery
./update-pieeprom.sh
../rpiboot -d .

Connect the rpi4 to the USB slave and disable eMMC boot with a jumper (the Waveshare board has a switch for this and uses usb c cable connected to its power port) . The cm4 will try booting via rpiboot, it should connect to the rpi4 and program the eeprom with the new bootloader. The screen goes green if this was successful. Re-enable eMMC boot by removing the jumper and remove the USB cable. It should then boot with the new bootloader after a powercycle.

You should see the following text on boot. I added the "pete" thing so you can confirm that you've updated your bootloader correctly. You can set your boot order back to 6 and set BOOT_UART=0

pete: f80b151a
0x00c03140 0x00000000 0x00001fff
MEM GPU: 76 ARM: 948 TOTAL: 1024
Starting start4.elf @ 0xfec00200 partition 0
NVME off

framps, 2022-09-30:
That's what I did:

1) Primed the nvme storage with the 09 lite image
2) Saved old start4.elf and copied your start4.elf into /boot
3) Saved fixup4.dat and copied your fixup4.dat into /boot
4) Saved old pieeprom.original.bin and copied pieeprom.original.tipwithfix.bin into /usbboot/recovery as pieeprom.original.bin
5) Executed on my Linux desktop
 ./update-pieeprom.sh
 ../rpiboot -d .
6) Rebooted cm4 with bootorder 1. Boot hangs. See attached UART log
[nvme_bootorder_1.log](https://github.com/RPi-Distro/repo/files/9685974/nvme_bootorder_1.log)

7) Changed bootorder to 6
8) Rebooted cm4 with bootorder 6. Boot hangs. See attached UART log
[nvme_bootorder_6.log](https://github.com/RPi-Distro/repo/files/9685975/nvme_bootorder_6.log)

As you can see in the logs your bootloader `pete` is used. :smiley: 

The second log shows `NVME off` but this doesn't fix the issue. It even doesn't boot with bootmode 1 from eMMC :cry: 

> 0x00c03140 0x00000000 0x00001fff

I noticed this line is different than yours `0x00a03140 0x00000000 0x00001fff`. 

Anything else I can do for you to locate the root cause of the issue?

framps, 2022-09-30:
I just wanted to revert all changes and detected I updated start4.elf and fixup4.dat on the emmc boot partition :worried:

Then I copied these files on the nvme boot partition and rebooted cm4 again (I still have bootorder 6). 

cm4 booted, generated ssh key once - not twice what happens without the fix - and asked for keyboard layout and uid and pwd of user. 

Success :smile: You fix fixed the issue :+1: 

Do I have to wait for an official fix now or can I continue and use this patched cm4 and any future updates will replace the patched code?

peterharperuk, 2022-09-30:
Future updates should replace those files.

framps, 2022-09-30:
Great. 

I just primed both emmc and nvme storage with the 09 image and updated start4.elf and fixup4.dat on both boot partitions in order to make sure there is no regression with emmc boot. 

Everything works fine now.

But I noticed during first boot from emmc the ssh key was generated twice. And when I booted from nvme there was no ssh key generated any more. Looks like the ssh generation is executed on all available rootfs during first boot. Not sure whether this works as designed ...

@leidrin You may be interested in this issue :wink:

framps, 2022-09-30:
I just used raspi-config and configured

1) 16 GPU size
2) Changed hostname to `raspberrypi-buster-nvme`
3) Add EN-US language
4) Set time to Berlin
5) Resized nvme partition to use whole space

The system reboots and hangs now again

framps, 2022-09-30:
I nailed the boot issue down to the GPU memory size reduction. With the previous image I was able to reduce the GPU memory.

Not sure whether this is related to the nvme boot issue or whether it's another issue in the 09 image.

peterharperuk, 2022-09-30:
Why are you messing with GPU memory?

framps, 2022-09-30:
I run all my Raspberries headless and just need the minimal GPU memory. If it's no longer supported to lower the GPU memory to 16 MB with 09 image and more memory is required with the 09 image this has to be documented and changed in raspi-config accordingly.

framps, 2022-09-30:
I primed eMMC and NMVe with the same 09 image and your fix. Note I can change GPU memory to 16MB on eMMC without any boot issue.

timg236, 2022-10-01:
If gpu_mem in config.txt is < 32 then startcd4.elf is loaded instead of start4.elf so you will be seeing the old firmware release.

framps, 2022-10-01:
Sounds reasonable. Please provide a fixed version of startcd4.elf and I will verify the fix :wink:

peterharperuk, 2022-10-06:
If you run rpi-update you'll get the fix.

framps, 2022-10-06:
Thank you very much for the update.

That's what I did now to verify the fix:

1) Flashed 0404 image on nvme storage
2) Booted system
3) Executed sudo rpi-update
4) Kept the default GPU memory
5) Rebooted the CM4
No boot. System hangs :disappointed: 

I checked the UART log and detected there are still the debug messages `pete` and `NVME off`. 

```
RPi: BOOTLOADER release VERSION:3165691d DATE: 2022/09/29 TIME: 15:56:43 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1664463403 0x34a87b72 0x00a03140 0x000747a9
PM_RSTS: 0x00001000
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Samsung' 8Gb x1 total-size: 8 Gbit 3200
DDR 3200 0 0 8 152

Boot mode: NVME (06) order f2514
VID 0x1c5c MN BC711 NVMe SK hynix 128GB              
NVME on
MBR: 0x00002000,  524288 type: 0x0c
MBR: 0x00082000, 3129344 type: 0x83
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
Trying partition: 0
type: 32 lba: 8192 oem: 'mkfs.fat' volume: ' boot       '
rsc 32 fat-sectors 1020 c-count 130554 c-size 4
root dir cluster 2 sectors 0 entries 0
FAT32 clusters 130554
Trying partition: 0
type: 32 lba: 8192 oem: 'mkfs.fat' volume: ' boot       '
rsc 32 fat-sectors 1020 c-count 130554 c-size 4
root dir cluster 2 sectors 0 entries 0
FAT32 clusters 130554
Read config.txt bytes     2075 hnd 0x163
Read start4.elf bytes  2249280 hnd 0x3cdb
Read fixup4.dat bytes     5399 hnd 0x169
pete: f80b151a
0x00a03140 0x00000000 0x00001fff
MEM GPU: 76 ARM: 948 TOTAL: 1024
Firmware: 102f1e848393c2112206fadffaaf86db04e98326 Aug 26 2022 14:03:16
Starting start4.elf @ 0xfec00200 partition 0
NVME off
+
```

Not sure whether the following output helps which I created when the rpi-update finished:

```
root@raspberrypi:/boot# ls -la
total 49851
drwxr-xr-x  3 root root    4096 Jan  1  1970 .
drwxr-xr-x 19 root root    4096 Oct  6 16:49 ..
-rwxr-xr-x  1 root root      41 Oct  6 16:51 .firmware_revision
-rwxr-xr-x  1 root root   18693 Mar 31  2022 COPYING.linux
-rwxr-xr-x  1 root root    1594 Mar 31  2022 LICENCE.broadcom
-rwxr-xr-x  1 root root   28418 Oct  6 16:50 bcm2708-rpi-b-plus.dtb
-rwxr-xr-x  1 root root   27790 Oct  6 16:50 bcm2708-rpi-b-rev1.dtb
-rwxr-xr-x  1 root root   28111 Oct  6 16:50 bcm2708-rpi-b.dtb
-rwxr-xr-x  1 root root   28032 Oct  6 16:50 bcm2708-rpi-cm.dtb
-rwxr-xr-x  1 root root   29267 Oct  6 16:50 bcm2708-rpi-zero-w.dtb
-rwxr-xr-x  1 root root   27856 Oct  6 16:50 bcm2708-rpi-zero.dtb
-rwxr-xr-x  1 root root   29305 Oct  6 16:50 bcm2709-rpi-2-b.dtb
-rwxr-xr-x  1 root root   29304 Oct  6 16:50 bcm2709-rpi-cm2.dtb
-rwxr-xr-x  1 root root   30170 Oct  6 16:50 bcm2710-rpi-2-b.dtb
-rwxr-xr-x  1 root root   32533 Oct  6 16:50 bcm2710-rpi-3-b-plus.dtb
-rwxr-xr-x  1 root root   31922 Oct  6 16:50 bcm2710-rpi-3-b.dtb
-rwxr-xr-x  1 root root   30157 Oct  6 16:50 bcm2710-rpi-cm3.dtb
-rwxr-xr-x  1 root root   31230 Oct  6 16:50 bcm2710-rpi-zero-2-w.dtb
-rwxr-xr-x  1 root root   31230 Oct  6 16:50 bcm2710-rpi-zero-2.dtb
-rwxr-xr-x  1 root root   52052 Oct  6 16:50 bcm2711-rpi-4-b.dtb
-rwxr-xr-x  1 root root   52184 Oct  6 16:50 bcm2711-rpi-400.dtb
-rwxr-xr-x  1 root root   52637 Oct  6 16:50 bcm2711-rpi-cm4.dtb
-rwxr-xr-x  1 root root   49951 Oct  6 16:50 bcm2711-rpi-cm4s.dtb
-rwxr-xr-x  1 root root   52460 Oct  6 16:50 bootcode.bin
-rwxr-xr-x  1 root root     103 Jan  1  1980 cmdline.txt
-rwxr-xr-x  1 root root    2075 Apr  4  2022 config.txt
-rwxr-xr-x  1 root root    7262 Oct  6 16:50 fixup.dat
-rwxr-xr-x  1 root root    5395 Oct  6 16:50 fixup4.dat
-rwxr-xr-x  1 root root    3170 Oct  6 16:50 fixup4cd.dat
-rwxr-xr-x  1 root root    8379 Oct  6 16:50 fixup4db.dat
-rwxr-xr-x  1 root root    8379 Oct  6 16:50 fixup4x.dat
-rwxr-xr-x  1 root root    3170 Oct  6 16:50 fixup_cd.dat
-rwxr-xr-x  1 root root   10226 Oct  6 16:50 fixup_db.dat
-rwxr-xr-x  1 root root   10224 Oct  6 16:50 fixup_x.dat
-rwxr-xr-x  1 root root     145 Apr  4  2022 issue.txt
-rwxr-xr-x  1 root root 6269448 Oct  6 16:50 kernel.img
-rwxr-xr-x  1 root root 6630752 Oct  6 16:50 kernel7.img
-rwxr-xr-x  1 root root 7040648 Oct  6 16:50 kernel7l.img
-rwxr-xr-x  1 root root 8193235 Oct  6 16:50 kernel8.img
drwxr-xr-x  2 root root   24064 Oct  6 16:50 overlays
-rwxr-xr-x  1 root root 2974112 Oct  6 16:50 start.elf
-rwxr-xr-x  1 root root 2249856 Oct  6 16:50 start4.elf
-rwxr-xr-x  1 root root  804380 Oct  6 16:50 start4cd.elf
-rwxr-xr-x  1 root root 3746024 Oct  6 16:50 start4db.elf
-rwxr-xr-x  1 root root 2997320 Oct  6 16:50 start4x.elf
-rwxr-xr-x  1 root root  804380 Oct  6 16:50 start_cd.elf
-rwxr-xr-x  1 root root 4817896 Oct  6 16:50 start_db.elf
-rwxr-xr-x  1 root root 3720968 Oct  6 16:50 start_x.elf
root@raspberrypi:/boot# cat .firmware_revision 
9bd03428a49f638b4a2c57015d0f7aea49cf16fa
```

peterharperuk, 2022-10-06:
This doesn't appear to match the version of start4.elf you have in your boot folder?
Read start4.elf bytes  2249280 hnd 0x3cdb
Firmware: 102f1e848393c2112206fadffaaf86db04e98326 Aug 26 2022 14:03:16

It should show...
Read start4.elf bytes  2249856 hnd 0x6c2d
Firmware: 6c57d97fa8066929a2fee6e4fac6d64ad62506c9 Oct  5 2022 15:51:11

peterharperuk, 2022-10-06:
you'll see the "pete" line until you update your bootloader. We have a beta in apt but we've found an issue with it, so you might want to hold on a bit.

framps, 2022-10-06:
> This doesn't appear to match the version of start4.elf you have in your boot folder?

Good catch :+1: 

I executed steps 1-3 again and checked UART log again. Now the filesize of start4.elf in /boot matches the UART boot log. I have no idea why it didn't match in my first attempt. 

```
RPi: BOOTLOADER release VERSION:3165691d DATE: 2022/09/29 TIME: 15:56:43 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1664463403 0x34a87b72 0x00a03140 0x00074127
PM_RSTS: 0x00000020
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Samsung' 8Gb x1 total-size: 8 Gbit 3200
DDR 3200 0 0 8 152

Boot mode: NVME (06) order f2514
VID 0x1c5c MN BC711 NVMe SK hynix 128GB              
NVME on
MBR: 0x00002000,  524288 type: 0x0c
MBR: 0x00082000,249537200 type: 0x83
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
Trying partition: 0
type: 32 lba: 8192 oem: 'mkfs.fat' volume: ' boot       '
rsc 32 fat-sectors 4033 c-count 516190 c-size 1
root dir cluster 2 sectors 0 entries 0
FAT32 clusters 516190
Trying partition: 0
type: 32 lba: 8192 oem: 'mkfs.fat' volume: ' boot       '
rsc 32 fat-sectors 4033 c-count 516190 c-size 1
root dir cluster 2 sectors 0 entries 0
FAT32 clusters 516190
Read config.txt bytes     2075 hnd 0x51b
Read start4.elf bytes  2249856 hnd 0x1a08d
Read fixup4.dat bytes     5395 hnd 0x232a9
pete: f80b151a
0x00a03140 0x00000000 0x00001fff
MEM GPU: 76 ARM: 948 TOTAL: 1024
Firmware: 6c57d97fa8066929a2fee6e4fac6d64ad62506c9 Oct  5 2022 15:51:11
Starting start4.elf @ 0xfec00200 partition 0
NVME off
+
```
Then I executed `sudo apt-get update; sudo apt-get upgrade` but didn't reboot the system. (I forgot to add this step in my list above and updated the list right now). Now there is an old start4.elf in /boot.

```
-rwxr-xr-x 1 root root 2249280 Oct  6 18:37 /boot/start4.elf
```
Looks like the apt upgrade reverted the update of start4.elf because now when I restart the CM4 it hangs.

> We have a beta in apt but we've found an issue with it, so you might want to hold on a bit.

Take your time :wink:

framps, 2022-10-06:
It's OT now @peterharperuk - but maybe you can help me to get access to my eMMC again. I just executed `sudo dd if=/dev/zero of=/dev/sdf bs=2GiB count=10` to my 32GB eMMC (maybe it was count=20). My command window with the dd command hung and I had to force close the window.
 
I thought I should clean the eMMC storage to make sure there is no code retrieved from there during boot in case of some storage lookup error. Now eMMC doesn't show up if I use mass-storage-gadget nor if I boot from nvme and check with `sudo blkid`.

Any help is greatly appreciated.

peterharperuk, 2022-10-06:
RPI Image has an option to Erase / format a card as Fat32. Try that.

framps, 2022-10-06:
Thank you very much for your reply on my OT question.

Please note: It's the internal eMMC, no SD card.

That's what I tried now: I downloaded a fresh usbboot git repo from the raspberry git, executed make and followed the README in rpi-imager-embedded. But the system boots from nvme :cry: . 

Given it's OT: Is there any place I can open an issue on this or can ask for help? A RaspberryForum thread or to open an issue in some git repo? I think we shouldn't have OT discussions in this issue.

framps, 2022-10-08:
eMMC storage is back again now :smile: 

So now I wait for the beta.

framps, 2022-10-18:
Is there any outlook for the beta?

timg236, 2022-10-18:
The rpi-eeprom APT package was updated today.

framps, 2022-10-18:
Great :+1: 

Not sure what I have to do now: Should I run rpi-update? Should I run rpi-eeprom-update from the eeprom repo? Or is there anything else I have to do in order to grab the beta?

peterharperuk, 2022-10-18:
rpi-update will give you the latest kernel (irrelevant to this problem) and update the "firmware" in the /boot folder (which is relevant to this problem). The /boot "firmware" fix was released a few weeks ago via rpi-update.

The rpi-eeprom APT package contains the beta bootloader. If you "sudo apt update" then "sudo apt upgrade" it updates the bootloader binaries in /lib/firmware/raspberrypi/bootloader. That's fairly irrelevant for a cm4 as the only way to update a cm4 bootloader is using usbboot.

You can do something like this to update the cm4 bootloader...
git clone https://github.com/raspberrypi/rpi-eeprom.git
cp rpi-eeprom/firmware/beta/pieeprom-2022-10-18.bin usbboot/recovery/pieeprom.original.bin
cd usbboot/recovery
./update-pieeprom.sh
../rpiboot -d.

Or wait until we update the bootloader in usbboot.

framps, 2022-10-18:
That's what I did right now:

1) Installed 04-04 lite OS
2) Executed rpi-update
3) Cloned rpi-eeprom and executed the steps you described above with rpiboot

Reboot was possible. Next I executed `sudo apt-get update; sudo apt-get upgrade` to get the latest code.

Reboot was no longer possible :cry: 

That's the UART output I get:

```
Pi: BOOTLOADER release VERSION:23aa699d DATE: 2022/10/18 TIME: 11:57:44 BOOTMODE: 0x00000006 part: 0 BUILD_TIMESTAMP=1666090664 0x34a87b72 0x00a03140 0x0007419c
PM_RSTS: 0x00001000
part 00000000 reset_info 00000000
uSD voltage 3.3V
Initialising SDRAM 'Samsung' 8Gb x1 total-size: 8 Gbit 3200
DDR 3200 0 0 8 152

Boot mode: NVME (06) order f2514
VID 0x1c5c MN BC711 NVMe SK hynix 128GB              
NVME on
MBR: 0x00002000,  524288 type: 0x0c
MBR: 0x00082000,249537200 type: 0x83
MBR: 0x00000000,       0 type: 0x00
MBR: 0x00000000,       0 type: 0x00
Trying partition: 0
type: 32 lba: 8192 oem: 'mkfs.fat' volume: ' NO NAME    '
rsc 32 fat-sectors 4033 c-count 516190 c-size 1
root dir cluster 2 sectors 0 entries 0
FAT32 clusters 516190
Trying partition: 0
type: 32 lba: 8192 oem: 'mkfs.fat' volume: ' NO NAME    '
rsc 32 fat-sectors 4033 c-count 516190 c-size 1
root dir cluster 2 sectors 0 entries 0
FAT32 clusters 516190
Read config.txt bytes     2075 hnd 0x51b
Read start4.elf bytes  2249280 hnd 0x456c5
Read fixup4.dat bytes     5399 hnd 0x4a182
0x00a03140 0x00000000 0x00001fff
MEM GPU: 76 ARM: 948 TOTAL: 1024
Firmware: 102f1e848393c2112206fadffaaf86db04e98326 Aug 26 2022 14:03:16
Starting start4.elf @ 0xfec00200 partition 0
NVME off
+
```

peterharperuk, 2022-10-18:
For some reason you don't have the latest firmware. This line should show something after "Oct 5 2022"

Firmware: 102f1e848393c2112206fadffaaf86db04e98326 Aug 26 2022 14:03:16

framps, 2022-10-18:
> For some reason you don't have the latest firmware.

I executed rpi-update which updated the firmware. Unfortunately I didn't check the UART messages and verified it happened before I updated and upgraded to the latest OS version. Looks like update and upgrade replaced the firmware with an older version. I will execute steps 1 and 2 now again and check the UART boot messages for the firmware update.

framps, 2022-10-18:
`Firmware: 23aa699d0e70ed543c1ddcd17bed6446bfb6866e Oct 18 2022 17:40:43` shows up after step 2 and step 3. 

> Looks like update and upgrade replaced the firmware with an older version

qed :smiley:

peterharperuk, 2022-10-18:
ah yes - of course, sorry that catches me out as well.

framps, 2022-10-18:
NP, so there is no way I can test the fix until the next RaspbianOS > 09-22 is published?

framps, 2022-10-19:
That's what I did now:

1) Installed 04-04 lite OS
2) Upgraded the 04-04 OS to the current version with apt
3) Executed rpi-update (has to be executed last because the OS update installs an old version of start4.elf)

Congratulations :+1: Now the OS boots successfully :smile:

framps, 2022-10-20:
I'll close this issue when the next OS Release is available and I verified an upgrade from 04-04 to this next release will boot. :wink:

framps, 2022-10-26:
@peterharperuk Is there already a target date for next OS Release or can you tell me where I can subscribe to get notified when the next OS Release is available? It's boring to check https://www.raspberrypi.com/software/operating-systems/ from time to time.

framps, 2023-02-27:
I just upgraded the 09-06-2022 image to new 01-21-2023. The system boot now again :+1: 

I close this issue now as verified.

closed 2023-02-27 by framps