RPi-Distro/repo #312: dhcpcd5 NTP hook scripts restart ntpd on lease renewal

eric1tran, 2022-10-10:
In common situations where your DHCP server does not provide any NTP server info, the dhcpcd `66-ntp.conf` fragment will continuously restart ntpd. 

The issue is reported in the dhcpcd repo [here](https://github.com/NetworkConfiguration/dhcpcd/issues/78). According to this issue, the hook scripts have been rewritten in dhcpcd 9.4.1. However, Bullseye looks to be on `Version: 1:8.1.2-1+rpt7`, which still has this hook script.

I'm looking to rebuild the Raspbian source with a fix for this. Is the source code maintained somewhere? I can't seem to find it.

seamusdemora, 2022-10-10:
Well - I'll put it this way: the source code is located somewhere. I divined that "somewhere" by dead-reckoning. See the first few steps of my "recipe" for a patch I applied - [posted here on my GitHub repo](https://github.com/seamusdemora/PiFormulae/blob/master/FixDhcpcd.md). 

If you'd like, you can make a pull-request against my code. Let me know if you're interested in this as I'll probably want to invest a little time in making a dhcpcd ver 8 stand-alone repo, and incorporating another change made to RPi's source archives (which aren't published on any git repo). 

Otherwise, if you want to share your change with the RPi archive keeper, let me know & I'll share my experience (I'll have to dig through some archived material to find that).

mutability, 2022-10-12:
I think you need to apply at least this patch from upstream Debian to the dhcpcd5 that Raspbian ships: https://salsa.debian.org/smlx-guest/dhcpcd5/-/commit/62162b20e2fb14336f6d884ec6603ebf1d3ac463

Also, the dhcpcd5 packages present in `archive.raspberrypi.org` have misleading maintainer / version-control information in their debian/control files -- they refer to the upstream Debian maintainer & VCS, which does not appear to be correct for these packages -- perhaps you could fix that at the same time?

seamusdemora, 2022-10-12:
@mutability : 

Debian's Salsa repo hasn't been maintained for a while now, and it's not the same version used by RPi. The Salsa repo for dhcpcd is at version 7; RPi is at version 8, and the [actual upstream sources](https://github.com/NetworkConfiguration/dhcpcd) are at version 9. I do agree that the *package information* for dhcpcd5 - shown via `apt-cache showsrc dhcpcd5` - is confusing.

mutability, 2022-10-12:
@seamusdemora : the bug is in hook scripts that are specific to Debian, that hook script does not exist in upstream dhcpcd

Regardless of which upstream repo is the right one, the bug continues to exist in Raspbian and the commit I linked (or something equivalent) should fix it. What do I need to do to get that fix applied to the package that Raspbian ships?

XECDesign, 2022-10-12:
> I think you need to apply at least this patch from upstream Debian to the dhcpcd5 that Raspbian ships: https://salsa.debian.org/smlx-guest/dhcpcd5/-/commit/62162b20e2fb14336f6d884ec6603ebf1d3ac463

A build containing that fix is now available from the `untested` component. Could you please try it and report back whether it resolves the problem for you?

To install it, edit `/etc/apt/sources.list.d/raspi.list` and add `untested` after `main` to the line starting with `deb`:

```
deb http://archive.raspberrypi.org/debian/ bullseye main untested
```

Then, run `sudo apt update && sudo apt install dhcpcd5` and reboot.

After that, it's best to remove the `untested` component from `raspi.list` to avoid accidentally pulling in other untested software in the future.

eric1tran, 2022-10-12:
@XECDesign That build resolves the issue. No longer seeing recurring restarts of ntp

XECDesign, 2022-10-12:
Thank for checking. The package has been moved to `main`.

closed 2022-10-12 by XECDesign

seamusdemora, 2022-10-12:
@mutability 
> @seamusdemora : the bug is in hook scripts that are specific to Debian, that hook script does not exist in upstream dhcpcd
>
> Regardless of which upstream repo is the right one, the bug continues to [...]

I'm not clear on why you think the scripts are specific to Debian... have you confirmed that by comparing to the upstream sources?

And it appears the patch has now been applied, and moved to `main` - `apt update` has picked it up in my `bullseye` systems, but not in buster` yet.

mutability, 2022-10-13:
> have you confirmed that by comparing to the upstream sources?

Yes.