RPi-Distro/repo #318: VNC server's "virtual desktop" mode broken in Raspberry Pi OS Bullseye

andrum99, 2023-01-26:
The built-in VNC server can run in "virtual desktop" mode, i.e. not backed by a real display, but instead running the desktop in memory, and allowing clients to connect via VNC in the otherwise normal way. See official documentation here: https://www.raspberrypi.com/documentation/computers/remote-access.html#creating-a-virtual-desktop. In this case, the VNC server starts up on VNC display number 1, i.e. port 5901.

This works fine on Raspberry Pi OS Buster (legacy), but on Bullseye I see two different breakages.

Breakage 1: On my Pi 4 with 4GB of RAM, the mutter window manager is used by default, and it won't run on the virtual desktop.
Here's a Pi 4 with 4GB RAM after VNCing in and starting a terminal from the menu:
[image]
Notice no window manager, terminal starts top left, panel menu no longer accessible.

This is a known problem, which for the console graphical desktop is fixed by forcing the use of openbox when the VNC server is enabled. Which leads me to breakage number 2.

Breakage 2: With the openbox window manager, the panel appears to start up and crash a few times, then vanish, when started from the virtual desktop. The final position looks like this on a Pi 4B 4GB:
[image]

Notice no panel, so the desktop is unusable.
(The few white pixels in the middle of the top of the window client area is the VNC toolbar drawn by the VNC viewer, which drops down when you mouseover it).

andrum99, 2023-01-26:
I've done some more testing to see if I could get it working with openbox by disabling some of the panel applets. It seems the applets entitled `Volume Control (PulseAudio)` and `Microphone Control (PulseAudio)` are the ones that cause the panel to crash wen run on the virtual desktop. Removing both of those from the panel, using the console's graphical desktop, then restarting the VNC virtual desktop allows the panel to start up OK and stay running, making the VNC session usable again.

andrum99, 2023-01-26:
Just for completeness, here's a screenshot of it working correctly on a Pi 4 with 4GB RAM with openbox, after disabling the two aforementioned panel applets:
[image]

andrum99, 2023-01-26:
Minor issue: the default X cursor doesn't get set correctly:
[image]

andrum99, 2023-01-26:
And here's what it looks like connecting to the VNC "virtual desktop" mode on a Pi 2 running Raspberry Pi OS Buster:
[image]
Note the volume control applet works and doesn't crash the panel. (Although since the VNC server doesn't support sending audio, it doesn't do anything useful).

XECDesign, 2023-01-26:
@spl237

spl237, 2023-01-26:
At a guess, the fact that audio isn't supported means those plugins are looking for some sort of virtual audio device which they cannot connect to.

Given the likelihood is that this area will be changing significantly in future, investigating problems with the current remote desktop configuration are a low priority - if I get time I'll have a look.