RPi-Distro/repo #324: Trying to change screen orientation crashes Xorg

aakoskin, 2023-03-26:
A fresh up-to-date 64-bit bullseye desktop install on Raspberry Pi 3 Model B with the official 7" touchscreen.

Trying to change the screen orientation with

  Preferences > Screen Configuration > Layout > Screens > DSI-1 > Orientation

always results in Xorg server crash. The same happens also when trying with xranrd.

Crash log:

```
[    14.709] 
X.Org X Server 1.20.11
X Protocol Version 11, Revision 0
[    14.710] Build Operating System: linux Debian
[    14.710] Current Operating System: Linux raspberrypi 6.1.19-v8+ #1637 SMP PREEMPT Tue Mar 14 11:11:47 GMT 2023 aarch64
[    14.710] Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_headphones=0 snd_bcm2835.enable_headphones=1 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_hdmi=0 video=Composite-1:720x480@60i vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  console=tty1 root=PARTUUID=d018ffcb-02 rootfstype=ext4 fsck.repair=yes rootwait
[    14.710] Build Date: 15 March 2023  02:39:27PM
[    14.711] xorg-server 2:1.20.11-1+rpt2+deb11u5 (https://www.debian.org/support) 
[    14.711] Current version of pixman: 0.40.0
[    14.711] 	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
[    14.711] Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    14.711] (==) Log file: "/var/log/Xorg.0.log", Time: Sun Mar 26 20:49:36 2023
[    14.740] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    14.788] (==) No Layout section.  Using the first Screen section.
[    14.788] (==) No screen section available. Using defaults.
[    14.788] (**) |-->Screen "Default Screen Section" (0)
[    14.788] (**) |   |-->Monitor "<default monitor>"
[    14.790] (==) No device specified for screen "Default Screen Section".
	Using the first device section listed.
[    14.790] (**) |   |-->Device "kms"
[    14.790] (==) No monitor specified for screen "Default Screen Section".
	Using a default monitor configuration.
[    14.790] (**) Option "Debug" "dmabuf_capable"
[    14.790] (==) Automatically adding devices
[    14.790] (==) Automatically enabling devices
[    14.790] (==) Automatically adding GPU devices
[    14.790] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    14.796] (WW) The directory "/usr/share/fonts/X11/misc" does not exist.
[    14.797] 	Entry deleted from font path.
[    14.797] (WW) The directory "/usr/share/fonts/X11/cyrillic" does not exist.
[    14.797] 	Entry deleted from font path.
[    14.797] (WW) The directory "/usr/share/fonts/X11/100dpi/" does not exist.
[    14.797] 	Entry deleted from font path.
[    14.797] (WW) The directory "/usr/share/fonts/X11/75dpi/" does not exist.
[    14.797] 	Entry deleted from font path.
[    14.797] (WW) The directory "/usr/share/fonts/X11/Type1" does not exist.
[    14.797] 	Entry deleted from font path.
[    14.797] (WW) The directory "/usr/share/fonts/X11/100dpi" does not exist.
[    14.797] 	Entry deleted from font path.
[    14.797] (WW) The directory "/usr/share/fonts/X11/75dpi" does not exist.
[    14.797] 	Entry deleted from font path.
[    14.797] (==) FontPath set to:
	built-ins
[    14.797] (==) ModulePath set to "/usr/lib/xorg/modules"
[    14.798] (II) The server relies on udev to provide the list of input devices.
	If no devices become available, reconfigure udev or disable AutoAddDevices.
[    14.798] (II) Loader magic: 0x555ded1e28
[    14.798] (II) Module ABI versions:
[    14.798] 	X.Org ANSI C Emulation: 0.4
[    14.798] 	X.Org Video Driver: 24.1
[    14.798] 	X.Org XInput driver : 24.1
[    14.798] 	X.Org Server Extension : 10.0
[    14.806] (++) using VT number 7

[    14.806] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[    14.814] (II) xfree86: Adding drm device (/dev/dri/card0)
[... 223 lines cut ...]
[    18.208] (**) Option "AccelerationScheme" "none"
[    18.208] (**) generic ft5x06 (79): (accel) selected scheme none/0
[    18.208] (**) generic ft5x06 (79): (accel) acceleration factor: 2.000
[    18.208] (**) generic ft5x06 (79): (accel) acceleration threshold: 4
[    18.219] (II) event5  - generic ft5x06 (79): is tagged by udev as: Touchscreen
[    18.223] (II) event5  - generic ft5x06 (79): device is a touch device
[    22.820] (II) modeset(0): Disabling kernel dirty updates, not required.
[   135.772] (EE) 
[   135.773] (EE) Backtrace:
[   135.781] (EE) 0: /usr/lib/xorg/Xorg (OsLookupColor+0x188) [0x555de30518]
[   135.782] (EE) unw_get_proc_info failed: no unwind info found [-10]
[   135.782] (EE) 
[   135.782] (EE) Segmentation fault at address 0x798a9200798a92
[   135.782] (EE) 
Fatal server error:
[   135.783] (EE) Caught signal 11 (Segmentation fault). Server aborting
[   135.783] (EE) 
[   135.783] (EE) 
Please consult the The X.Org Foundation support 
	 at http://wiki.x.org
 for help. 
[   135.783] (EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
[   135.783] (EE) 
[   135.853] (EE) Server terminated with error (1). Closing log file.
```

aakoskin, 2023-03-26:
Rotation works when glamor is enabled (raspi-config, Advanced Options > Enable/disable glamor graphics acceleration).

XECDesign, 2023-03-27:
@popcornmix Is this an issue or is the non-glamour use case no longer supported?

popcornmix, 2023-03-27:
I believe using gl is the only way rotation is handled in drm/kms world.
So, I would expect rotation to not work when glamor (hence gl rendering) is disabled.

The seg fault shouldn't really happen though. 
And it might be nice if choosing rotation + no glamor results in a "screen rotation requires glamor, do you want to enable" type dialogue.

aBUGSworstnightmare-rpi, 2023-03-27:
> I believe using gl is the only way rotation is handled in drm/kms world. So, I would expect rotation to not work when glamor (hence gl rendering) is disabled.
>
> The seg fault shouldn't really happen [...]

Sorry, but can you explain why one would need glamor on devices <2GB RAM in order to rotate the display?

Just posted some results here https://forums.raspberrypi.com/viewtopic.php?p=2094323#p2094323: rotated a display on a Pi3B with no issues - **without glamour enabled**

aakoskin, 2023-03-28:
> Just posted some results here https://forums.raspberrypi.com/viewtopic.php?p=2094323#p2094323: rotated a display on a Pi3B with no issues - **without glamor enabled**

Can you share the Xorg log from the working system?

Just in case, I also tried the 32-bit install image, it was still the same. Then I made a fresh install using 64-bit "Raspberry Pi OS with desktop" image on another RPi3B, using just a HDMI monitor, and again the behavior is the same. Rotation crashes Xorg unless glamor is enabled. I also noticed that during the setup phase, if I select "Reduce the size of the desktop..." with HDMI monitor, there is a similar crash.

So perhaps there is something wrong with the install image? Some module/package/dependency missing etc.

aBUGSworstnightmare-rpi, 2023-03-28:
Hmm .. that seems to be 'one time lucky punch'!
Tried to replicate it but not possible. X-crashes and restarts with logon-screen.

Sorry, but no idea why it has worked yesterday but refuses to work today. Also tested with a fresh install -> not working.

aakoskin, 2023-03-28:
> Hmm .. that seems to be 'one time lucky punch'! Tried to replicate it but not possible. X-crashes and restarts with logon-screen.
> 
> Sorry, but no idea why it has worked yesterday but refuses to work today. Also tested with a fresh install -> not working.

I found out that it will work if you edit /usr/share/X11/xorg.conf.d/20-noglamor.conf and replace:

`Option "AccelMethod" "msdri3"`
with
`Option "AccelMethod" "none"`

It seems that "msdri3" is something that is newly introduced for older Pi devices (< 4), and the previous disable glamor setting was "none". Probably what happened to you was that you played around with raspi-config so that it rewrote the file with that new setting (and the previous file was from an older install, maybe check from a timestamp of the file).

The acceleration option in raspi-config should be tristate: glamor, msdri3 or none. The current one is totally confusing. Also msdri3 should be enabled by default as it clearly has issues.

aakoskin, 2023-03-28:
This is probably the change after which the rotation got broken:

```
xorg-server (2:1.20.11-1+rpt1+deb11u5) bullseye; urgency=medium

  * glamor: Avoid using GL_QUADS on V3D
  * fb: do not use pixman_blt to perform copies. This is faster on Raspberry
    Pi
  * modesetting: Add DRI3 support to modesetting without glamor
  [ Simon Long ]
  * Enable dmabuf by default

 -- Rob Thomas <rob.thomas@raspberrypi.com>  Thu, 09 Feb 2023 15:15:54 +0000

```

aakoskin, 2023-03-30:
The issue is now fixed in today's xorg-server update. Great job, thanks!

```
xorg-server (2:1.20.11-1+rpt3+deb11u6) bullseye; urgency=medium

  * modesetting: msdri3 should use dumb buffers for randr
    - https://github.com/RPi-Distro/repo/issues/324

  [ Rob Thomas ]
  * modesetting: Fix DPMS behaviour on all RPi devices
  * glamor: Avoid using GL_QUADS on V3D
  * fb: do not use pixman_blt to perform copies. This is faster on Raspberry
    Pi
  * modesetting: Add DRI3 support to modesetting without glamor

  [ Simon Long ]
  * Enable dmabuf by default

 -- Serge Schneider <serge@raspberrypi.com>  Thu, 30 Mar 2023 14:46:14 +0100
```

closed 2023-03-30 by aakoskin