RPi-Distro/repo #327: package raspberrypi-kernel-headers 1:1.20230405-1 do not include headers for kernel8

Pcdwarf, 2023-04-30:
raspberrypi-kernel-headers 1:1.20230405-1 do not include headers for kernel8 

```
root@raspberrypi:~# grep ^Model /proc/cpuinfo
Model		: Raspberry Pi 4 Model B Rev 1.5
```

```
root@raspberrypi:~# apt update
Hit:1 http://raspbian.raspberrypi.org/raspbian bullseye InRelease
Hit:2 http://archive.raspberrypi.org/debian bullseye InRelease   
Reading package lists... Done                             
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
```

```
root@raspberrypi:~# dpkg -l | grep raspberrypi-kernel-headers
ii  raspberrypi-kernel-headers             1:1.20230405-1                   armhf        Header files for the Raspberry Pi Linux kernel
```

```
root@raspberrypi:~# uname -a
Linux raspberrypi 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64 GNU/Linux
```

```
root@raspberrypi:~# ls -l /usr/src/
total 16
drwxr-xr-x 25 root root 4096 30 avril 17:08 linux-headers-6.1.21+
drwxr-xr-x 25 root root 4096 30 avril 17:08 linux-headers-6.1.21-v7+
drwxr-xr-x 25 root root 4096 30 avril 17:08 linux-headers-6.1.21-v7l+
drwxr-xr-x  3 root root 4096 30 oct.   2021 sense-hat
```

XECDesign, 2023-05-02:
https://github.com/raspberrypi/linux/issues/5408

Pcdwarf, 2023-05-02:
As said in https://github.com/raspberrypi/linux/issues/5408
I switched to 64bit version of PiOS

everything works as expected.

closed 2023-05-02 by Pcdwarf