RPi-Distro/repo #330: Buster Lite (5/3/2023) Keyboard Disconnects

K4ZAD, 2023-07-10:
I have used Buster Lite (last install was a 2021 version?) on RPi 1 Version 1 for many years with no problems. Recent SD card corruption forced me to burn the 5/3/2023 version to the card. Now everything appears OK to start with. However, if either of my wired USB keyboards (US settings) are unused for a while, they fails to work when I try to use them again. I don't know the time to failure. This has forced several cycles of unplugging the power - card corruption - burn the OS again (sometimes to a second card) - only to experience the same failure. 

Seems like a bug. Fix needed, or Is it possible to download an earlier version of Buster lite?

K4ZAD, 2023-07-18:
Please cancel this bug report.
Finally having the Keyboard disconnected immediately upon a fresh boot made me consider this as a hardware problem. Examination of the USB socket solder connections showed some to be questionable, and re-soldering them cured the problem.

XECDesign, 2023-07-19:
Out of curiosity, did that solve the SD corruption issues you were seeing? Did you see any under-voltage warnings?