RPi-Distro/repo #333: nodered preinst script fails if nodered was not already installed

antoneliasson, 2023-08-06:
nodered 2.2.3-2 packaged for bullseye on http://archive.raspberrypi.org/debian contains this preinst script:
```
#!/bin/sh -e
service nodered stop >/dev/null 2>&1; exit 0
```
The invocation of `service` fails with exit status 5 if nodered was not already installed, because the nodered service does not exist. Due to `sh -e` the whole script terminates early, causing apt to fail the installation:
```
Selecting previously unselected package nodered.
Preparing to unpack .../273-nodered_2.2.3-2_armhf.deb ...
dpkg: error processing archive /tmp/apt-dpkg-install-5RPgJ8/273-nodered_2.2.3-2_armhf.deb (--unpack):
 new nodered package pre-installation script subprocess returned error exit status 5
```
One way to fix it is to change the preinst script to
```
#!/bin/sh -e
service nodered stop || true
```
so that the return status from `service` is ignored. I tested this locally by manually repackaging the deb archive and installing it with dpkg. Afterwards I was able to "upgrade" to the version from apt without errors.

antoneliasson, 2023-08-06:
Ping maintainer @dceejay

dceejay, 2023-08-11:
Thanks - will update the script.

audax1764, 2024-07-12:
For any non-advanced user like me who might fall into this pithole (current raspbian seems to still ship the broken version), here's how I applied the fix mentioned by @antoneliasson:

```
# Download package into current directory
apt-get download nodered

# Unpack into temporary dir
mkdir tmp
dpkg-deb -R nodered_2.2.3-2_armhf.deb tmp

# Now, edit the script
nano tmp/DEBIAN/preinst

# Repack deb file
dpkg-deb -b tmp nodered_2.2.3-2_armhf.deb

# Install fixed deb file
sudo dpkg -i nodered_2.2.3-2_armhf.deb
```

XECDesign, 2025-04-14:
Closing as an upstream nodered issue.

closed 2025-04-14 by XECDesign