RPi-Distro/repo #339: Screen blanking not working

sakittelsen, 2023-08-15:
Latest Raspberry Pi OS on a Pi 400, the screen blanking does not work. It's turned on in raspi-config. Only modifications done to the monitor is setting it to 50Hz.

popcornmix, 2023-08-15:
Are you saying blanking was working before you changed monitor refresh rate?

sakittelsen, 2023-08-15:
I will have to try it, I always run 50hz.
-------- Opprinnelig melding --------Fra: popcornmix ***@***.***> Dato: 15.08.2023  19:40  (GMT+01:00) Til: RPi-Distro/repo ***@***.***> Ko: sakittelsen ***@***.***>, Author ***@***.***> Emne: Re: [RPi-Distro/repo] Screen blanking not working (Issue #339) 
Are you saying blanking was working before you changed monitor refresh rate?

sakittelsen, 2023-08-15:
How many minutes before the screen blanker kicks in?

I have also used Raspberry Pi Lite OS with XFCE, and when I set the screen
blanker in XFCE, the screen will blank for 2 seconds and then come back to
the desktop. This will repeat with each timeout for the screen blanker. The
same might happen with stock Raspberry Pi OS, I just need to know when I
should look  for those 2 seconds of blank screen. I installed a fresh copy
without any modifications, and the screen was not blank after 10 minutes.

sakittelsen, 2023-08-24:
It's confirmed on both Raspberry Pi 4B and 400. After 10 minutes in Raspi OS the screen will go blank for about 2-3 seconds, and then go back to the desktop again. This will repeat every 10 minutes. This will happen on a fresh install without any changes to the OS. With Raspi OS lite and Gnome the screen will go blank a few seconds, and then the login screen will be shown. Other DE's will have a blank screen with just the mouse cursor showing. The monitor is a 24" AOC LCD with HDMI input.

popcornmix, 2023-08-24:
I have just flashed "Raspberry Pi OS with desktop" 64-bit Release data: May 3rd 2023. ([here](https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-64-bit)).

I booted on a Pi3 connected to a 4k monitor.
I did the minimum I could, so I chose username and password. Skipped wifi setup. Skipped updates. Said desktop fitted monitor. I rebooted when asked.

Then I left it for 10 minutes. Screen went blank. And it is still blank some minutes later.
Waggle mouse and it comes back on.

I have hdmi monitor, ethernet, usb mouse and keyboard connected.

So I need to know what is different with your setup.

If you could repeat my exact steps and report if that fails.
If it does, then we need to see if anything in your environment is causing it.
Can you disconnect mouse, keyboard and ethernet after the reboot, and see if behaviour changes?

If not , can you add `vc4.force_hotplug=1` to `cmdline.txt` (on existing line).
That will rule out any waggling of hotplug the display may be doing when it detects the blanking.

sakittelsen, 2023-08-24:
Copy:
If not , can you add vc4.force_hotplug=1 to cmdline.txt (on existing line).

Adding this command fixed the screen blanking, and it's now working.

popcornmix, 2023-08-24:
At a guess, your monitor is detecting the blanking and doing something with hotplug detect (most monitors do not do this).
Possibly it is deasserting/asserting hotplug to indicate the EDID has changed (not quite sure what the change would be).
That has the side effect of triggering X to update its list of hdmi modes supported and reevaluate the best one to choose.
This behaviour very likely resets the screen blanking state.

If you are curious, you can enable debugging. I think this may generate suitable info:
```
echo 0x14 | sudo tee /sys/module/drm/parameters/debug
```
Run that before the blanking occurs, the grab `dmesg` output after blanking has engaged, then aborted.
(obviously this is only interesting without `vc4.force_hotplug=1` present).