RPi-Distro/repo #343: 100% CPU in PCManFM on Buster

AnthonyRKing, 2023-09-17:
This relates directly to #https://github.com/RPi-Distro/repo/issues/145, where I left a comment as I was surprised to see there had been a fix for the '100% of one CPU consumed by PCManFM' issue, which I am currently affected by.

The original poster, @basilrabi, like me, had an RPi 3B running Buster. They never came back to say the fix worked for them, so I wonder whether or not the fix became available for Buster, or just for Bullseye.

basilrabi, 2023-09-17:
I think buster is not supported anymore. I was not able to reproduce the issue anymore so I guess it was fixed. I'm using bullseye now.

spl237, 2023-09-18:
We're no longer supporting buster. The fix is for bullseye only.

closed 2023-09-18 by spl237