RPi-Distro/repo #344: kernel8.img missing in /boot after (re)installing raspberrypi-kernel on RaspberryPi 3

jorisvergeer, 2023-09-21:
I have a lot raspberry pies running on raspbian bullseye 64bit. So I boot from kernel8.img.

I was shocked that when I tested the reinstallation/upgrade on one of my devices by running 
```sudo apt install raspberrypi-kernel --reinstall```

There somewhere in the logs is:
```
You do not have enough space in /boot to install this package.
Skipping Pi 4 support
```

In this snippet in postinst of raspberrypi-kernel diversions of the kernel.img are removed under conditions:

```
if [ "$SKIP_FILES" != "1" ] || [ "${SKIP_PI4}" = "0" ]; then
  if [ -f /usr/share/rpikernelhack/kernel7l.img ]; then
    rm -f /boot/kernel7l.img
    dpkg-divert --package rpikernelhack --rename --remove /boot/kernel7l.img
    sync
  fi
  if [ -f /usr/share/rpikernelhack/kernel8.img ]; then
    rm -f /boot/kernel8.img
    dpkg-divert --package rpikernelhack --rename --remove /boot/kernel8.img
    sync
  fi
fi
```

Here the condition of "SKIP_PI4" is not met since it is 1 on a RPI3, and thus the diversion of kernel8 is not removed, leaving /boot devoid of a kernel8.img file. 

I think that this condition should not even be in the arm64 build of the package since kernel8.img is the only kernel present in the arm64 build of raspberrypi-kernel, and the diversion should always be removed in this case.

XECDesign, 2023-09-21:
Yup, that doesn't sound great. I'll take a look, thanks.

jorisvergeer, 2023-09-25:
@XECDesign Will you have looked into this, and if you also conclude that it needs to be fixed, that this is done before the next kernel update?

And, do you need some info from me to speed things up?

RoiArthurB, 2023-09-26:
I can relate to this issue, and reinstalling both kernel and bootloader (`sudo apt install raspberrypi-kernel raspberrypi-bootloader --reinstall`) as jorisvergeer advised didn't fix it for me.

However, interestingly, the `kernel8.img` is still in `/usr/share/rpikernelhack` : 

```
pi@raspberrypi:~ $ ls -lh /boot
total 26M
-rwxr-xr-x 1 root root  28K Sep 26 05:19  bcm2708-rpi-b.dtb
-rwxr-xr-x 1 root root  29K Sep 26 05:19  bcm2708-rpi-b-plus.dtb
-rwxr-xr-x 1 root root  28K Sep 26 05:19  bcm2708-rpi-b-rev1.dtb
-rwxr-xr-x 1 root root  28K Sep 26 05:19  bcm2708-rpi-cm.dtb
-rwxr-xr-x 1 root root  28K Sep 26 05:19  bcm2708-rpi-zero.dtb
-rwxr-xr-x 1 root root  29K Sep 26 05:19  bcm2708-rpi-zero-w.dtb
-rwxr-xr-x 1 root root  30K Sep 26 05:19  bcm2709-rpi-2-b.dtb
-rwxr-xr-x 1 root root  30K Sep 26 05:19  bcm2709-rpi-cm2.dtb
-rwxr-xr-x 1 root root  30K Sep 26 05:19  bcm2710-rpi-2-b.dtb
-rwxr-xr-x 1 root root  32K Sep 26 05:19  bcm2710-rpi-3-b.dtb
-rwxr-xr-x 1 root root  32K Sep 26 05:19  bcm2710-rpi-3-b-plus.dtb
-rwxr-xr-x 1 root root  30K Sep 26 05:19  bcm2710-rpi-cm3.dtb
-rwxr-xr-x 1 root root  31K Sep 26 05:19  bcm2710-rpi-zero-2.dtb
-rwxr-xr-x 1 root root  31K Sep 26 05:19  bcm2710-rpi-zero-2-w.dtb
-rwxr-xr-x 1 root root  52K Sep 26 05:19  bcm2711-rpi-400.dtb
-rwxr-xr-x 1 root root  52K Sep 26 05:19  bcm2711-rpi-4-b.dtb
-rwxr-xr-x 1 root root  52K Sep 26 05:19  bcm2711-rpi-cm4.dtb
-rwxr-xr-x 1 root root  38K Sep 26 05:19  bcm2711-rpi-cm4-io.dtb
-rwxr-xr-x 1 root root  50K Sep 26 05:19  bcm2711-rpi-cm4s.dtb
-rwxr-xr-x 1 root root  52K Sep 26 05:21  bootcode.bin
-rwxr-xr-x 1 root root  142 Jun 24  2021  cmdline.txt
-rwxr-xr-x 1 root root 1.6K Nov 30  2021  config.txt
-rwxr-xr-x 1 root root  19K Sep 26 05:19  COPYING.linux
-rwxr-xr-x 1 root root 3.1K Sep 26 05:21  fixup_cd.dat
-rwxr-xr-x 1 root root 7.1K Sep 26 05:21  fixup.dat
-rwxr-xr-x 1 root root  10K Sep 26 05:21  fixup_db.dat
-rwxr-xr-x 1 root root  10K Sep 26 05:21  fixup_x.dat
-rwxr-xr-x 1 root root  145 Jun 27  2018  issue.txt
-rwxr-xr-x 1 root root 6.5M Sep 26 05:19  kernel7.img
-rwxr-xr-x 1 root root 6.2M Sep 26 05:19  kernel.img
-rwxr-xr-x 1 root root 1.6K Sep 26 05:21  LICENCE.broadcom
-rwxr-xr-x 1 root root  19K Jun 27  2018  LICENSE.oracle
drwxr-xr-x 2 root root  24K Sep 26 05:21  overlays
-rwxr-xr-x 1 root root 787K Sep 26 05:21  start_cd.elf
-rwxr-xr-x 1 root root 4.6M Sep 26 05:21  start_db.elf
-rwxr-xr-x 1 root root 2.9M Sep 26 05:21  start.elf
-rwxr-xr-x 1 root root 3.6M Sep 26 05:21  start_x.elf
drwxr-xr-x 2 root root  512 Feb 16  2021 'System Volume Information'

pi@raspberrypi:~ $ ls -lh /usr/share/rpikernelhack
total 25M
drwxr-xr-x   2 root root 4.0K Sep 26 05:21 .
drwxr-xr-x 191 root root  12K Apr 19 04:34 ..
-rw-r--r--   1 root root 3.1K Apr  5 13:32 fixup4cd.dat
-rw-r--r--   1 root root 5.3K Apr  5 13:32 fixup4.dat
-rw-r--r--   1 root root 8.2K Apr  5 13:32 fixup4db.dat
-rw-r--r--   1 root root 8.2K Apr  5 13:32 fixup4x.dat
-rw-r--r--   1 root root 6.9M Apr  5 13:32 kernel7l.img
-rw-r--r--   1 root root 7.9M Apr  5 13:32 kernel8.img
-rw-r--r--   1 root root 787K Apr  5 13:32 start4cd.elf
-rw-r--r--   1 root root 3.6M Apr  5 13:32 start4db.elf
-rw-r--r--   1 root root 2.2M Apr  5 13:32 start4.elf
-rw-r--r--   1 root root 2.9M Apr  5 13:32 start4x.elf
```

jorisvergeer, 2023-09-29:
@RoiArthurB Are you using the 32bit build of raspiOS? That's different form mine, that version includes both 32bit and 64bit kernels and bootloader files.

That version requires more space in /boot so probably in your case SKIP_FILES was probably still set to "1".  Whereas in my situation having all kernel and bootloader files redirected to /usr/share/rpikernelhack, left SKIP_FILES to "0" is my setup. 

kernel8.img is indeed still in /usr/share/rpikernelhack because the postinst script doesn't cleanup the redirection for kernel8.img, so that is never directing it back to /boot.

RoiArthurB, 2023-09-29:
@jorisvergeer oh yeah you're right. I have an old remote 32bit install. I thought I could use your documented issue to migrate to a 64bits kernel 😅

XECDesign, 2023-10-02:
> @XECDesign Will you have looked into this, and if you also conclude that it needs to be fixed, that this is done before the next kernel update?
> 
> And, do you need some info from me to speed things up?

It's fairly high on the todo list, but not as high as it normally would be, because of the pi5 launch.

I'll take a look before any further bullseye raspberrypi-kernel updates go out.