RPi-Distro/repo #347: Package newer kernels?

solskogen, 2023-10-13:
I know I can install newer kernels with rpi-update, but I would rather have them packaged so I can install them the "normal" way with apt instead. Could you do that, please? :-)

ghollingworth, 2023-10-13:
We do package new kernels for apt update.  That's how everyone gets kernel updates...  rpi-update is specifically only for testing individual changes that we make in the process of developing a new release.

But we can't just haphazardly release random changes to the 55million Raspberry Pis in the world, each patch goes into an rpi-update kernel for some time where it is tested by lots of people (generally led by the LibreELEC.tv project who usually take that kernel), before being merged across into the master branch (from which the apt package is created).

So if you need a particular fix, you'll have to use rpi-update in the short term until the next apt package is released

closed 2023-10-13 by ghollingworth

solskogen, 2023-10-13:
Sorry, I was being unclear. I was thinking about the other branches like 6.3, 6.4, 6.5 and 6.6

ghollingworth, 2023-10-13:
I think the biggest problem here is that at any one time there is only one supported and reliable kernel version.  What you suggest would make it too easy for people to accidentally end up with a bleeding edge kernel version which introduces as many issues as it fixes.

In the short-term while there is a fix in rpi-update you require, you should probably use that, with the knowledge that this is not a stable kernel and you are doing so at your own risk.