RPi-Distro/repo #348: Warning from 'apt update' in Raspberry Pi OS Bookworm 32-bit

reraikes, 2023-10-16:
root@raspberrypi:~# apt update
Hit:1 http://archive.raspberrypi.com/debian bookworm InRelease
Hit:2 http://raspbian.raspberrypi.com/raspbian bookworm InRelease
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
W: http://raspbian.raspberrypi.com/raspbian/dists/bookworm/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.

MichaIng, 2023-12-14:
```sh
ln -s /etc/apt/trusted.gpg.d/raspbian-archive-keyring.gpg /usr/share/keyrings/raspbian-archive-keyring.gpg
rm /etc/apt/trusted.gpg
apt update
```

WLBI, 2024-06-05:
The fix works great, thanks! 

> ln -s /usr/share/keyrings/raspbian-archive-keyring.gpg /etc/apt/trusted.gpg.d/raspbian-archive-keyring.gpg
> rm /etc/apt/trusted.gpg

XECDesign, 2024-06-12:
Closing since the issue is with the Raspbian repo. https://bugs.launchpad.net/raspbian

closed 2024-06-12 by XECDesign

MichaIng, 2024-06-12:
I created a bug report years ago about this already. I'll refresh it, since now a 2nd warning appeared: https://bugs.launchpad.net/raspbian/+bug/1727874

However, AFAIK, Raspbian maintainers are volunteers, hence it might not be solved quickly, and it would be better to contribute the fix directly. Does anyone know whether Raspbian's own package code is hosted somewhere publicly?

The issue is in `raspbian-archive-keyring`'s `postinst` maintainer script:
```sh
#! /bin/sh
# postinst script for raspbian-archive-keyring

set -e

case "$1" in
    configure)
        apt-key add /usr/share/keyrings/raspbian-archive-keyring.gpg > /dev/null
    ;;

    remove|upgrade|failed-upgrade|abort-install|abort-upgrade|disappear)
    ;;

    *)
        echo "postinst called with unknown argument \`$1'" >&2
        exit 1
    ;;
esac

# dh_installdeb will replace this with shell code automatically
# generated by other debhelper scripts.

#DEBHELPER#

exit 0
```
`apt-key add` is deprecated, and stores the key in the deprecated `/etc/apt/trusted.gpg`. The command could replaced with above symlink generation instead, or the package could ship `/etc/apt/trusted.gpg.d/raspbian-archive-keyring.gpg` directly, which is how `debian-archive-keyring` does it, though in ASCII-armored format since Bookworm.

beta-tester, 2024-11-13:
it is still an issue in [2024-10-22-raspios-bookworm-armhf-lite.img.xz](https://downloads.raspberrypi.org/raspios_lite_armhf/images/raspios_lite_armhf-2024-10-28/2024-10-22-raspios-bookworm-armhf-lite.img.xz)

[Warning from apt-key: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see te DEPRECATION](https://forums.raspberrypi.com/viewtopic.php?t=358016)

why is that issue closed, when it is still not fixed?