RPi-Distro/repo #350: Bookworm who command not returning correct information

pcollinson, 2023-10-26:
The 'who' command on 32-bit Raspbian bookworm returns strange output.

```
pc       tty1         1970-01-11 01:36
pc       tty7         1970-01-01 16:59 (:0)
pc       pts/0        723576232734386770 (192.168.10.10)
pc       pts/1        723576232734131466 (192.168.10.10)
```

I guess that it's not reading the information in /var/run/wtmp correctly. I've tried reloading the coreutils package, and its binaries come back dated Sep 20  2022 -- perhaps it needs recompiling?

popcornmix, 2023-10-26:
Yes, can confirm.
Bookworm on pi4 64-bits is okay:
```
pi@pi4:~ $ who
pi                    2023-10-25 18:00
pi       tty1         2023-10-25 18:00
pi       pts/0        2023-10-25 18:00 (10.3.14.142)
```
Bookworm on pi4 32-bits is surprising:
```
pi@pi4:~ $ who
pi                    1970-01-12 00:06
pi       tty1         1970-01-02 11:00
pi       pts/0        -8210621729164976728 (10.3.14.142)
```

popcornmix, 2023-10-26:
Forum had a link to this: https://debbugs.gnu.org/cgi/bugreport.cgi?bug=64937

pcollinson, 2023-10-27:
Getting to the end of this Debian bug, it seems that there are fixes 'upstream' to make things work on Raspbian from August 2023, but they don't seem to have materialised in the current release.