RPi-Distro/repo #353: apt update hangs on 512MB models running desktop

trejan, 2023-11-24:
There have been a few people complaining that apt update fails on Pi Zero 2 running desktop edition. The symptom is that it hangs during the reading changelogs step. The cause appears to be insufficient RAM as increasing CONF_SWAPSIZE in /etc/dphys-swapfile from 100 to 512 fixes it.

Running desktop on a 512MB model isn't optimal but updates should still work even if its slow. 64-bit Bookworm Lite does update without any problems but that has much more free RAM without the GUI running.

The dphys-swapfile package is already modified by Raspberry Pi to set 100MB so maybe time to increase the default?

XECDesign, 2024-01-02:
This seems to be happening because bookworm's version of dpkg has introduced multi-threaded decompression for xz by default. On the original, it would work fine, but then on a zero 2, using additional cores makes it eat more RAM.

The next image will have a version of sys-mods which limits dpkg to a single thread if the device does not have enough RAM. Don't know enough to say whether we should also increase swap.

Murray2015, 2024-05-26:
Hello! Dropping in to say a [few people on this thread](https://forums.raspberrypi.com/viewtopic.php?p=2224570&hilit=zero+2#p2224570) (inc. myself) are still finding this issue in the prod image. However, I don't have enough experience to be able to offer any help; sorry! 🙂

XECDesign, 2024-05-27:
Thanks for the heads up. Looks like the dpkg fix wasn't enough in some cases and bumping swap is necessary.

I'll try defaulting to 200 instead and monitor if that's enough.

https://github.com/RPi-Distro/raspberrypi-sys-mods/commit/1f88fa7abe59d878920cd02582a5f1c1ec69f531
https://github.com/RPi-Distro/pi-gen/commit/75fe47c7571c533ad52c43e03b440b0116a157ea

znmeb, 2024-06-02:
I've run into this issue and related ones in `dpkg` with every Bookworm variant I've tried on my Zero 2 Ws. The "lock up at 33%" is just one of the symptoms; I've also seen `dpkg` spin for minutes at 100 percent CPU unpacking a large file like Firefox. There just isn't enough RAM in a Pi Zero 2 W to run Bookworm - case closed.

Making a bigger swapfile is hardly a fix; swapping to an SD card every time you upgrade is nonsensical. Realistically, I'd recommend that the Raspberry Pi Imager limit the recommended options for the Pi Zero 2 W to just Bullseye and Ubuntu 22.04. Nothing else runs out of the box with no hassles.

closed 2024-06-12 by XECDesign

XECDesign, 2024-08-20:
Did you have anything else running at the time the update ran?

Kevvie250, 2024-08-20:
Yes, VNC for headless.  Other than that, right from Imager to Zero 2W and first boot

XECDesign, 2024-08-20:
Hmm, I don't remember the pre-6.6 kernels just locking up the whole system like that, which makes me wonder if there's some other underlying issue.

Maybe for now we do need to bump to 512 and take a look at zswap.

@popcornmix Any ideas? Is zram more appropriate? Are the default zswap parameters okay? Archlinux wiki says that [zsmalloc](https://docs.kernel.org/mm/zsmalloc.html) may be a better allocator for low-ram devices, but we're defaulting to zbud. Should we enable it at run-time when a 512MB device is detected or always?

popcornmix, 2025-05-30:
A desktop install is not recommended for 512M devices (even with swap).
There just isn't enough RAM for typical operations.

A lite image would be recommended.