RPi-Distro/repo #357: [Bookworm] Cleanup /boot/firmware on new kernel package removals

MichaIng, 2023-12-14:
I recognised that after removal of the new RPi kernel packages, the related kernel and initramfs images in `/boot/firmware` remain. It would make sense to clean this up via `postrm` maintainer script. I can open a PR for this, if someone can show me where the package build scripts are hosted. At least I could not find them in the Linux repo.

XECDesign, 2023-12-15:
The linux packages would install the kernel into /boot/ and then raspi-firmware contains the script that is responsible for copying and renaming the appropriate kernels. It's not on github, so I can't take a PR, but if you attach a patch for this and/or issue #358 , I can take a look. Otherwise, I can take update the package in the new year.

The bookworm image needed to be shipped a bit earlier than I wanted to, so a lot of this is marked as 'TODO' in the script.

MichaIng, 2023-12-15:
I just found the ToDo in the script:
```sh
# TODO:
# Handle removed files on upgrade
# Handle removal
# Handle custom kernel?
# Handle upstream kernel
# Handle wrong partition
```
So this patch handles the removal:
```diff
--- /etc/kernel/postinst.d/z50-raspi-firmware.old       2023-11-14 13:43:16.000000000 +0100
+++ /etc/kernel/postinst.d/z50-raspi-firmware   2023-12-15 16:09:25.759176315 +0100
@@ -93,5 +93,10 @@
       cp "$latest_kernel" "$kernel_dst"
       cp "$latest_initrd" "$initrd_dst"
     fi
+  else
+    if [ "$KERNEL" = "auto" ]; then
+      [ -e "$latest_kernel" ] || rm "$kernel_dst"
+      [ -e "$latest_initrd" ] || rm "$initrd_dst"
+    fi
   fi
 done
```
The whole block:
```sh
for flavour in $FLAVOURS; do
  # Copy kernel files to /boot/firmware
  if [ "$KERNEL" = "auto" ] ; then
    latest_kernel=$(ls -1 /boot/vmlinuz-*-rpi-"$flavour" 2> /dev/null | grep -E -v '\.(dpkg-bak|old-dkms)$' | sort -V -r | head -1)
    latest_initrd=$(ls -1 /boot/initrd.img-*-rpi-"$flavour" 2> /dev/null | grep -E -v '\.(dpkg-bak|old-dkms)$' | sort -V -r | head -1)
  fi

  kernel_dst="$firmware_dst/kernel$(echo "$flavour" | sed 's/^v//;s/^6//;s/2712/_2712/;').img"
  initrd_dst="$firmware_dst/initramfs$(echo "$flavour" | sed 's/^v//;s/^6//;s/2712/_2712/;')"
  case $flavour in
    v8|2712)
      dtb_path="/usr/lib/linux-image-${latest_kernel#/boot/vmlinuz-}/broadcom"
      ;;
    *)
      dtb_path="/usr/lib/linux-image-${latest_kernel#/boot/vmlinuz-}"
      ;;
  esac
  overlay_path="/usr/lib/linux-image-${latest_kernel#/boot/vmlinuz-}/overlays"
  if [ "$1" != "remove" ]; then
    if [ "$KERNEL" = "auto" ] ; then
      [ -e "$latest_kernel" ] && [ -e "$latest_initrd" ] || continue
      for dtb in "${dtb_path}"/bcm27*.dtb; do
        [ -e "${dtb}" ] || continue
        cp "${dtb}" /boot/firmware/
      done
      for dtbo in "${overlay_path}"/*.dtb* "${overlay_path}"/README; do
        [ -e "${dtbo}" ] || continue
        cp "${dtbo}" "${firmware_dst}/overlays"
      done
      if [ -e "${firmware_dst}/.firmware_revision" ]; then
        echo "WARNING: raspi-firmware: replacing rpi-update kernels" >&2
        rm -f "${firmware_dst}/.firmware_revision"
      fi
      cp "$latest_kernel" "$kernel_dst"
      cp "$latest_initrd" "$initrd_dst"
    fi
  else
    if [ "$KERNEL" = "auto" ]; then
      [ -e "$latest_kernel" ] || rm "$kernel_dst"
      [ -e "$latest_initrd" ] || rm "$initrd_dst"
    fi
  fi
done
```
- `/etc/kernel/postrm.d/z50-raspi-firmware` calls `/etc/kernel/postinst.d/z50-raspi-firmware`, so this script handles installs as well as uninstalls.
- The parent block loops through all kernel flavours.
- The added code runs if `[ "$1" != "remove" ]` is not true, hence on removals. I wanted to keep the diff minimal, but for better understanding it could be swapped to avoid doubled negation.
- I followed the condition on install to only do something if `"$KERNEL" = "auto"`, i.e. remove kernel/initrd only if it is installed as well. Side note: `INITRAMFS` is currently not used at all. I guess the idea was to handle kernel and initrd images separately?
- If no kernel/initrd image matching the flavour exists in `/boot`, as shipped by packages and generated by `initramfs-tools`, the related kernel/initrd image from `/boot/firmware` is removed.

XECDesign, 2024-01-17:
I've added new kernel and raspi-firmware packages to the 'untested' component. They should behave a bit better than what we were shipping previously.

closed 2024-01-29 by XECDesign