RPi-Distro/repo #360: lsb_release -a reports Debian on 64 bit platforms, Raspian on 32 bit Platforms

wcbonner, 2024-01-17:
I'm not sure where lsb_release pulls it's information, but I believe it is the source for the DISTRO details in CMake.

On a Pi ZeroW running Bookworm:
```
wim@WimPiZeroW:~ $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 12 (bookworm)
Release:        12
Codename:       bookworm
```

On a Pi4:
```
wim@WimPi4:~ $ lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 12 (bookworm)
Release:        12
Codename:       bookworm
```

See this post for the issues I was having with CMake: https://forums.raspberrypi.com/viewtopic.php?t=363747

popcornmix, 2024-01-17:
This is correct.

RPiOS uses the the Debian repo for arm64.
RPiOS uses the the Raspbian repo for arm32.

[raspbian](https://www.raspbian.org) is a build of debian, optimised to run on all Pi devices (specifically a 32-bit build with the hard float ABI).

wcbonner, 2024-01-17:
Is the proper method of recognizing the Raspberry platform during a CMake build documented somewhere?

popcornmix, 2024-01-17:
I think it's rare for a build to care what it is running on. These days the Pi has removed many proprietary ways of doing things, and follows standard linux APIs.

It's more likely what you really care about is 32-bit vs 64-bit, what libraries are available, hardware features etc, which all have different ways of detection.

But if you want to know if you are running on RPiOS, then /etc/rpi-issue (as mentioned in forum) seems reasonable.

closed 2024-01-18 by XECDesign