RPi-Distro/repo #361: service rc-local.service fails to execute /etc/rc.local

aspymid, 2024-01-22:
In a headless installation the rc-local.service will fail to execute the /etc/rc.local file. The failure is as follows:

Jan 21 18:29:16 dclock3 systemd[1]: Starting /etc/rc.local Compatibility...
Jan 21 18:29:16 dclock3 systemd[324]: rc-local.service: Failed to set up standard output: No such device
Jan 21 18:29:16 dclock3 systemd[324]: rc-local.service: Failed at step STDOUT spawning /etc/rc.local: No such device
Jan 21 18:29:16 dclock3 systemd[1]: rc-local.service: Control process exited, code=exited, status=209/STDOUT
Jan 21 18:29:16 dclock3 systemd[1]: rc-local.service: Failed with result 'exit-code'.
Jan 21 18:29:16 dclock3 systemd[1]: Failed to start /etc/rc.local Compatibility.

**Work around:**
Modify /etc/systemd/system/rc-local.service.d/ttyoutput.conf to read as follows:
[Service]
StandardOutput=journal+console
StandardError=journal+console

instead of:
[Service]
StandardOutput=tty

This bug is also found on raspberry pi 2w and I suspect all the others.
Steps to reproduce the behaviour

**The reproduce:**

    make a fresh installation of raspbian, (a headless installation)
    make sure that the servide rc-local.service is enabled.
    reboot
    get disagnostic of the failure with : journalctl -u rc-local

Device (s)

Raspberry Pi 1 Mod. B+
System

Raspberry Pi reference 2023-12-05
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 446f3e34c8ceb33dfe7bd38578f1cd6b6afe19da, stage4
root@dclock3:  vcgencmd version
Mar 17 2023 10:52:42
Copyright (c) 2012 Broadcom
version 82f3750a65fadae9a38077e3c2e217ad158c8d54 (clean) (release) (start)
root@dclock3:  uname -a
Linux dclock3 6.1.21+ https://github.com/raspberrypi/linux/issues/1642 Mon Apr 3 17:19:14 BST 2023 armv6l GNU/Linux

XECDesign, 2024-01-23:
It looks like Debian have added their own equivalent fix in `/lib/systemd/system/rc-local.service.d/debian.conf`, so I can just remove our drop-in altogether.

Should be fixed with the next release of raspberrypi-sys-mods.

Thanks!

XECDesign, 2024-01-29:
Should be fixed with today's `raspberrypi-sys-mods` update.

closed 2024-01-30 by XECDesign