RPi-Distro/repo #363: VS Code is stuck on 1.83.0 in the Pi Bookworm repo.

trejan, 2024-01-25:
The latest VS Code is 1.85.2 but the archive.raspberrypi.com Bookworm repo only shows 1.83.0. The Bullseye and Buster repos on archive.raspberrypi.com do have the latest 1.85.2. It is only the Bookworm repo that is out of date for some reason.

Bookworm:
```
$ apt policy code
code:
  Installed: (none)
  Candidate: 1.83.0-1696349969
  Version table:
     1.83.0-1696349969 500
        500 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages
```

Bullseye:
```
$ apt policy code
code:
  Installed: (none)
  Candidate: 1.85.2-1705560689
  Version table:
     1.85.2-1705560689 500
        500 http://archive.raspberrypi.org/debian bullseye/main arm64 Packages

```

XECDesign, 2024-01-25:
Thanks for reporting the issue. It should be resolved now.

trejan, 2024-01-25:
Thanks for the quick fix. I confirm that 1.83.2 is available from the Bookworm repo.

closed 2024-01-25 by trejan