RPi-Distro/repo #364: Chromium-browser ignores `--force-dark-mode` flag on Bookworm

Botspot, 2024-01-27:
I prefer running Chromium with full UI dark mode and have been doing that with the `--force-dark-mode` flag for years. This continues to work on latest Chromium on Raspberry Pi OS Bullseye, but on PiOS Bookworm the flag seems to have no effect.

To reproduce:
```
chromium-browser --force-dark-mode
```
And notice that it still launches with a light UI theme.
Expected behavior: The tabs list, context menus, omnibox, and other top-level widgets are darkened.

I am not sure what is causing this issue, as it does not seem to be from any new [custom patches](https://github.com/RPi-Distro/chromium-browser/tree/master/debian/patches) on your end. Also, I highly doubt that chromium dropped support for this flag at the same time that Bookworm dropped, so for this issue I am not sure where to even start looking for a cause.

Just wanted to mention that installing a dark mode theme on Chromium **is not** a complete fix. Yes, the better ones can make a fairly complete dark theme, but the selection lists that appear on right-click and when clicking the hamburger button are still in light mode. More importantly though, these dark theme extensions cannot inform websites that you are using dark mode, making several of them, including Github.com, change their behavior and introduce additional contrast issues sitewide.

ghollingworth, 2024-01-28:
Yeah, the chromium-browser script doesn't do anything with the command line argument, if you go to chrome://version you can see it's still in the command line, so it must be something in Chromium 120, or possibly related to an interaction with the system (ozone wayland backend for example).

Botspot, 2024-09-14:
This issue seems to have been resolved.
I am unsure what was responsible. Here are three possible causes:

- `chromium-browser` was upgraded to `chromium` as a result of the new packaging format.
- I managed to accidentally remove chromium:arm64 and install chromium:armhf, then I fixed it by installing chromium:arm64 again.
- I reset all of my custom command-line flags in /etc/chromium.d

And most of those occurred in one session, so I am not sure which one fixed it.

closed 2024-09-14 by Botspot