RPi-Distro/repo #366: EOM image viewer covers the screen with certain image resolutions

Botspot, 2024-01-30:
This is likely a bug with EOM, but maybe with wayfire.
With an image here that is 2263x2262 pixels, opening it using default PiOS 64-bit will launch a Image Viewer window that covers the screen.
[image]
I cannot even click the Maximize button as It covers both the taskbar and the titlebar is above the top edge of the screen. Furthermore, there is no way to drag the window around. The only way to regain control of the system is to use Alt-F4.

I am using a normal 1080p screen - nothing special here to my knowledge.
Attached below is an image with these dimensions for others to replicate.
[image]
Please note that this not some crazy, artificial edge case. This happens with about one third of images that I open, including all of the photos taken using my digital camera.

spl237, 2024-01-30:
Wayfire, like all Wayland compositors, opens windows centred on the screen.

EOM appears to be opening images at a 1:1 pixel mapping by default, and this image is larger than your screen in the vertical dimension, so a centred EOM window has the top and bottom outside the visible screen area.

There's nothing that can be done in Wayfire to fix this. I'd suggest that the EOM developers might want to change the default scaling with which an image is opened if an image is larger than a certain size, ideally based on monitor dimensions. I suggest you raise an issue at https://github.com/mate-desktop/eom

Botspot, 2024-01-30:
The strange thing here is that 2262x2262 works and 2263x2263 works - they open in a reduced zoom level.
I suspect this is more nuanced, but will try to get someone on the eom repo to reproduce it.

closed 2024-01-30 by Botspot