RPi-Distro/repo #368: Track source code for linux-image-rpi-* package releases

febus982, 2024-03-17:
**Context:**
I need to install a custom kernel on a fleet of PIs. Ideally I'd like to track the official package release versions and each time a new version is released:
* download the sources
* apply custom kernel configuration
* build kernel, modules, overlays, etc...
* build and self host an equivalent `deb` package to replace `linux-image-rpi-XX` 

**Questions**
1. How can I retrieve the original source code for a specific `linux-image-rpi-XX` version?

The current version is 6.6.20 but I was't able to find a tag or a release in https://github.com/raspberrypi/linux (they seem to maintain only the main development branch for the major version). I have also tried to download the sources using `apt-get source` but I wasn't able to build the kernel (it's not [the recommended way](https://www.raspberrypi.com/documentation/computers/linux_kernel.html) to build the kernel anyway)

2. Is the [documentation](https://www.raspberrypi.com/documentation/computers/linux_kernel.html) for building the kernel still up to date?

The documentation mention copying the overlays in `/boot/firmware` but the `linux-image-rpi-XX` has them in `/usr/lib/linux-image-XX`

3. Are the scripts used to build the official `linux-image-rpi-XX` images publicly available?

It would be great if they were, so it wouldn't necessary to reverse-engineer the existing `deb` packages and changes to the OS structure (such as overlays in different directories) would be inherited automatically.

ghollingworth, 2024-03-18:
We've been discussing how to develop images and updates for Pi-Gen built images (which is how we build the lite, desktop and full images).  It would be useful to understand your use case in more detail.

Email me "firstname" at raspberrypi.com

Thanks

Gordon

febus982, 2024-03-21:
> We've been discussing how to develop images and updates for Pi-Gen built images (which is how we build the lite, desktop and full images). It would be useful to understand your use case in more detail.
> 
> Email me "firstname" at raspberrypi.com
> 
> Thanks
> 
> Gordon

Hey Gordon, just in case my email ends up in a spam folder, I have sent one to you. It's coming from "firstname.lastname" at gmail [dot] com

febus982, 2024-04-01:
Hey @ghollingworth , I was wondering if there was any update on this. 👋

XECDesign, 2024-04-02:
To avoid leading you hanging for much longer, I'll leave a quick response before I have time for a full one...

> How can I retrieve the original source code for a specific `linux-image-rpi-XX` version?

The source packages are in the repo are based on Debian's packaging. The readme files in the debian/ directory explain how it works. The main thing I've done is added an 'rpi' patch set, a 'flavour' for each pi family and removed unnecessary architectures.

You can get it by using the standard tools for fetching source packages 'apt source linux' or something like `dget -u https://archive.raspberrypi.com/debian/pool/main/l/linux/linux_6.6.20-1+rpt1.dsc`

We should be adding tags to the linux repo, but we need to figure out a way to do so consistently. Keep in mind that the packaging adds additional patches, so the generated kernels won't be identical.

> Is the [documentation](https://www.raspberrypi.com/documentation/computers/linux_kernel.html) for building the kernel still up to date?

It's still up to date. Debian packages are unable to install files to filesystems which don't support permissions. The previous workaround was to use the dpkg-divert mechanism to install them to another location and then remove the divert in postinst. Now this is handled by raspi-firmware's kernel postinst hooks instead.

>  Are the scripts used to build the official `linux-image-rpi-XX` images publicly available?

Everything you need to build the .deb files is in the source package. For your use case, you'd  just need to change the .config files, update the changelog and rebuild.

The main thing missing is how I generated the .config files in the first place. I'll add it to the source package at some point, but it's not required to build the .deb files.

The plan is to push it all to github at some point.

febus982, 2024-04-05:
> Keep in mind that the packaging adds additional patches, so the generated kernels won't be identical.

That is the main problem, we'd need the exact source code including the patches. I see that a similar discussion is happening here: https://github.com/raspberrypi/linux/issues/6063

Let's keep a single thread going on.

XECDesign, 2024-04-05:
> That is the main problem, we'd need the exact source code including the patches.

It's all in the source package.

febus982, 2024-04-06:
> The main thing missing is how I generated the .config files in the first place. I'll add it to the source package at some point, but it's not required to build the .deb files.

I assume this is what is making impossible to run `make bcm2711_defconfig`, causing this:
```
  HOSTCC  scripts/basic/fixdep
  HOSTCC  scripts/kconfig/conf.o
  HOSTCC  scripts/kconfig/confdata.o
  HOSTCC  scripts/kconfig/expr.o
  LEX     scripts/kconfig/lexer.lex.c
  YACC    scripts/kconfig/parser.tab.[ch]
  HOSTCC  scripts/kconfig/lexer.lex.o
  HOSTCC  scripts/kconfig/menu.o
  HOSTCC  scripts/kconfig/parser.tab.o
  HOSTCC  scripts/kconfig/preprocess.o
  HOSTCC  scripts/kconfig/symbol.o
  HOSTCC  scripts/kconfig/util.o
  HOSTLD  scripts/kconfig/conf
***
*** Can't find default configuration "arch/arm64/configs/bcm2711_defconfig"!
***
make[2]: *** [scripts/kconfig/Makefile:94: bcm2711_defconfig] Error 1
make[1]: *** [/opt/kernel/aaa/linux-6.6.20/Makefile:708: bcm2711_defconfig] Error 2
make: *** [Makefile:246: __sub-make] Error 2
```

I have tried this process as a workaround for the missing config.

* Download the source with `apt source linux` (also `apt source linux-image-6.6.20+rpt-rpi-v8`, but they point to the same source package)
* Download the `deb` package (i.e. `apt download linux-image-6.6.20+rpt-rpi-v8`) and extract it with `dpkg -x downloaded_file.deb /some/output/folder`
* Use the config file in the deb package as base config: `cp /some/output/folder/boot/config-6.6.20+rpt-rpi-v8 /path/to/kernel/source/.config`
* Build and package the kernel in a new deb file

When building the kernel some of the configuration appears not to be set, which seems to suggest the source not being the same used to build the `deb` package. I might be missing something here.

After running a `make olddefconfig` and comparing the two configs this is the diff:

```
diff .config .config.old 
5c5
< CONFIG_CC_VERSION_TEXT="gcc (Debian 12.2.0-14) 12.2.0"
---
> CONFIG_CC_VERSION_TEXT="gcc-12 (Debian 12.2.0-14) 12.2.0"
21c21
< CONFIG_PAHOLE_VERSION=0
---
> CONFIG_PAHOLE_VERSION=124
34d33
< CONFIG_BUILD_SALT=""
1959a1959
> CONFIG_FW_CS_DSP=m
2116a2117
> CONFIG_OF_CONFIGFS=y
2160a2162
> CONFIG_BCM2835_SMI=m
3286a3289
> CONFIG_JOYSTICK_RPISENSE=m
3446a3450,3453
> CONFIG_BRCM_CHAR_DRIVERS=y
> CONFIG_BCM2708_VCMEM=y
> CONFIG_BCM_VCIO=y
> CONFIG_BCM2835_SMI_DEV=m
3562a3570
> CONFIG_RASPBERRYPI_GPIOMEM=m
3598a3607,3608
> CONFIG_I2C_BCM2708=m
> CONFIG_I2C_BCM2708_BAUDRATE=100000
3759a3770,3771
> CONFIG_PINCTRL_RP1=y
> CONFIG_PINCTRL_BCM2712=y
3783a3796
[... 146 lines cut ...]
> CONFIG_BCM_VC_SM_CMA=m
> CONFIG_VIDEO_CODEC_BCM2835=m
> CONFIG_VIDEO_ISP_BCM2835=m
6974a7092,7095
> CONFIG_COMMON_CLK_RP1=y
> CONFIG_COMMON_CLK_RP1_SDIO=y
> CONFIG_COMMON_CLK_HIFIBERRY_DACPLUSHD=m
> CONFIG_COMMON_CLK_HIFIBERRY_DACPRO=m
7042a7164
> CONFIG_BCM2712_IOMMU=y
7680a7803
> CONFIG_PWM_RP1=y
7694a7818
> CONFIG_BCM2712_MIP=y
7715a7840
> CONFIG_GENERIC_PHY_MIPI_DPHY=y
7753a7879
> CONFIG_RPI_AXIPERF=m
8475d8600
< CONFIG_SYSTEM_TRUSTED_KEYS=""
8604d8728
< CONFIG_DMA_BOUNCE_UNALIGNED_KMALLOC=y
```

What would you recommend?

XECDesign, 2024-04-06:
> I assume this is what is making impossible to run `make bcm2711_defconfig`, causing this:

No, the source package contains all the config data as well.

By running 'make' directly, you're skipping all the bits that the debian packaging does, which includes applying all the appropriate patches and configs. Build it like you would any other debian package.

The README files contain a lot of the information you need to understand how the config files are being generated and how to add additional patches.

febus982, 2024-04-06:
> The README files contain a lot of the information you need to understand how the config files are being generated and how to add additional patches.

Sorry, what README files are you referring to?

XECDesign, 2024-04-06:
> Sorry, what README files are you referring to?

> The readme files in the debian/ directory explain how it works.

I think there are only two (README.Debian and README.source) and they both contain useful information and links to further information.

febus982, 2024-04-06:
Oh the ones in the downloaded source! Gotcha! I'll give a look there, thanks.

febus982, 2024-04-07:
First and foremost: Thanks @XECDesign , I think I made the initial breakthrough.

In case anyone stumbles here, This is what I did to re-package the kernel packages:

This set of packages seems to be enough (there might be some unnecessary one like ncurses):

```
apt-get install -y --no-install-recommends \
    build-essential gcc \
    ca-certificates kmod \
    git bc bison flex libssl-dev \
    make libc6-dev libncurses5-dev \
    crossbuild-essential-arm64 dpkg-dev \
    rsync cpio debhelper quilt dh-exec
```

Steps:

1. Download the source: `apt-get source linux-image-rpi-v8` (I specifically need this version but I believe you can simply do `apt-get source linux` because all the configs are downloaded anyway)
2. Go in the source directory `cd linux-6.6.20`
3. Modify the configuration I need: in my case it's in `debian/config/arm64/rpi/config.v8`
4. Build the kernel `make -f debian/rules.gen -j$(( $(nproc) * 2 )) binary-arch_arm64_rpi_v8`

This produces the following deb packages in the parent directory (`..`):

```
linux-headers-6.6.20+rpt-rpi-v8_6.6.20-1+rpt1_arm64.deb
linux-headers-rpi-v8_6.6.20-1+rpt1_arm64.deb
linux-image-6.6.20+rpt-rpi-v8-dbg_6.6.20-1+rpt1_arm64.deb
linux-image-6.6.20+rpt-rpi-v8_6.6.20-1+rpt1_arm64.deb
linux-image-rpi-v8-dbg_6.6.20-1+rpt1_arm64.deb
linux-image-rpi-v8_6.6.20-1+rpt1_arm64.deb
```

I unpacked the `linux-image-rpi-v8_6.6.20-1+rpt1_arm64.deb` and verified the config file in `boot` directory contains my changes 🚀 

------------

This is still using the same package and image name `vmlinuz-6.6.20+rpt-rpi-v8` which might (or not) create some conflicts if installed together with the official package. 

@XECDesign Do you think there is an easy way to change the kernel LOCALVERSION and build another package? (i.e. `vmlinuz-6.6.20+rpt-rpi-v8-custom`)?

I fear the only way is changing `debian/control` file, which will need a new version of all configs/pre/post, plus regenerate `debian/rules.gen` (and I don't have a clue where to begin)

XECDesign, 2024-04-09:
raspi-firmware's hooks are in charge of copying recognised kernels and initramfs files into /boot/firmware whenever appropriate. I think if you modify LOCALVERSION, you may end up with something that gets installed into /boot, but never copied to /boot/firmware. So you may also need to fork the raspi-firmware package.

You can change abiname or add LOCALVERSION

Here is where I add the '+rpt' to abiname:
```
index f7485df372..399ed8e114 100755
--- a/debian/bin/gencontrol.py
+++ b/debian/bin/gencontrol.py
@@ -638,7 +638,7 @@ linux-signed-{vars['arch']} (@signedtemplate_sourceversion@) {dist}; urgency={ur
         elif version.linux_revision_backports:
             self.abiname = f'{version.linux_upstream_full}+bpo'
         else:
-            self.abiname = f'{version.linux_upstream_full}'
+            self.abiname = f'{version.linux_upstream_full}+rpt'

         self.vars = {
             'upstreamversion': self.version.linux_upstream,

```
LOCALVERSION just comes from CONFIG_LOCALVERSION in the config files.

febus982, 2024-04-09:
I have ended up in handling the package using in the apt repository priorities so it won't ever install the official one

In case anyone needs it:

`Release` file in the root of the apt repository:

```
Archive: kernel-custom
Component: kernel-custom
Origin: my-custom-apt-repo
Label: Custom Kernel overrides
Architecture: arm64
Description: Packages for the Raspberry Pi
```

`/etc/apt/preferences.d/kernel-override`

```
Package: linux-image-rpi-v8
Pin: release o=Debian
Pin-Priority: -10

Package: linux-image-rpi-v8
Pin: release o=my-custom-apt-repo
Pin-Priority: 500
```

closed 2024-04-10 by XECDesign