RPi-Distro/repo #369: What must be added to the Lite distro to make Bluetooth work?

seamusdemora, 2024-04-02:
I go through this occasionally, and it always takes way, way too much time...  I'm talking about trying to get Bluetooth working on the Lite distro. The Desktop//"Everything" distros seem to have what's required to make Bluetooth functional. I started earlier today trying to get Bluetooth working on `bookworm` Lite, and running into the usual roadblocks/lack of information. 

I'd really like to see a "Lite+" distro that included Bluetooth **_- OR -_** something in the documentation that explains roughly what must be done. Things like "add user pi to bluetooth group", "install pulseaudio (or is it pipewire this time around?)", "set config file XYZ to ABC", etc. 

Is this something that could be considered??

XECDesign, 2024-04-02:
When you say bluetooth, are you specifically referring to bluetooth audio devices? AFAIK, things like keyboards and mice should already just work.

For audio, installing pipewire would be the simplest option. I would recommend opening an issue in the documentation repo.

seamusdemora, 2024-04-03:
Not necessarily limited to audio (though that one has vexed me for years), but also for Pi-to-Pi and Pi-to-Device data transfer.  Does that (data transfer) 'just work' also?   Frankly, I've never tried a keyboard or mouse with a Lite system - I can't actually see the point in my use case, but good to hear that it's available for others. 

WRT opening an issue in the documentation repo: I can do that, but is there some way this issue could be "transferred" over there?

XECDesign, 2024-04-03:
I don't have much experience with bluetooth file transfer, but yeah, I think that's another thing that would require something like bt-obex from bluez-tools installed.

Unfortunately I can't transfer the issue, since github doesn't allow that across organisations.

ghollingworth, 2024-04-03:
No, that is on the raspberrypi organisation so we can't transfer it there...

Please close this one and open a new one

seamusdemora, 2024-04-03:
@ghollingworth  :     ??? *"the raspberrypi organisation"*  ???

I thought **all** of this was on "the raspberrypi organisation"?  What other organisation is there?

lurch, 2024-04-04:
> I thought **all** of this was on "the raspberrypi organisation"? What other organisation is there?

The "organisation" in this case is the first part of the URL after `github.com`; so the https://github.com/RPi-Distro/repo repository is in the `RPi-Distro` organisation (for historical reasons) and the https://github.com/raspberrypi/documentation repository is in the `raspberrypi` organisation. See https://raspberrypi.github.io/

seamusdemora, 2024-04-04:
@lurch : thanks for that.

FWIW: the documentation team has [denied responsibility](https://github.com/raspberrypi/documentation/issues/3585#event-12354374925) for this. So disappointing. Why does RPi even bother to include BT in the hardware if it's not going to be supported in software and documentation? ... that's an "honest" and "earnest" question.

popcornmix, 2024-04-05:
@seamusdemora asked to document how to make bluetooth work on *bullseye* Lite.
That, understandably was rejected, as bullseye is a no longer supported OS.

A request to document how to make bluetooth work on *bookworm* Lite might get a more positive response.

JamesH65, 2024-04-06:
I wish Debian had been a little more ambitious with their naming, three OS releases in a row with the same first later was a terrible idea.

But I'd be happy to write up how to do it on Bookworm as a whitepaper. If I know how its done.

seamusdemora, 2024-04-06:
@JamesH65 : I wish I could help you there...  But I think this is an ***excellent*** idea, and I ***hope*** you follow through with it! Please let me know when that *whitepaper* is available.

seamusdemora, 2024-04-06:
> @seamusdemora asked to document how to make bluetooth work on _bullseye_ Lite. That, understandably was rejected, as bullseye is a no longer supported OS.
> 
> A request to document how to make bluetooth work on _bookworm_ Lite might get a more positive response.  

@popcornmix : 
Somehow I **doubt** that would have been the case. She didn't offer `bookworm` documentation - she said... well, you read it, but yes - `bookworm` documentation would have been great!  Has Alistair left the documentation organization?

aallan, 2024-04-07:
> @popcornmix : Somehow I doubt that would have been the case. She didn't offer bookworm documentation - she said... well, you read it, but yes - bookworm documentation would have been great! Has Alistair left the documentation organization

No, I'm still around. But as Liz said, we're a small team, and it was a pretty niche request. It's unlikely to be something we'll get around to any time soon. 

Also you understand that @LizUpton is my boss, right? 🤔

MichaIng, 2024-04-09:
> I wish Debian had been a little more ambitious with their naming, three OS releases in a row with the same first letter was a terrible idea.

Absolutely. Luckily, next is Debian **T**rixie 😄.

For Bluetooth audio with plain ALSA on Bookworm:
```sh
apt install bluez-alsa-utils
```
`/etc/alsa/conf.d/20-bluealsa.conf` defines/configures the ALSA PCM. To set it as default system-wide:
```sh
echo -e 'pcm.!default bluealsa\nctl.!default bluealsa' | sudo tee /etc/asound.conf
```
or for the current user:
```sh
echo -e 'pcm.!default bluealsa\nctl.!default bluealsa' > ~/.asoundrc
```
So since Bookworm, I wouldn't say that using PipeWire makes it easier, when using RPi OS Lite.

In case PipeWire is used:
```sh
apt install libspa-0.2-bluetooth
```

For PulseAudio:
```sh
apt install pulseaudio-module-bluetooth
```
But I am not sure whether the PW/PA modules need to be additionally configured, or whether it works OOTB.

Not sure about raw file transfer either. As far as I understand, the Bluetooth device must provide a known connection profile, for the host to know what to do. For this to be known/available for audio devices, a BlueALSA, PA or PW instance (with BT plugin) must be running. I would assume that it works OOTB for BT storage devices, same as like for input devices. But I never tested it on Linux.

seamusdemora, 2024-04-10:
> No, I'm still around. But as Liz said, we're a small team, and it was a pretty niche request. It's unlikely to be something we'll get around to any time soon.
>
> Also you understand that @LizUpton is [...]

Glad you're still with us :)  I struggle understanding how this is a *niche* request. There is nothing in the documentation (certainly not for Lite users) on making Bluetooth work. And BT is *kinda' mainstream tech* - or at least I thought it was? 

But that's too bad - I am disappointed that no Bluetooth documentation is forthcoming. 

And I didn't know she was your boss - but will stand by my "parting shot" in the other thread.

seamusdemora, 2024-04-10:
> @seamusdemora asked to document how to make bluetooth work on _bullseye_ Lite. That, understandably was rejected, as bullseye is a no longer supported OS.
> 
> A request to document how to make bluetooth work on _bookworm_ Lite might get a more positive response.
@popcornmix : 

As we can now plainly see - bookworm did not get a more positive response. 

Also curious re yr statement that "bullseye is a no longer supported OS":  I still get the occasional update through `apt`... where does this come from if it's not supported?

lurch, 2024-04-10:
Bullseye is now our "Legacy OS" and so still gets security updates, but no further support beyond that. See https://www.raspberrypi.com/news/new-old-functionality-with-raspberry-pi-os-legacy/

aallan, 2024-04-10:
> I struggle understanding how this is a niche request. There is nothing in the documentation (certainly not for Lite users) on making Bluetooth work. And BT is kinda' mainstream tech - or at least I thought it was?

On Lite? Its niche. If I'm going to prioritise documentation around Lite it'll be more command-line management of the new Network Manager stack — which is poorly documented everywhere (not just on our site) — and is most definitely not niche.

More on Bluetooth is on the list for our [tutorials site](https://www.raspberrypi.com/tutorials/) but adding more content there isn't right at the top of our to do list right now. Too much other stuff going on. Sorry.

seamusdemora, 2024-04-10:
@lurch 
> Bullseye is now our "Legacy OS" and so still gets security updates, but no further support beyond that. See https://www.raspberrypi.com/news/new-old-functionality-with-raspberry-pi-os-legacy/

If Bullseye is legacy, why is it the only choice given for an OS to install (using Raspberry Pi Imager) on Raspberry Pi 3A+ ? 

This was a big surprise for me...  it's almost as if...  well, I won't say it, but why is this the case? Why isn't the RPi3A+ supported on bookworm?

seamusdemora, 2024-04-10:
> On Lite? Its niche. If I'm going to prioritise documentation around Lite it'll be more command-line management of the new Network Manager stack — which is poorly documented everywhere (not just on [...]

Oh, so it is 'Lite' that's "niche" - not Bluetooth? But surely there's much overlap... and there's certainly lots of precedent for documenting a "GUI" way and a "CLI way" of doing things (often the "CLI way" is much easier/faster!). 

Yes - I spend some time on "Stack Exchange", and there's been some confusion re `nmcli` vs `dhcpcd`... but then there was lots of confusion on `dhcpcd` before `nmcli` came around. Seems to me there are a lot of folks who just won't read the (usually very good) manuals/man pages, *and* lots of people who have no clue re how their network actually works - which can't be fixed w/ man pages alone.... _**But back to the point**_: There has been lots written/documented on `nmcli` both at RPi and many other places. Bluetooth OTOH seems to get almost nothing... the current documentation at RPi amounts to a few screenshots showing someone "clicking icons"... that's not what I'd call documentation. I think you'll agree on that point at least.

seamusdemora, 2024-04-10:
### In an effort to ***make my point*** re the need for Bluetooth documentation:

I just installed (and immediately updated & upgraded) a 64-bit version of RPi's `bookworm Lite` on my new-ish RPi 3A+. This is as close to pristine as one can get. I entered the following command: 
```bash
pi@rpi3a:~ $ systemctl status bluetooth.service
● bluetooth.service - Bluetooth service
     Loaded: loaded (/lib/systemd/system/bluetooth.service; enabled; preset: enabled)
     Active: active (running) since Wed 2024-04-10 22:48:45 UTC; 15min ago
       Docs: man:bluetoothd(8)
   Main PID: 599 (bluetoothd)
     Status: "Running"
      Tasks: 1 (limit: 380)
        CPU: 176ms
     CGroup: /system.slice/bluetooth.service
             └─599 /usr/libexec/bluetooth/bluetoothd

Apr 10 22:48:45 rpi3a bluetoothd[599]: src/plugin.c:plugin_init() Failed to init vcp plugin
Apr 10 22:48:45 rpi3a bluetoothd[599]: profiles/audio/mcp.c:mcp_init() D-Bus experimental not enabled
Apr 10 22:48:45 rpi3a bluetoothd[599]: src/plugin.c:plugin_init() Failed to init mcp plugin
Apr 10 22:48:45 rpi3a bluetoothd[599]: profiles/audio/bap.c:bap_init() D-Bus experimental not enabled
Apr 10 22:48:45 rpi3a bluetoothd[599]: src/plugin.c:plugin_init() Failed to init bap plugin
Apr 10 22:48:45 rpi3a bluetoothd[599]: Bluetooth management interface 1.22 initialized
Apr 10 22:48:45 rpi3a bluetoothd[599]: profiles/sap/server.c:sap_server_register() Sap driver initialization failed.
Apr 10 22:48:45 rpi3a bluetoothd[599]: sap-server: Operation not permitted (1)
Apr 10 22:48:46 rpi3a bluetoothd[599]: Failed to set privacy: Rejected (0x0b)
Apr 10 22:48:50 rpi3a bluetoothd[599]: src/adv_monitor.c:btd_adv_monitor_power_down() Unexpected NULL btd_adv_monitor_manager object upon power down
``` 

9 of the 10 lines in the "diagnostic" section of this report fm `systemd` are highlighted in yellow or orange in my terminal; i.e. "problems/issues". My Bluetooth ***will not work*** with all of this. A bit of documentation would be a big help here. A bit of pre-release configuration would also be a big help. But all that seems not to matter here - nobody wants to take any responsibility for this. After many years, no one has even bothered to add the default `user` to the `bluetooth` `group`; has someone in your organization decided that Bluetooth is not going to be supported in the 'Lite' distro? 

For the life of me - I do not understand this attitude... can someone explain? If the 'Lite' distro is not to be maintained, why even have a 'Lite' distro?

seamusdemora, 2024-04-11:
I've done a survey of all the bluetooth, alsa, audio apps that came in the bookworm Lite distro I installed earlier today: 

pi-bluetooth 		Raspberry Pi 3 bluetooth; Loads BCM43430A1 firmware on boot
libbluetooth3		Library to use the BlueZ Linux Bluetooth stack
bluez				BlueZ is the official Linux Bluetooth protocol stack.
bluez-firmware		Firmware; required for BT dongles based on Broadcom BCM203x and Raspberry Pi chipset.
alsa-topology-conf	ALSA config files; contains ALSA topology configuration files used by libasound2 for specific audio hardware  
alsa-ucm-conf		ALSA Use Case Manager config files; ALSA UCM config of audio i/o names & routing for specific audio hardware; used with the alsaucm tool
alsa-utils			Utilities for configuring and using ALSA; incl alsactl, alsaucm, aplay, arecord, speaker-test & several others

I'll *guess* that this means someone thinks that Bluetooth audio can be made to work using `bluez` and `alsa`.  I'd really appreciate it if someone could spend a few minutes, and search for some kind of documentation that explains where to start with alsa and bluez. I looked for hours, and found nothing but chaos.

lurch, 2024-04-11:
> If Bullseye is legacy, why is it the only choice given for an OS to install (using Raspberry Pi Imager) on Raspberry Pi 3A+ ?

See e.g. https://github.com/raspberrypi/rpi-imager/issues/730

MichaIng, 2024-04-11:
> I'd really appreciate it if someone could spend a few minutes, and search for some kind of documentation that explains where to start with alsa and bluez.

See my post above: https://github.com/RPi-Distro/repo/issues/369#issuecomment-2045753694

ghollingworth, 2024-04-11:
@seamusdemora 

You seem to believe we are responsible for maintaining the whole Linux bluetooth subsystem and that someone at Raspberry Pi would have some special knowledge that makes what you ask "easy for them".  But that's just not true, we do not, and have not ever done the thing you are trying to do and therefore we no better than you at searching the various systems, applications and audio interfaces to understand this.

On the desktop image we instead base our support on something originally implemented in an lxplugin and just update it to keep it working where we can.

We'd be happy to add some documentation for the Bookworm lite image or point to a suitable bit of external documentation if it did exist, but we don't have the time right now to research, develop, bug-fix and maintain this on the lite image.  I was under the impression that it was possible, but it's not something I've done myself.

Gordon

closed 2024-04-11 by ghollingworth

seamusdemora, 2024-04-11:
@ghollingworth 
> You seem to believe we are responsible for maintaining the whole Linux bluetooth subsystem and that someone at Raspberry Pi would have some special knowledge that makes what you ask "easy for them". [...]

Gordon,
Yeah - but you seem to believe that you have **no responsibilities!** I do believe you have responsibilities - whether you can admit them or not. I completely agree that your responsibilities **do not** extend to **"maintaining the whole Linux bluetooth subsystem"**, and I feel it's disingenuous of you to say such a thing. ... I was going to write more, but I feel that would be a waste of time in your case.

seamusdemora, 2024-04-11:
@lurch 
> > If Bullseye is legacy, why is it the only choice given for an OS to install (using Raspberry Pi Imager) on Raspberry Pi 3A+ ?
> 
> See e.g. [raspberrypi/rpi-imager#730](https://github.com/raspberrypi/rpi-imager/issues/730)

I guess that clears it up... but that's rather an *obscure* announcement. 

I don't know... nothing personal with anyone here, but I am losing faith with this Raspberry Pi thing...  esp after reading Gordon's denial of responsibility for anything and everything - that sort of did me in. Sheesh