RPi-Distro/repo #372: [Bullseye] Imagemagick convert utility results in Illegal instruction

darksidelemm, 2024-06-27:
I'm debugging an issue on a Raspberry Pi Zero W, using the imagemagick 'convert' utility. Currently this is occurring on a machine run by someone else, so apologies for the screenshots. I hope to replicate this when I'm back from a work trip and can access a RPi Zero W myself.

Platform: Raspberry Pi Zero W v1.1, running Bullseye. Kernel 6.1.21, armv6l

Essentially, when it's run, sometimes it breaks with an Illegal Instruction error:
[image]

I was a bit confused by this so confirmed what package they were using:
[image]

I've read elsewhere that this kind of problem can be from an armv7 package being used on an armv6 system. Apparently armhf means different things in raspbian and debian worlds?

Anyway. I compared the MD5sum of the packages from raspbian and debian:
raspbian: http://raspbian.raspberrypi.org/raspbian/pool/main/i/imagemagick/imagemagick_6.9.11.60+dfsg-1.3+deb11u3_armhf.deb
debian: http://security.debian.org/debian-security/pool/updates/main/i/imagemagick/imagemagick_6.9.11.60+dfsg-1.3+deb11u3_armhf.deb
Both have the same md5sum of 7d717709b5519047c3649296dadb2d4e

**Is this right? Should these packages be the same?**

I'm hoping to replicate this issue and do a bit more debugging when I can get access to a RPi Zero W next.

XECDesign, 2024-07-03:
I wouldn't have expected them to have the exact same checksums, but debian is aiming for reproducible builds, so it's not too surprising. I'll have a quick look to see if there's anything obviously wrong, but this looks more like a [Raspbian issue](https://bugs.launchpad.net/raspbian/+bugs).

darksidelemm, 2024-07-03:
Apologies if this was the wrong place to post this!  If this is the wrong place, I can post it on the raspbian bug tracker you linked...

I've also heard that Illegal Instruction errors can occur when the RPis overheat, and that underclocking can help reduce how often this occurs. Hopefully this ends up just being some kind of overheating bug, but I was wondering about the armv6 vs armv7 stuff...

popcornmix, 2024-07-08:
@darksidelemm can you try adding `over_voltage=2` to config.txt and reboot?

closed 2024-07-16 by XECDesign