RPi-Distro/repo #376: raspi-utils-core misses Breaks/Replaces on libraspberrypi-bin

MichaIng, 2024-09-03:
I hope reports regarding the `raspi-utils` DEB packages are correct here, since the other repo is for the raw executables only? https://github.com/raspberrypi/utils

Else I can re-open the issue over there.

`raspi-utils` has been split into `raspi-utils-core` and other packages, which is great since one has now much more flexibility to select the actually needed tools. However, `raspi-utils-core` contains `vcgencmd` but has no Breaks/Replaces on `libraspberrypi-bin`, hence the migration from old/Bullseye to new/Bookworm packages currently fails. The Breaks/Replaces is still present on the `raspi-utils` meta package, where it is not needed, since it is a dummy package only. `Provides: libraspberrypi-bin` however still makes sense only on the dummy package, since only that one pulls in all the tools previously part of `libraspberrypi-bin`?

XECDesign, 2024-09-06:
This is the right place and I can reproduce the issue. Will push a fixed version a bit later today.

Thanks!

XECDesign, 2024-09-06:
I'm having a bit of a problem with that approach. While it makes it possible to install `raspi-utils-core` without any issues, installing `raspi-utils` doesn't work:

```
$ sudo apt install raspi-utils
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 raspi-utils-core : Breaks: libraspberrypi-bin
 raspi-utils-dt : Breaks: libraspberrypi-bin
E: Unable to correct problems, you have held broken packages.

```

``` diff
 Package: raspi-utils-core
 Architecture: any
-Breaks: raspi-utils (<< 20240402-4)
-Replaces: raspi-utils (<< 20240402-4)
+Breaks: raspi-utils (<< 20240402-4), libraspberrypi-bin,
+Replaces: raspi-utils (<< 20240402-4), libraspberrypi-bin,
 Depends:
  ${shlibs:Depends},
  ${misc:Depends},
@@ -58,9 +56,11 @@ Description: Tools for reading and setting Raspberry Pi OTP bits
 Package: raspi-utils-dt
 Architecture: any
 Breaks:
+ libraspberrypi-bin,
  ovmerge (<< 20230123-1),
  raspi-utils (<< 20240402-4),
 Replaces:
+ libraspberrypi-bin,
  ovmerge (<< 20230123-1),
  raspi-utils (<< 20240402-4),
 Depends:  ${misc:Depends}, ${perl:Depends}, ${shlibs:Depends}, device-tree-compiler,
```

I've probably got something wrong there but I've always found that section of Debian's policy manual a bit vague.

XECDesign, 2024-09-07:
The main problem was the `Provides: libraspberrypi-bin` line for raspi-utils. I'll need to clean up other packages which reference libraspberrypi-bin. At least one of them is provided by a third party, so it may take a while to sort this out fully.

MichaIng, 2024-09-08:
So a package can provide and break another package. But a dependency cannot break a package provided by the dependant? Nasty. I mean usually such breaks, replaces, provides are on the very same replacement package. But when splitting features into multiple packages, having a meta package to depend on all of them, it sort of makes sense to keep the "provides" on that one.

Does it maybe work if `raspi-utils` itself keeps the `Breaks: libraspberrypi-bin` as well? But maybe the `Provides: libraspberrypi-bin` is not so important in this case.

XECDesign, 2024-09-09:
> a dependency cannot break a package provided by the dependant?

Yeah, that's my understanding from trying a few different approaches.

> Does it maybe work if `raspi-utils` itself keeps the `Breaks: libraspberrypi-bin` as well? But maybe the `Provides: libraspberrypi-bin` is not so important in this case.

Tried that as well, with no luck. I think the 'Provides' line is safe to remove as long as there are no other packages that depend on `libraspberrypi-bin`. In most packages I had `raspi-utils | libraspberrypi-bin`, which works fine, but there are a few older ones that need fixing.

XECDesign, 2024-09-11:
Shipped a version with these changes.

MichaIng, 2025-01-02:
I forgot to say "thanks", it is working as expected, thank you very much! And wishing you and the RPi team a Happy New Year 2025 🙂 🥳 !

closed 2025-01-02 by MichaIng

XECDesign, 2025-01-02:
Happy new year! And thanks for all the great and detailed issue reports and help with solving them over the years.