RPi-Distro/repo #378: scratch3 does not run on Pi2

koshoid, 2024-10-05:
Package: scratch3
Version: 3.30.9

On a current Raspberry Pi OS (bookworm, armv7l) running on a Pi2, I installed scratch3. After starting it with scratch-desktop, it took several minutes just to show the error message that WebGL is not supported. If this is intended and there is actually a hardware requirement not met, this should be in the package description to save the user from the experience of downloading, installing and booting the software just for an error message.

XECDesign, 2024-10-05:
The scratch 3 package is not managed by us. I think the right place to report the issue is here - https://github.com/RaspberryPiFoundation/scratch-desktop

XECDesign, 2024-10-05:
Actually, there's no issue tracker there. @spl237 Do you know who the right contact is?

closed 2024-10-05 by XECDesign

spl237, 2024-10-05:
Greg Annandale, I think.

But we've never supported Scratch 3 on older platforms - that's why Scratch 1 is still available. A Pi 2 doesn't have the processing power for 3.

XECDesign, 2024-10-06:
Thanks

@grega, any idea about this one?

grega, 2024-10-07:
Thanks for the feedback all. We'll update the package description to make the requirements clearer, as suggested 👍 

I've also enabled issues on https://github.com/RaspberryPiFoundation/scratch-desktop/issues