RPi-Distro/repo #379: Better Flatpak support

AsciiWolf, 2024-10-18:
Flatpak has become a well-established and widely used modern Linux packaging format in recent years and there are more than two thousands aarch64 apps available on [Flathub](https://flathub.org). However, the Flatpak support in Raspberry Pi OS is still far from being ideal and is not suitable for regular users.

- There is not any package manager GUI for Flatpak in Raspberry Pi OS and users are forced to use a terminal.
- Flatpak is not preinstalled and its installation requires using a terminal.
- The Flatpak support in Raspberry Pi OS is still considered unofficial.

XECDesign, 2024-10-18:
This repo is mostly for reporting issues with apt packages in the archive.raspberrypi.com repo.

Is the Raspberry Pi OS situation any worse than a default Debian install? If there's something they do to support flatpaks that we don't, I'm happy to look into it.

@spl237 is responsible for the desktop UX and may have some thoughts.

AsciiWolf, 2024-10-18:
>Is the Raspberry Pi OS situation any worse than a default Debian install? If there's something they do to support flatpaks that we don't, I'm happy to look into it.

Yes, it is. Debian provides "GNOME Software" and "KDE Discover" package management GUIs, both with Flatpak plugins, and while the Flatpak support is also not installed out-of-box there, it works flawlessly after following [these simple steps](https://flatpak.org/setup/Debian). I even remember that it was/is possible to enable the Flatpak support in a few clicks by using GUI - there were buttons for it in both GNOME Software and KDE Discover on Debian.

Then there is for example Linux Mint that develops their own, very lightweight [Software Manager](https://github.com/linuxmint/mintinstall) GUI that also provides top-notch Flatpak support.

Basically every bigger Linux distro provides a good Flatpak support nowadays. Except maybe Ubuntu, where Snaps are promoted instead, and some power-user distros like Gentoo or LFS.

spl237, 2024-10-18:
There's nothing preventing you from installing GNOME Software on Pi - it's in the apt repo, so if that supports Flatpak, then that will do what you need. (As far as I am concerned, its UI is pretty flawed, so we won't be installing it by default, though.)

Given all the packages we recommend are in apt, it's not going to be worth the effort to add Flatpak support to our own Recommended Software application.

AsciiWolf, 2024-10-18:
Installing GNOME Software is an overkill and since it also provides apt support, automatic updates etc., it will add unwanted notifications and other things not wanted by the user.