RPi-Distro/repo #381: /etc/os-release should specify NAME as 'Raspberry Pi OS'

andrum993, 2024-11-07:
Firstly, apologies if this is not a good place for this issue - it seemed like the most appropriate place to raise this.

The file /etc/os-release currently reports different values for NAME in the 32-bit version of Raspberry Pi OS, compared to the 64-bit version. On 32-bit it reports `Raspbian`, while on 64-bit it reports `Debian`. In both cases this is due to that version of Raspberry Pi OS being based on Raspbian (32-bit) and Debian (64-bit).

This issue is to propose that both be changed to read `Raspberry Pi OS`. This would make it clear when the machine boots that Raspberry Pi OS is the running OS, which is the intended use of the NAME variable. The documentation for the /etc/os-release file - https://www.freedesktop.org/software/systemd/man/latest/os-release.html - does not specify whether this variable needs to be the actual distro name or not; I would argue that doing so is a good thing as it avoids any ambiguity. (It also has the handy property of removing what is hopefully the last instance of `Raspbian` from the Raspberry Pi OS UI).

Note that if this proposal is accepted, it will also be necessary to make similar changes to the PRETTY_NAME variable in the same file, and to both NAME and PRETTY_NAME in the /var/lib/os-release file.

XECDesign, 2024-11-08:
That would break way too many scripts out there which are usually just checking whether they're running on Debian or Ubuntu.

andrum993, 2024-11-08:
> That would break way too many scripts out there which are usually just checking whether they're running on Debian or Ubuntu.

Which are presumably broken on Raspbian already. Is this file the best and recommended place to determine the running distro?

I don't have a strong view on this, but I thought it was worth raising this issue.

popcornmix, 2024-11-08:
> Is this file the best and recommended place to determine the running distro?

I'd suggest checking for /etc/rpi-issue if you want to know if you are running on RPiOS.

andrum993, 2024-11-08:
OK, so it sounds like the best thing is to leave this file as it is. Thank you both for your input; I will close this issue.

closed 2024-11-08 by andrum993

XECDesign, 2024-11-08:
> Which are presumably broken on Raspbian already.

Raspbian started with their own version of that file, so scripts were written with that in mind from the start. Although I still ocasionally see cases where it causes issues.

I'd rethink why you need to know that in the first place. If you need to know something about the kernel you're running, use `uname`. If you need to know the architecture of userland packages, use `dpkg --print-architecture` and so on. I can't think of many cases where software needs to do something different based on whether it's running under Raspberry Pi OS or Debian. 

Whatever the difference between the two is that makes your software want to behave differently is the thing I'd be checking for directly.

andrum993, 2024-11-08:
Apologies. My issue with the current file contents was that at boot up Raspberry Pi OS misidentifies itself as either Raspbian or Debian, however I had misunderstood where this is set. To me the CLI should state `Raspberry Pi OS` at boot, not `Raspbian GNU/Linux` or `Debian GNU/Linux`. I think it would have been preferrable if Raspberry Pi had made this change when Raspberry Pi OS was first released, as Raspbian had done at some point (I seem to remember it was not done immediately by Raspbian).

Do you think there is any mileage is suggesting that Raspberry Pi OS change the OS name reported in /etc/issue to `Raspberry Pi OS`?

XECDesign, 2024-11-11:
I am not against updating `/etc/issue`. I've actually been meaning to make it show the IP address as well, so I'll do that when I get a chance.

XECDesign, 2024-11-11:
Unfortunately `/etc/issue` comes from base-files as well, which I don't want to adopt as a Raspberry Pi OS package. I can't override the file using `dpkg-divert` since it specifically says it shouldn't be used for conffiles.

There's /etc/issue.d for adding additional text, which I can use for the IP thing, but it doesn't really help with what's already in /etc/issue.

Another alternative could be to add `--issue-file /etc/issue.pios` to `agetty`, but messing with `ExecStart` in a drop-in service file is going to be a minefield. For example, if somebody has autologin enabled, the same mechanism is used. So they'll either not get the new issue file or autologin will stop working.

I'm not seeing a nice way to do it and it's not enough of an issue to be worth messing with the edge cases in postinst. At the end of the day, whether it says Raspberry Pi OS, Raspbian or Debian, it's not wrong.

andrum993, 2024-11-11:
No problem - thanks for considering it.