RPi-Distro/repo #382: /etc/initramfs-tools/initramfs.conf change in 2024-11-13 IMG fails in systemd-nspawn

gitbls, 2024-11-18:
The 2024-11-13 RasPiOS IMG made a change to `/etc/initramfs-tools/initramfs.conf` which creates strange breakage.

Previously, that file had `MODULES=most`. With the 2024-11-13 update that is changed to `MODULES=dep`

With `MODULES=dep` running an update/upgrade in a systemd-nspawn container elicits an error. NB Since there are no updates in apt for this just-released IMG, installing an initramfs-relevant program (e.g., btrfs-progs) causes update-initramfs to run.
```
pw/ssdy/work# ./mini-nspawn 2024-11-13-raspios-bookworm-lite.img
mount: /dev/loop0p2 mounted on /mnt/sdm.
mount: /dev/loop0p1 mounted on /mnt/sdm/boot/firmware.
Spawning container sdm on /mnt/sdm.
Press ^] three times within 1s to kill container.
root@sdm:~# apt install btrfs-progs --yes
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  liblzo2-2
Suggested packages:
  duperemove
The following NEW packages will be installed:
  btrfs-progs liblzo2-2
0 upgraded, 2 newly installed, 0 to remove and 0 not upgraded.
Need to get 769 kB of archives.
After this operation, 4781 kB of additional disk space will be used.
Get:1 http://deb.debian.org/debian bookworm/main arm64 liblzo2-2 arm64 2.10-2 [51.8 kB]
Get:2 http://deb.debian.org/debian bookworm/main arm64 btrfs-progs arm64 6.2-1+deb12u1 [717 kB]
Fetched 769 kB in 0s (15.3 MB/s)    
Selecting previously unselected package liblzo2-2:arm64.
(Reading database ... 56291 files and directories currently installed.)
Preparing to unpack .../liblzo2-2_2.10-2_arm64.deb ...
Unpacking liblzo2-2:arm64 (2.10-2) ...
Selecting previously unselected package btrfs-progs.
Preparing to unpack .../btrfs-progs_6.2-1+deb12u1_arm64.deb ...
Unpacking btrfs-progs (6.2-1+deb12u1) ...
Setting up liblzo2-2:arm64 (2.10-2) ...
Setting up btrfs-progs (6.2-1+deb12u1) ...
Processing triggers for man-db (2.11.2-2) ...
Processing triggers for initramfs-tools (0.142+rpt1+deb12u1) ...
update-initramfs: Generating /boot/initrd.img-6.6.51+rpt-rpi-v8
mkinitramfs: failed to determine device for /
mkinitramfs: workaround is MODULES=most, check:
grep -r MODULES /etc/initramfs-tools

Error please report bug on initramfs-tools
Include the output of 'mount' and 'cat /proc/mounts'
update-initramfs: failed for /boot/initrd.img-6.6.51+rpt-rpi-v8 with 1.
dpkg: error processing package initramfs-tools (--configure):
 installed initramfs-tools package post-installation script subprocess returned error exit status 1
Processing triggers for libc-bin (2.36-9+rpt2+deb12u9) ...
Errors were encountered while processing:
 initramfs-tools
E: Sub-process /usr/bin/dpkg returned an error code (1)

root@sdm:~# mount
/dev/loop0p2 on / type ext4 (rw,relatime)
/dev/loop0p1 on /boot/firmware type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,size=813580k,nr_inodes=409600)
sysfs on /sys type sysfs (ro,nosuid,nodev,noexec,relatime)
tmpfs on /dev type tmpfs (rw,nosuid,size=4096k,nr_inodes=65536,mode=755)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev,size=813580k,nr_inodes=409600)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=666)
tmpfs on /run type tmpfs (rw,nosuid,nodev,size=1627160k,nr_inodes=819200,mode=755)
tmpfs on /run/host type tmpfs (ro,nosuid,nodev,noexec,size=1627160k,nr_inodes=819200,mode=755)
/dev/sda2 on /run/host/os-release type ext4 (ro,nosuid,nodev,noexec,noatime,commit=300)
tmpfs on /run/host/incoming type tmpfs (ro,relatime,size=813580k,mode=755)
/dev/sda2 on /run/host/os-release type ext4 (rw,noatime,commit=300)
tmpfs on /run/host/incoming type tmpfs (rw,nosuid,nodev,noexec,relatime,size=813580k,mode=755)
proc on /proc type proc (rw,nosuid,nodev,noexec,relatime)
proc on /proc/sys type proc (ro,nosuid,nodev,noexec,relatime)
proc on /proc/asound type proc (ro,nosuid,nodev,noexec,relatime)
[... 155 lines cut ...]
    mount -v ${loopdev}p1 /mnt/$mpt/boot/firmware
else
    mount -v ${loopdev}p1 /mnt/$mpt/boot
fi
systemd-nspawn --directory /mnt/$mpt < $(tty)
sts=$?
#if [ $sts -eq 127 ]
#then
#    echo "? invalid kernel page size"
#fi
trap EXIT
if [ -d /mnt/$mpt/boot/firmware ]
then
    umount /mnt/$mpt/{boot/firmware,}
else
    umount /mnt/$mpt/{boot,}
fi
losetup -d $loopdev
```
Save it as `mini-nspawn`, for instance, and then:
* `mini-nspawn 2024-11-13-raspios-bookworm-arm64-lite.img`
* At the prompt: `apt install btrfs-progs --yes`
* btrfs-progs will be installed and then update-initramfs will run and the above-mentioned failure happens.

Thanks!

XECDesign, 2024-11-19:
Since 'dep' uses the current state of the system to determine which modules you're likely to need in initramfs, I don't think it's a good idea to update images like that. If you use chroot, I'm guessing you bind-mount /proc and /sys, so it inherits the system state of the host, whereas systemd-nspawn probably uses something like unshare to create its own namespaces for things.

When pi-gen builds the image, we use `most` to get the initial initramfs files and then switch over to `dep` on the final image. The idea being that when initramfs gets updated in the future, it will be a bit faster and the initramfs files will be a bit smaller.

Is that a viable approach for sdm as well?

Another thing we could do is use the 'list' option, and hard-code modules like vc4, v3d, btrfs and so on in there, but it's hard to predict a list that will be suitable for everyone.

gitbls, 2024-11-19:
> Since 'dep' uses the current state of the system to determine which modules you're likely to need in initramfs, I don't think it's a good idea to update images like that. If you use chroot, I'm [...]
If I understand your question, you're asking if sdm could use `most` when needed and then switch back to `dep`. Yes, that's actually what I was assuming I'd end up doing but wanted to be sure that I fully understood the scenario and ensure that approach was workable, which you just confirmed.

> Another thing we could do is use the 'list' option, and hard-code modules like vc4, v3d, btrfs and so on in there, but it's hard to predict a list that will be suitable for everyone.

That doesn't seem like a great approach from either the RasPiOS or sdm perspective. 

Thanks for the quick response and insight!

XECDesign, 2024-11-19:
No worries. If you have any better ideas, I'm open to them. But I don't think reverting the change altogether is the way to go.

gitbls, 2024-11-19:
Since sdm can't really predict when update-initramfs might be run, I set `most` before an apt operation and `dep` afterwards, which works perfectly at the cost of a pair of `sed` commands.

This is a perfect solution from the sdm perspective. Thanks for the insights!

gitbls, 2024-12-07:
@XECDesign

sdm works fine with the above change, but I've got a related question about `update_initramfs=yes` in /etc/initramfs-tools/update-initramfs.conf that I'd appreciate your thoughts on so that I can be sure I'm doing this correctly.

I'm seeing a strange result with the following scenario:
* lightly customized 2024-11-19 Lite IMG boots to command prompt on a Pi5 with kernel 6.6.51+rpt-rpi-2712
* run `sdm-cryptconfig` from the sdm website and don't modify `update_initramfs=yes` to `update_initramfs=all` (Note that the version on the website currently makes this change)
* After update-initramfs runs, initramfs8 is updated and includes the sdm modules:
```
pt/usr/local/sdm# lsinitramfs /boot/firmware/initramfs8 | grep sdm
etc/modprobe.d/blacklist-sdm-bluetooth.conf
etc/modprobe.d/blacklist-sdm-wifi.conf
etc/sdmcrypto
etc/sdmkeyfile
usr/bin/sdmcryptfs
usr/bin/sdmluksunlock
```
* BUT initramfs_2712 does NOT include the sdm modules. In fact, it's untouched, and unsurprisingly fails to properly boot.
```
pt/usr/local/sdm# lsinitramfs /boot/firmware/initramfs_2712 | grep sdm
pt/usr/local/sdm# 
```
If I also  modify `update_initramfs=yes` to `update_initramfs=all` then both initramfs (2712 and 8) are updated correctly.

I'm confused as to why the initramfs of the booted OS is NOT updated, but the other, non-booted initramfs IS updated. 

I want to make sure that a) I do the right thing when configuring rootfs encryption, and b) leave initramfs configured correctly for future OS updates, etc.

Thanks!

XECDesign, 2024-12-09:
The logic in kernel and initramfs-tools packages is a bit questionable. It will usually assume that the relevant kernel is the 'latest' one based on a straight forward version number comparison and v8 > 2712.

I updated initramfs-tools a few days ago, so I would expect it to behave better with 'yes' now. What version do you have installed (it should be +rpt2)

gitbls, 2024-12-09:
Yes, currently have initramfs-tools 0.142+rpt2+deb12u1.

I'll give the new version a go when it shows up.

Thanks!

XECDesign, 2024-12-09:
Hmm, no if you already have that version, then I'd expect it to do the right thing.

What does `linux-version list` say?

gitbls, 2024-12-09:
```
pit/nv# linux-version list
6.6.51+rpt-rpi-2712
6.6.51+rpt-rpi-v8
6.6.62+rpt-rpi-2712
6.6.62+rpt-rpi-v8
pit/nv# 
```

XECDesign, 2024-12-09:
In that case, I would expect 6.6.62+rpt-rpi-2712 and 6.6.62+rpt-rpi-v8 to be updated if you run `sudo update-initramfs -u`

gitbls, 2024-12-09:
> In that case, I would expect 6.6.62+rpt-rpi-2712 and 6.6.62+rpt-rpi-v8 to be updated if you run `sudo update-initramfs -u`
That's what I expected as well.

I will do a retest later this morning (GMT-8 here) to document and verify versions, outputs, etc. Will update this thread with result, or would you prefer a new issue if it has an unexpected result?

XECDesign, 2024-12-09:
New issue would be best

XECDesign, 2024-12-09:
Ah no, that's good. Thanks for confirming it's working as expected.

b4shful, 2025-11-22:
I also just experienced the config file change while updating as outlined by @ali1234 

Did anyone figure out what was going on?