RPi-Distro/repo #383: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.

beta-tester, 2024-11-21:
today is made a fresh install of `2024-11-19-raspios-bookworm-armhf-lite.img.xz` for my RPi 1 and RPi Zero W, and i still get that warning, when i do
`sudo apt update` or `sudo apt-get update`

```bash
$ sudo apt update
Get:1 http://raspbian.raspberrypi.com/raspbian bookworm InRelease [15.0 kB]
Hit:2 http://archive.raspberrypi.com/debian bookworm InRelease                                                                                                              
Fetched 15.0 kB in 12s (1205 B/s)                                                                                                                                           
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.
W: http://raspbian.raspberrypi.com/raspbian/dists/bookworm/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
```

```bash
$ sudo apt-get update
Hit:1 http://raspbian.raspberrypi.com/raspbian bookworm InRelease
Hit:2 http://archive.raspberrypi.com/debian bookworm InRelease
Reading package lists... Done
W: http://raspbian.raspberrypi.com/raspbian/dists/bookworm/InRelease: Key is stored in legacy trusted.gpg keyring (/etc/apt/trusted.gpg), see the DEPRECATION section in apt-key(8) for details.
```

```bash
$ cat /boot/firmware/issue.txt 
Raspberry Pi reference 2024-11-19
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, b15527e15ba4d9e65126a454bfec2040b909dc1c, stage2
```

```bash
$ cat /etc/os-release 
PRETTY_NAME="Raspbian GNU/Linux 12 (bookworm)"
NAME="Raspbian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

only on the Lite images of RaspiOS is observed this issue since about a year / "bookworm"

XECDesign, 2024-11-22:
Duplicate of https://github.com/RPi-Distro/repo/issues/348

closed 2024-11-22 by XECDesign

beta-tester, 2024-11-22:
i thought RaspiOS is something different from the previouse Raspbian. so it just the name changed?
i don't know if my observation is correct, but i think only the lite version does shows this issue, and the desktop version don't.

XECDesign, 2024-11-26:
The name was more of clarification.

Raspbian is a volunteer-run project to recompile Debian arm(v7)hf into arm(v6)hf. Raspberry Pi (Ltd, Trading or Foundation) has never owned that name. We used Raspbian as a base to build our armhf images on and people generally called these Raspbian although they were not done by the Raspbian organisation. This sometimes caused some confusion on the forum and on the issue trackers.

When arm64 images were added, they were build on Debian and had no trace of Raspbian in them. So that seemed like a good opportunity to put the Raspberry Pi OS  name on them to make it clearer that our images are not stock Raspbian or Debian.